espressif/arduino-esp32 #9421: esp_netif_action_start crashes in Arduino but not ESP-IDF

i-am-shodan, 2024-03-26:
### Board

LilyGo T-Dongle-S3

### Device Description

ESP32-S3

### Hardware Configuration

No

### Version

v2.0.11

### IDE Name

Arduino IDE & Platform IO

### Operating System

all

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm trying to use my board as a Ethernet NCM module. In ESP-IDF I can compile the following sketch and use this to create an ethernet interface I can send and receive packets with. I've tested the code and it works well.

I want to use the same functionality in an Arduino environment however I get a crash.

The following is a small test case that does nothing with USB, just creates the interface. The crash occurs when esp_netif_action_start is called.

### Sketch

```cpp
static esp_err_t netif_transmit(void *h, void *buffer, size_t len) {
  Serial.println("Not called in this example");
  return ESP_OK;
}

static void l2_free(void *h, void *buffer) {
  Serial.println("Not called in this example");
  free(buffer);
}

void setup() {
  Serial.begin(115200);

  delay(5 * 1000);

  uint8_t mac[8];
  if (esp_efuse_mac_get_default(mac) == ESP_OK) {
    Serial.println("Got mac address");
    esp_base_mac_addr_set(mac);
  }
  else
  {
    Serial.println("mac address error");
  }

  if (esp_netif_init() == ESP_OK) {
    Serial.println("esp_netif_init() success");
  } else {
    Serial.println("esp_netif_init() failed");
    return;
  }

  tcpip_adapter_set_default_eth_handlers();  // not sure if this actually needs to be called

  esp_netif_inherent_config_t base_cfg = {
    .flags = (esp_netif_flags_t)(ESP_NETIF_DHCP_SERVER | ESP_NETIF_FLAG_EVENT_IP_MODIFIED | ESP_NETIF_FLAG_AUTOUP),
    .get_ip_event = IP_EVENT_ETH_GOT_IP,
    .lost_ip_event = IP_EVENT_ETH_LOST_IP,
    .if_key = "usb_eth",
    .if_desc = "usb ncm config device",
    .route_prio = 10
  };
  Serial.println("Created esp_netif_inherent_config_t");

  esp_netif_driver_ifconfig_t driver_cfg = {
    .handle = (void *)1,
    .transmit = netif_transmit,
    .driver_free_rx_buffer = l2_free
  };
  Serial.println("Created esp_netif_driver_ifconfig_t");

  esp_netif_config_t cfg = {
    .base = &base_cfg,
    .driver = &driver_cfg,
    .stack = ESP_NETIF_NETSTACK_DEFAULT_ETH,
  };
  Serial.println("Created cfg, about to create new netif");

  esp_netif_t *s_netif = esp_netif_new(&cfg);
  if (s_netif != NULL) {
    Serial.println("s_netif was not NULL");
    Serial.flush();
    esp_netif_action_start(s_netif, 0, 0, 0);
    Serial.println("esp_netif_action_start returned");
  } else {
    Serial.println("esp_netif_new returned and was NULL");
  }
}

void loop() {
  Serial.println("loop");
  delay(10 * 1000);
}
```

### Debug Message

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
13:30:55.378 -> 
13:30:55.378 -> Core  0 register dump:
13:30:55.378 -> PC      : 0x4201fe5e  PS      : 0x00060630  A0      : 0x820198a4  A1      : 0x3fced3a0  
13:30:55.378 -> A2      : 0x00000001  A3      : 0x00000004  A4      : 0x3fced3dc  A5      : 0x3fcf44b0  
13:30:55.378 -> A6      : 0x00000001  A7      : 0x3fceca20  A8      : 0x00005002  A9      : 0x00000000  
13:30:55.378 -> A10     : 0x3fced3c8  A11     : 0x00000000  A12     : 0x80000020  A13     : 0x00000007  
13:30:55.378 -> A14     : 0x00000005  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
13:30:55.378 -> EXCVADDR: 0x00000015  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  
13:30:55.378 -> 
13:30:55.378 -> 
13:30:55.378 -> Backtrace: 0x4201fe5b:0x3fced3a0 0x420198a1:0x3fced3d0 0x4200d48d:0x3fced400 0x4200b59a:0x3fced430 0x4200b3ba:0x3fced470 0x4200ce23:0x3fced490

PC: 0x4201fe5e: esp_eth_ioctl at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_eth/src\esp_eth.c:349
EXCVADDR: 0x00000015

Decoding stack results
0x4201fe5b: esp_eth_ioctl at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_eth/src\esp_eth.c:407
0x420198a1: ethernetif_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/port/esp32/netif\ethernetif.c:223
0x4200d48d: netif_add at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core\netif.c:381
0x4200b59a: esp_netif_start_api at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_netif/lwip\esp_netif_lwip.c:617
0x4200b3ba: esp_netif_api_cb at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_netif/lwip\esp_netif_lwip.c:115
0x4200ce23: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api\tcpip.c:208
```

### Other Steps to Reproduce

Compile sketch, reset

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

i-am-shodan, 2024-03-26:
Serial output

```
13:30:55.055 -> Got mac address
13:30:55.055 -> esp_netif_init() success
13:30:55.055 -> Created esp_netif_inherent_config_t
13:30:55.055 -> Created esp_netif_driver_ifconfig_t
13:30:55.055 -> Created cfg, about to create new netif
13:30:55.055 -> s_netif was not NULL
13:30:55.378 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```

i-am-shodan, 2024-03-26:
I've also tried calling the following to ensure the Ethernet stack comes up. Still crashes
```
ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
```

me-no-dev, 2024-03-27:
What I can suggest is to switch to the current development version (install from Git master) and then after `ETH.begin()` you can call `esp_netif_action_start(ETH.netif(), 0, 0, 0);`. Other than that I have no idea why you are getting the exception. Must be something specific to `esp_netif_action_start`

i-am-shodan, 2024-03-27:
I can confirm that it works on the latest v3. Problem is I have a lot of dependencies that are incompatible so 2.x is what I need to target.

Jason2866, 2024-04-05:
Have you tried v2.0.14? If you are lucky it may works.

wachidsusilo, 2024-04-10:
is there any example of how to use USB NCM in arduino environment?
I want to establish tcp/ip connection over usb but the examples out there are mostly written for esp-idf.

i-am-shodan, 2024-04-10:
None that I've found. You'll struggle with anything that is lwip based as Arduino ESP32 already ships that stack.

I've had some limited success with using picotcp but really the above code should work.

Jason2866, 2024-04-10:
There are several reasons for Arduino 3.0.0 One of them is/was to optimize the network stack.
I see only the way adopt your code to or find a workaround for core 2.0.14.
Wouldn't hold my breath for waiting fixing in core 2.0.x

wachidsusilo, 2024-04-10:
> There are several reasons for Arduino 3.0.0 One of them is/was to optimize the network stack. I see only the way adopt your code to Arduino 3.0.0 or find a workaround for core 2.0.14. Wouldn't hold my breath for waiting fixing in core 2.0.x

So it's working in 3.0.0? any example?

closed 2025-01-15 by Parsaabasi