espressif/arduino-esp32 #9428: WiFiProv.ino example fails on ESP32-C6-MINI-1 - v3.x Arduino library

Dario503, 2024-03-28:
### Board

ESP32-C6 (SparkFun Qwiic Pocket Development Board)

### Device Description

None

### Hardware Configuration

USB C to MacBook development system

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE v2.3.2

### Operating System

macOS 14.4.1

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

The example code works fine on the ESP32-C3-WROOM-02 Dev board.  BLE is seen on mobile device and WiFi provisioning occurs properly using the Espressif ESP BLE Provisioning App. In the serial monitor, progress is seen, QR code is completely printed, all good.

On the ESP32-C6, in the Arduino Serial monitor the first indication of issues is right after the notification:
"Begin Provisioning using BLE", when the QR code is typically printed to the console. It stops after about 7 rows:
[image]
It never shows up in the provisioning App, program appears to have crashed.

### Sketch

```cpp
Use the source code as is from the V3.x example:

Files -> Examples -> ESP32 -> WiFiProv -> WiFiProv

/*
Please read README.md file in this folder, or on the web:
https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFiProv/examples/WiFiProv

Note: This sketch takes up a lot of space for the app and may not be able to flash with default setting on some chips.
  If you see Error like this: "Sketch too big"
  In Arduino IDE go to: Tools > Partition scheme > chose anything that has more than 1.4MB APP
   - for example "No OTA (2MB APP/2MB SPIFFS)"
*/

#include "WiFiProv.h"
#include "WiFi.h"

// #define USE_SOFT_AP // Uncomment if you want to enforce using the Soft AP method instead of BLE
const char * pop = "abcd1234"; // Proof of possession - otherwise called a PIN - string provided by the device, entered by the user in the phone app
const char * service_name = "PROV_123"; // Name of your device (the Espressif apps expects by default device name starting with "Prov_")
const char * service_key = NULL; // Password used for SofAP method (NULL = no password needed)
bool reset_provisioned = true; // When true the library will automatically delete previously provisioned data.

// WARNING: SysProvEvent is called from a separate FreeRTOS task (thread)!
void SysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {
    case ARDUINO_EVENT_WIFI_STA_GOT_IP:
        Serial.print("\nConnected IP address : ");
        Serial.println(IPAddress(sys_event->event_info.got_ip.ip_info.ip.addr));
        break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
        Serial.println("\nDisconnected. Connecting to the AP again... ");
        break;
    case ARDUINO_EVENT_PROV_START:
        Serial.println("\nProvisioning started\nGive Credentials of your access point using smartphone app");
        break;
    case ARDUINO_EVENT_PROV_CRED_RECV: {
        Serial.println("\nReceived Wi-Fi credentials");
        Serial.print("\tSSID : ");
        Serial.println((const char *) sys_event->event_info.prov_cred_recv.ssid);
        Serial.print("\tPassword : ");
        Serial.println((char const *) sys_event->event_info.prov_cred_recv.password);
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_FAIL: {
        Serial.println("\nProvisioning failed!\nPlease reset to factory and retry provisioning\n");
        if(sys_event->event_info.prov_fail_reason == WIFI_PROV_STA_AUTH_ERROR)
            Serial.println("\nWi-Fi AP password incorrect");
        else
            Serial.println("\nWi-Fi AP not found....Add API \" nvs_flash_erase() \" before beginProvision()");
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_SUCCESS:
        Serial.println("\nProvisioning Successful");
        break;
    case ARDUINO_EVENT_PROV_END:
        Serial.println("\nProvisioning Ends");
        break;
    default:
        break;
    }
}

void setup() {
  Serial.begin(115200);
  WiFi.onEvent(SysProvEvent);

// BLE Provisioning using the ESP SoftAP Prov works fine for any BLE SoC, incuding ESP32, ESP32S3 and ESP32C3.
#if CONFIG_BLUEDROID_ENABLED && !defined(USE_SOFT_AP)
    Serial.println("Begin Provisioning using BLE");
    // Sample uuid that user can pass during provisioning using BLE
    uint8_t uuid[16] = {0xb4, 0xdf, 0x5a, 0x1c, 0x3f, 0x6b, 0xf4, 0xbf,
                        0xea, 0x4a, 0x82, 0x03, 0x04, 0x90, 0x1a, 0x02 };
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name, service_key, uuid, reset_provisioned);
    log_d("ble qr");
    WiFiProv.printQR(service_name, pop, "ble");
#else
    Serial.println("Begin Provisioning using Soft AP");
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name, service_key);
    log_d("wifi qr");
    WiFiProv.printQR(service_name, pop, "softap");
#endif
}

void loop() {
}
```

### Debug Message

```plain
None
```

### Other Steps to Reproduce

Works fine on ESP32-C3 and ESP32-S3 with same source above.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-07-31:
Hi @Dario503, I did retest the WiFiProv example on ESP32-C6 and the QR code is printed correctly.
BUT on ESP32C6 BLE provisioning is currently not working, as there is an issue with BlueDroid in IDF.
The fix is already pending. I will retest once the fix is done.

P-R-O-C-H-Y, 2024-07-31:
Solved by PR in library builder https://github.com/espressif/esp32-arduino-lib-builder/pull/197

closed 2024-07-31 by P-R-O-C-H-Y