espressif/arduino-esp32 #9429: ETH_PHY_POWER set as output after ETH.begin() causes board to restart on 3.0.0-alpha3

Stanimir-Petev, 2024-03-28:
### Board

Olimex ESP32-POE

### Device Description

USB-micro cable to power the board and for monitoring.
Ethernet cable connected to the local network.

### Hardware Configuration

GPIO 34 is attached to a button with active (pressed) state 0.

### Version

latest master (checkout manually)

### IDE Name

Arduino 1.8.19

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hello!

I am an Olimex employee and recently found a strange issue with the pin of the Ethernet phy power with the latest version (at the moment of posting this issue 3.0.0-alpha3).

I was experimenting something related to an issue report from our customer and I wanted to try resetting the phy by disabling and then reenabling the power of the phy (using the phy power pin). And for my surprise the board reset right after the pin is set as output if the ethernet has already been initialized (check the sketch I am posting with this issue).
If the pin is initialized as output BEFORE the ETH.begin() call then the board doesn't reset but the pin level can't be manipulated to low and high level. Basically not behaving as an output at all.

I decided to try it with the older 2.0.14 version. Not only that it didn't reset but it worked as I expected - when the phy is high the ethernet is working (successfully pinged the given IP) and when it's low it isn't working (ping timeouts).

I tried to simplify to sketch as much as I can to get as close as possible to the problem. But from what I have seen I guess it is something related to the latest changes of the ethernet library and the way the pins are defined. Could be wrong though.

### Sketch

```cpp
#ifndef ETH_PHY_TYPE
  #define ETH_PHY_TYPE        ETH_PHY_LAN8720
  #define ETH_PHY_ADDR         0
  #define ETH_PHY_MDC         23
  #define ETH_PHY_MDIO        18
  #define ETH_PHY_POWER       12
  #if defined BOARD_HAS_PSRAM  // when PSRAM is enabled pins 16 and 17 are used for the PSRAM and alternative pins are used for respectively I2C SCL and Ethernet Clock GPIO
  #define ETH_CLK_MODE        ETH_CLOCK_GPIO0_OUT
  #else
  #define ETH_CLK_MODE        ETH_CLOCK_GPIO17_OUT
  #endif
#endif

#include <ETH.h>
#define BUTTON 34

void setup()
{
  Serial.begin(115200);
  pinMode (BUTTON, INPUT);
  
  ETH.begin();
  while (!(unsigned int)ETH.localIP());
  Serial.print ("\n\rEthernet IP address: ");
  Serial.println (ETH.localIP());
  Serial.printf ("Ethernet PHY configuration:\n\r  ETH_PHY_TYPE = %d\n\r  ETH_PHY_ADDR = %d\n\r  ETH_PHY_MDC = %d\n\r  ETH_PHY_MDIO = %d\n\r  ETH_PHY_POWER = %d\n\r  ETH_CLK_MODE = %d\n\r",
  ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_MDC, ETH_PHY_MDIO, ETH_PHY_POWER, ETH_CLK_MODE);
  
  pinMode (ETH_PHY_POWER, OUTPUT);
}

void loop()
{
  if (!digitalRead(BUTTON))
  {
    digitalWrite (ETH_PHY_POWER, !digitalRead(ETH_PHY_POWER));
    Serial.printf ("Eth phy power %d\n\r", digitalRead(ETH_PHY_POWER));
    while (!digitalRead(BUTTON));
  }
}
```

### Debug Message

```plain
Output with 3.0.0-alpha3:

10:43:19.368 -> Ethernet IP address: 192.168.0.183
10:43:19.368 -> Ethernet PHY configuration:
10:43:19.368 ->   ETH_PHY_TYPE = 0
10:43:19.368 ->   ETH_PHY_ADDR = 0
10:43:19.368 ->   ETH_PHY_MDC = 23
10:43:19.368 ->   ETH_PHY_MDIO = 18
10:43:19.368 ->   ETH_PHY_POWER = 12
10:43:19.368 ->   ETH_CLK_MODE = 3
10:43:19.368 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
10:43:19.368 -> 
10:43:19.368 -> Core  0 register dump:
10:43:19.368 -> PC      : 0x400f008d  PS      : 0x00060530  A0      : 0x800f0273  A1      : 0x3ffb9740  
10:43:19.415 -> A2      : 0xd7eafe67  A3      : 0x00000005  A4      : 0x00000000  A5      : 0x3ffb2e90  
10:43:19.415 -> A6      : 0x00000005  A7      : 0x00000001  A8      : 0x00000002  A9      : 0x3ffb9720  
10:43:19.415 -> A10     : 0x00000002  A11     : 0x400f4c00  A12     : 0x00000000  A13     : 0x00000000  
10:43:19.415 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
10:43:19.415 -> EXCVADDR: 0xd7eafe80  LBEG    : 0x400894bc  LEND    : 0x400894d2  LCOUNT  : 0xffffffff  
10:43:19.415 -> 
10:43:19.415 -> 
10:43:19.415 -> Backtrace: 0x400f008a:0x3ffb9740 0x400f0270:0x3ffb9760 0x400e5a96:0x3ffb9780 0x400e6435:0x3ffb97a0 0x400e6495:0x3ffb9820 0x400f52ca:0x3ffb9840 0x400f4c0e:0x3ffb9860 0x400e5616:0x3ffb9880
10:43:19.463 -> 
10:43:19.463 -> 
10:43:19.463 -> 
10:43:19.463 -> 
10:43:19.463 -> ELF file SHA256: a2ae87b53011ae87
10:43:19.463 -> 
10:43:19.463 -> Rebooting...
10:43:19.463 -> ets Jun  8 2016 00:22:57
10:43:19.463 -> 
10:43:19.463 -> rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
10:43:19.463 -> configsip: 0, SPIWP:0xee
10:43:19.463 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:43:19.463 -> mode:DIO, clock div:1
10:43:19.463 -> load:0x3fff0030,len:1256
10:43:19.463 -> load:0x40078000,len:13832
10:43:19.463 -> load:0x40080400,len:4
10:43:19.463 -> load:0x40080404,len:3048
10:43:19.463 -> entry 0x40080590

Output with 2.0.14 (and when I press the button few times to disable/enable phy power pin):

10:26:59.181 -> Ethernet IP address: 192.168.0.201
10:26:59.181 -> Ethernet PHY configuration:
10:26:59.181 ->   ETH_PHY_TYPE = 0
10:26:59.181 ->   ETH_PHY_ADDR = 0
10:26:59.181 ->   ETH_PHY_MDC = 23
10:26:59.181 ->   ETH_PHY_MDIO = 18
10:26:59.181 ->   ETH_PHY_POWER = 12
10:26:59.181 ->   ETH_CLK_MODE = 3
10:26:59.181 -> Eth phy power 0
10:27:02.696 -> Eth phy power 1
10:27:09.212 -> Eth phy power 0
10:27:18.868 -> Eth phy power 1
10:27:19.148 -> Eth phy power 0
10:27:19.148 -> Eth phy power 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-03-28:
@Stanimir-Petev what are you using alpha3 or latest master?

Stanimir-Petev, 2024-03-28:
> @Stanimir-Petev what are you using alpha3 or latest master?

It's 3.0.0-alpha3. My bad about the selection in the report issue dropdown menu. I edited it the issue text now.

me-no-dev, 2024-03-28:
@Stanimir-Petev does it also happen with `esp32-gateway`? And please install from git the latest master (and remove alpha3) There have been some changes in between.

Stanimir-Petev, 2024-03-28:
@me-no-dev so I have tested it with the master branch (with alpha3 removed) and in fact it behaves differently but still doesn't work like it did in 2.0.14 (tested it with ESP32-POE and ESP32-Gateway same result).
Now when I try to initialize the phy power pin as output after the ETH.begin() call instead of reseting the board (like it did in the alpha3 version) I get the following message:
`14:29:54.682 -> E (1605) esp_eth: esp_eth_driver_uninstall(263): 2 ethernet reference in use`
After which I can toggle the pin by pressing the button but it doesn't make the phy to actually turn on/off. And it is instead disabled (judging by the fact that I receive no ping reply). If I remove the line 
`pinMode (ETH_PHY_POWER, OUTPUT);`
then the ethernet works but I can't toggle the phy power pin and I think the point of having such pin is that you can enable/disable the phy via software.

Stanimir-Petev, 2024-03-28:
I tried what you suggested and it worked so I guess in that case it's not an actual issue as I initially thought.
It's just that in 2.0.14 worked like a GPIO and that confused me. Thanks for the clarification!

me-no-dev, 2024-03-28:
in 3.0.0 we introduced the peripheral manager, which keeps track of used peripherals and pins that are attached to them. When you try to use a pin for something else, it will first deinit whatever it's currently attached to, before it switches to the mode you requested (GPIO in that case, while pin was attached to ETH)

VojtechBartoska, 2024-09-12:
Closing this ticket as answered. :)

closed 2024-09-12 by VojtechBartoska