espressif/arduino-esp32 #9432: modify timestamp of an existing file

tius, 2024-03-28:
### Related area

esp32 filesystem

### Hardware specification

esp32

### Is your feature request related to a problem?

Sorry if this issue should have been addressed before. I could not find a way to set the timestamp (last modification time) for an existing file (LittleFS, SD and SD_MMC).  

Using _utimes()_ results in a linker error.

### Describe the solution you'd like

int utimes (const char *__path, const struct timeval *__tvp);

should work as expected.

### Describe alternatives you've considered

What I have tried so far:

- **setTimeCallback()**: not implemented for esp32 
- **utimes()**: Linker error 
- **lfs_setattr()**: pointer to _vfs_t_ is not available, only LittleFS supported

### Additional context

A workaround would also be fine.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-03-28:
If utimes is not available, arduino would have no way to change the time either.  Ask this at [esp-idf issues](https://github.com/espressif/esp-idf/issues).

tius, 2024-03-29:
Thank you very much @lbernstone!

It looks like I gave up too early. 

_utimes()_ is not available, but _utime()_ seems to work as expected :-)

closed 2024-03-29 by tius