espressif/arduino-esp32 #9433: Partition default for esp-idf with Arduino as component

handmade0octopus, 2024-03-29:
### Board

Custom ESP32-S3 board (N8R2)

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest development Release Candidate (RC-X)

### IDE Name

Platformio + VCS

### Operating System

Win11

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I can't seem to be able to change default partition config. I created file that I previously stored in `/boards` directory, but I do not seem to be able to use it with Arduino as component - it seems to overwrite to `default` and there is no way to disable it.

I want to create configuration that is used as stock with Arduino so I would be able to issue OTA update to boards that I supply.

### Sketch

```cpp
### boards.json

{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "boards/partitions-4MB-tinieruf2.csv"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_USB_CDC_ON_BOOT=0",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_USB_MODE=0",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM",
      "-DPCB_VER=50"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "memory_type": "qspi_qspi",
    "flash_mode": "qio",
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
[... 1620 lines cut ...]
I (516) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (522) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (529) spiram: Adding pool of 1984K of external SPI memory to heap allocator
I (537) spi_flash: detected chip: gd
I (541) spi_flash: flash io: dio
I (545) sleep: Configure to isolate all GPIO pins in sleep state
I (551) sleep: Enable automatic switching of GPIO sleep configuration
I (559) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (574) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
[   578][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
I (609) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (610) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (617) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
Firmware update started. Please wait, may take several minutes!
UPDATE ERROR!
```

### Other Steps to Reproduce

https://github.com/handmade0octopus/Gauge.S-sd-updater - trying to use this as a base

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-29:
So you have renamed your file to partitions.csv in the project directory, but it isn't using that? Do you get a warning message that the partition file was not found?

Jason2866, 2024-03-29:
Try
```json
{
  "build": {
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_USB_CDC_ON_BOOT=0",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_USB_MODE=0",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM",
      "-DPCB_VER=50"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "memory_type": "qspi_qspi",
    "flash_mode": "qio",
    "mcu": "esp32s3",
    "variant": "esp32s3",
    "partitions": "boards/partitions-4MB-tinieruf2.csv"
  },
  "connectivity": [
    "wifi",
    "can"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "openocd_board": "esp32s3-builtin.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Gauge.S V5 Module",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "http://gauge.sorek.uk/",
  "vendor": "sorek.uk"
}
```
and to override in Platformio.ini
```
[env:esp32-s3]
board = esp32-s3-your-board
board_build.filesystem = littlefs
board_build.partitions = partitions_custom.csv
```

handmade0octopus, 2024-03-29:
> So you have renamed your file to partitions.csv in the project directory, but it isn't using that? Do you get a warning message that the partition file was not found?

@lbernstone Sorry for confusion, I still have `partitions-4MB-tinieruf2.csv` in my /boards, but since I am using menuconfig to compile with espidf I also copied `partitions.csv` to main folder for it to pick it up (hopefully).

@Jason2866 thanks! Adding/moving line from arduino to top level in boards.json did the trick :)

Questions for guys there:
Can I get somewhere stock sdkconfig used for ESP32-S3? I found some in the builder repo but tbh it's lacking like 99% configuration bits, and I would like to have same configuration as starting point.

Jason2866, 2024-03-29:
@handmade0octopus There https://github.com/espressif/esp32-arduino-libs/blob/idf-release/v5.1/esp32s3/sdkconfig

handmade0octopus, 2024-03-29:
> @handmade0octopus There https://github.com/espressif/esp32-arduino-libs/blob/idf-release/v5.1/esp32s3/sdkconfig

Thank you sir!

closed 2024-03-29 by handmade0octopus