espressif/arduino-esp32 #9435: WiFi static IP configuration doesn't work after networking refactoring

JAndrassy, 2024-03-29:
### Board

any

### Device Description

any

### Hardware Configuration

any

### Version

latest master (checkout manually)

### IDE Name

any

### Operating System

any

### Flash frequency

any

### PSRAM enabled

yes

### Upload speed

any

### Description

WiFi doesn't use static IP configuration after the network refactoring. Settings are not applied. DHCP is used.

### Sketch

```cpp
WiFiClientStaticIP example
```

### Debug Message

```plain
not aplicable
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-04-02:
@JAndrassy please try https://github.com/espressif/arduino-esp32/pull/9445

JAndrassy, 2024-04-02:
output from my test
```
ttempting to connect to SSID "M20" with static IP 192.168.1.177
.
	...success

MAC: 30:AE:A4:08:B2:70
SSID: M20
BSSID: C0:25:E9:C3:2F:8A
RSSI:-56 dBm
IP Address: 192.168.1.177
gateway IP Address: 192.168.1.1
subnet IP mask: 255.255.255.0
DNS server: 192.168.1.1

Attempt to connect to port 80 on 192.168.1.1
	...success

[  2890][W][STA.cpp:157] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
Attempting to connect without resetting static IP configuration

	...success

MAC: 30:AE:A4:08:B2:70
SSID: 
BSSID: 00:00:00:00:00:00
RSSI:0 dBm
IP Address: 192.168.1.177
gateway IP Address: 192.168.1.1
subnet IP mask: 255.255.255.0
DNS server: 192.168.1.1
DNS server2: 8.8.8.8

Attempt to connect to port 80 on 192.168.1.1
[  2923][E][NetworkClient.cpp:258] connect(): connect on fd 48, errno: 118, "Host is unreachable"
	...ERROR

[  2937][W][STA.cpp:548] disconnect(): STA already disconnected.
```

me-no-dev, 2024-04-02:
enable verbose debug please. Not sure what I am looking at, but here I ran a constant loop of connect, wait 10 seconds, disconnect, wait 5 seconds, repeat. Static IP was kept the same and was able to reconnect to the same AP just fine. Anyway, your IP is there just fine.

JAndrassy, 2024-04-02:
there is nothing in verbose log for the second `begin` with static IP. I will debug it now to find out where it fails

JAndrassy, 2024-04-02:
sorry. I can't replicate it. then it happened on the first and second try. now it works. In many attempts it sometimes doesn't connect to AP in different parts of the test, but that could be a problem of my AP.

so the PR is OK and the problem is solved

closed 2024-04-02 by JAndrassy