espressif/arduino-esp32 #9437: USBSerial cannot Send Data

andychu2888, 2024-03-30:
### Board

ESP32-S3 Dev-Module

### Device Description

The device Board is ESP32-S3 bought from乐鑫 ,USB mode was set to USB-OTG(TinyUSB),other were default setting.

### Hardware Configuration

nothing

### Version

v2.0.14

### IDE Name

Arduino IDE 2.32

### Operating System

Windows 11

### Flash frequency

Flash Mode QIO 80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

//-----USB mode was set to USB-OTG(TinyUSB)
void setup() {
  USB.onEvent(usbEventCallback);
  USB.begin();
  USBSerial.onEvent(usbEventCallback);
  
  USBSerial.begin();
  
}

void loop() {
  USBSerial.printf("This is a test\n");
  delay(100);      
}
nothing happen !

### Sketch

```cpp
//-----USB mode was set to USB-OTG(TinyUSB)
void setup() {
  USB.onEvent(usbEventCallback);
  USB.begin();
  USBSerial.onEvent(usbEventCallback);
  
  USBSerial.begin();
  
}

void loop() {
  USBSerial.printf("This is a test\n");
  delay(100);      
}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[    90][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface(): Interface CDC enabled
[   131][D][esp32-hal-tinyusb.c:569] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
```

### Other Steps to Reproduce

I have tried another board, the issue still exists

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-30:
Does the [USBSerial](https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBSerial/USBSerial.ino) example work if you just change the loop function to look like yours?  Do you see the USB device in Arduino/Windows Device Manager?

SuGlider, 2024-03-30:
@andychu2888 - There should be 2 COM ports that can be used in the Arduino IDE Serial Monitor.
In the screenshot below I have COM35 as the UART0 port and COM37 as the USB CDC port (don't worry about the ESP32H2 name).

[image]

One for the UART0 **(COM35)** that will display:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[    90][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface(): Interface CDC enabled
[   131][D][esp32-hal-tinyusb.c:569] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
```

Another for the USB CDC **(COM37)** that shall output:
```
This is a test
This is a test
This is a test
This is a test
This is a test
This is a test
This is a test
This is a test
This is a test
This is a test

```

andychu2888, 2024-04-01:
[lbernstone](https://github.com/lbernstone) Yes I See the virtual com port(com 14) but no any output at all

lbernstone, 2024-04-01:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html
As mentioned, there are 2 ports.  One is USB-OTG, the other is USB-UART.  Read @SuGlider 's post carefully.

SuGlider, 2024-04-01:
@andychu2888, sorry for the basic question: 
Do you have 2 USB cables connected to the ESP32-S3 DevKit? 
Boths cable shall be also connected to the Win11 computer.
If this is correct you must see 2 COM ports available in the Arduino IDE:

[image]

SuGlider, 2024-04-01:
I see no issue...

[image]

SuGlider, 2024-04-04:
Closing the issue. Feel free to open another one if necessary.

closed 2024-04-04 by SuGlider