espressif/arduino-esp32 #9450: ESP32 S3: Cannnot start a debugging session using the integrated usb.

LordRafa, 2024-04-02:
### Board

LilyGo T relay s3

### Device Description

This board contains a ESP32S3 Wroom 1U

### Hardware Configuration

It has a I2C RCT :
static const uint8_t SDA = 16;
static const uint8_t SCL = 17;
static const uint8_t RTC_INT = 18;

And a shift register(HT74HC595ARZ):
static const uint8_t SHIFT_REG_CLK   = 5;
static const uint8_t SHIFT_REG_LATCH = 6;
static const uint8_t SHIFT_REG_DATA  = 7;

### Version

v2.0.14

### IDE Name

ArduinoIDE 2.3.2

### Operating System

Linux

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

I cannot debug using the integrated usb, I set a breakpoint at Serial.println("WOLOLO"); but openocd loos to have a problem writing on the flash

Note: I can program it and use the usb monitor without issues.

### Sketch

```cpp
// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  Serial.println("WOLOLO");
  delay(1000);  // delay in between reads for stability
}
```

### Debug Message

```plain
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 50004 for tcl connections
Info : Listening on port 50005 for telnet connections
Info : esp_usb_jtag: serial (DC:DA:0C:4C:3F:24)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 50003
Info : Listening on port 50003 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : accepting 'gdb' connection on tcp/50003
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 47 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 127 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 47 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 127 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 128 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 47 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 127 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 48 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Target halted, PC=0x403B244A, debug_reason=00000001
Error: Failed to erase flash (-1)!
Error: failed erasing sectors 0 to 11
Error: flash_erase returned -4
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
shutdown command invoked
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-04-04:
@me-no-dev - PTAL, Thanks!

LordRafa, 2024-04-14:
Any idea about why this could be happening?, I cannot find the reason

hintermayr, 2024-06-09:
@LordRafa could you solve the problem?
I have the Same message

closed 2025-01-15 by Parsaabasi