espressif/arduino-esp32 #9456: Make tweaks in sdkconfig also possible in Arduino

weoiss1998, 2024-04-05:
### Related area

General

### Hardware specification

All ESP32 modules

### Is your feature request related to a problem?

My current problem is that my ESP32 is booting too slowly. I have seen in the ESP-IDF docs that it is possible to switch off the logs so that the ESP32 boots much faster. 
I have seen that it is possible to move the entire project from Arduino to the ESP-IDF, but this involves considerable effort.

For this reason, I wanted to ask whether it is possible to make these settings accessible in the Arduino-ESP project.

Translated with DeepL.com (free version)

### Describe the solution you'd like

I would be happy if I have a config file where I can select this performance options.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lucasssvaz, 2024-04-05:
If you need to tweak the IDF configuration you can either use Arduino as a component or manually recompile the libs using https://github.com/espressif/esp32-arduino-lib-builder

SuGlider, 2024-04-08:
As already said, Arduino IDE uses precompiled libraries and it is not possible to change any sdkconfig setting.
A way to use Arduino and be able to change the sdkconfig is by using Arduino as IDF Component.
It will allow to build the Arduino project using IDF tools, such as `idf.py build` and `idf.py menuconfig`

-------
Related to the itme it takes to boot: Does your project use PSRAM? 
Arduino checks the whole memory (scanning it) before starting the sketch. 
There is a way to skip it (that takes about 500ms). 
For not testing PSRAM, just add `BYPASS_SPIRAM_TEST(true);` in the sketch code.

https://github.com/espressif/arduino-esp32/issues/5737#issuecomment-1012532384
https://github.com/espressif/arduino-esp32/pull/6135

closed 2024-10-17 by weoiss1998