espressif/arduino-esp32 #9463: Wifi SmartConfig - cannot discover ESP32S3 AP.

kristsm, 2024-04-08:
### Board

ESP32S3

### Device Description

ESP32S3 Module 16MB flash with dual USB-C connectors (native USB and FTDI)

### Hardware Configuration

ESP32S3 16MB FLASH

### Version

latest master (checkout manually)

### IDE Name

Plaformio  Espressif 6.6.0

### Operating System

MacOS 14.3.1 on M1

### Flash frequency

n/a/

### PSRAM enabled

yes

### Upload speed

115200

### Description

From mobile phone I cannot discover ESP32 Access Point, there is no new AP discovered when searching WIFI Networks. I tried to burn in another SoftAP and WIFI_AP_STA examples and they are working as expected ( discoverable from mobile phone WIFI network search). The only problem is with Smartconfig. Maybe I am doing something wrong? Should I push some button to start AP_STA?

### Sketch

```cpp
#include "WiFi.h"

void setup()
{
  Serial.begin(115200);

  // Init WiFi as Station, start SmartConfig
  WiFi.mode(WIFI_AP_STA);
  WiFi.beginSmartConfig();

  // Wait for SmartConfig packet from mobile
  Serial.println("Waiting for SmartConfig.");
  while (!WiFi.smartConfigDone())
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("SmartConfig received.");

  // Wait for WiFi to connect to AP
  Serial.println("Waiting for WiFi");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("WiFi Connected.");

  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
}

void loop()
{
  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

kristsm, 2024-04-08:
BTW, the serial prints dots (........) so it seems the code is working.

Orel-A, 2024-04-09:
Also happens on the original esp32, can't discover the created access point.

kristsm, 2024-04-09:
I discovered that actually I can connect esp32s3 to wifi with one of the smartconfig codes running on laptop when the code is provided with router ssid/bssid/pswd. But still - the module's ssid is not visible on any of the  phone/pc wifi networks.Should it be so?

bearkillerPT, 2024-04-27:
I'm also facing this issue. I need the AP created by the ESP32 to be visible to phones to use the app that configures the wifi.

kristsm, 2024-04-27:
Actually it works, just the SSID is not visible. Smartconfig code run on Android can connect and configure esp32 with AP credentials.

closed 2025-01-15 by Parsaabasi