espressif/arduino-esp32 #9465: SD.remove() fails on file name that has no extension, confuses it for a directory name

frankcohen, 2024-04-08:
### Board

ESP32-S3 Dev Module

### Device Description

Reflections open-source project uses ESP32-S3 Mini 1 with a Nand (an SD in SMT format) XTSD01GLGEAG chip. Schematic and Gerber files at https://github.com/frankcohen/ReflectionsOS/tree/main/Devices/Yertle

### Hardware Configuration

// SPI Bus
#define SPI_MOSI      35
#define SPI_MISO      37
#define SPI_SCK       36
#define SPI_SPEED     10000000

// NAND Storage
#define NAND_SPI_CS   15

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

MacOS Sonoma 14.4.1

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Remove file named /REFLECTIONS/log1 and get an error: File or Directory does not exist. It appears to confuse log1 for a directory.
Remove file named /REFLECTIONS/log1.txt and it removes the file successfully.

### Sketch

```cpp
void deleteFile(fs::FS &fs, const char * path){
    Serial.printf("Deleting file: %s\n", path);
    if(fs.remove(path)){
        Serial.println("File deleted");
    } else {
        Serial.println("Delete failed");
    }
}
```

### Debug Message

```plain
No debug info given when it fails.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-04-08:
Please provide code.  Unable to reproduce
https://wokwi.com/projects/394638383953711105

frankcohen, 2024-04-09:
The whole of the code is at https://github.com/frankcohen/ReflectionsOS/tree/main/src
The delete isn't working in logger.cpp. That's too much to send. I'm going to write a smaller sketch showing the problem. Will post tomorrow. -Frank

P-R-O-C-H-Y, 2024-06-21:
Hi @frankcohen, is this issue still valid? If yes, can you please provide as you mentioned a small reproducible sketch? Thanks

frankcohen, 2024-06-27:
Working on a write up of this and other issues found. Need another few days. -Frank

closed 2025-01-16 by Parsaabasi