espressif/arduino-esp32 #9472: Can't connect to WiFi networks that contain a '

assman, 2024-04-09:
### Board

ESP32 S3

### Device Description

It's an ESP32 S3 WROOM from Freenove. It's got 2 USB C ports, one camera connector. https://store.freenove.com/products/fnk0090

### Hardware Configuration

None

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

macOS 14.4.1 Sonoma

### Flash frequency

240 MHz (WiFi)

### PSRAM enabled

no

### Upload speed

115200

### Description

My home WiFi SSID is Varun's Network. First I used the WifiScan example from the Arduino IDE to see if the board is able to see my network and ensure that it is in range, which it is. Then I use the following code to connect the board to my WiFi. Unfortunately it doesn't work and the board keeps throwing a `WiFi.status()` of `1 (WL_NO_SSID_AVAIL)`. I changed my SSID to one without the `'` character and it is able to connect without any issues.

```
#include<WiFi.h>

const char *ssid = "Varun's Network";

const char *password = "varunvarun";
        
void initWiFi() {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.print("Connecting to WiFi ");
  Serial.print(ssid);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    Serial.print(WiFi.status());
    delay(1000);
  }
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);
  initWiFi();
  Serial.print("RRSI: ");
  Serial.println(WiFi.RSSI());
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Sketch

```cpp
#include<WiFi.h>

const char *ssid = "Varun's Network";

const char *password = "varunvarun";
        
void initWiFi() {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.print("Connecting to WiFi ");
  Serial.print(ssid);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    Serial.print(WiFi.status());
    delay(1000);
  }
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);
  initWiFi();
  Serial.print("RRSI: ");
  Serial.println(WiFi.RSSI());
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
NA
```

### Other Steps to Reproduce

I have checked for similar existing issues within this repo and outside, could not find anything with a solution. I want to be able to connect to networks with special characters because I can't control what people name their SSIDs.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-04-09:
Can you try `const char *ssid = "Varun\'s Network";`
Make very sure that you are using the same character as well (copy/paste from the serial monitor).  Sometimes browsers/OS can change those quote marks into "smart" quotes. AFAIK apostrophe is allowed in esp32 SSID names.

assman, 2024-04-10:
Are you able to connect to a network with an apostrophe? I tried the escape mechanism you suggested but still not able to connect. To confirm that it is the apostrophe causing the issue and not anything else, I did a bunch of tests. My personal hotspot SSID is called **iPhone 7**, to which the ESP32 is easily able to connect. I renamed the hotspot to **Varun's iPhone 7** and I'm facing the same issue, it just doesn't connect.

lbernstone, 2024-04-10:
Confirmed.  This is an arduino-esp32 issue.  Works in IDF
Failure: 
```
10:36:50.912 -> [     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
10:36:50.946 -> [    39][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
10:36:51.012 -> [   109][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
10:36:51.012 -> [   111][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
10:36:51.012 -> [   111][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
10:36:55.751 -> [  4828][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: lb'moto, BSSID: 5a:4f:da:76:21:5e, Reason: 2
10:36:55.751 -> [  4829][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
10:36:55.751 -> [  4836][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
```

Success:
```
10:38:28.452 -> [     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
10:38:28.485 -> [    39][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
10:38:28.551 -> [   108][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
10:38:28.551 -> [   110][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
10:38:28.585 -> [   125][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
10:38:30.043 -> [  1584][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: lbmoto, BSSID: 1e:9b:1d:36:6f:27, Channel: 1, Auth: WPA2_WPA3_PSK
10:38:30.043 -> [  1585][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
10:38:30.076 -> [  1616][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.120.252
10:38:30.076 -> [  1617][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
10:38:30.076 -> [  1620][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.120.252, MASK: 255.255.255.0, GW: 192.168.120.235
```

SuGlider, 2024-04-11:
@me-no-dev - PTAL. It may be some that you may have fixed within the new Networking Arduino layer.

SuGlider, 2024-04-11:
@lbernstone - Thanks for confirming the issue. Did you test it with 2.0.14 or 3.0.0-Alpha2? Or somthing else...

SuGlider, 2024-04-11:
Thanks for the confirmation @lbernstone!

lbernstone, 2024-04-12:
Original issue is as posted.  A simple begin with ssid containing apostrophe fails with AUTH_EXPIRE.  If a static config is set first, it connects properly.

me-no-dev, 2024-04-17:
I can not confirm this:
```
[  1311][V][NetworkEvents.cpp:121] checkForEvent(): Network Event: 9 - WIFI_READY
[  1386][V][STA.cpp:205] _onStaEvent(): STA Started
[  1391][V][NetworkEvents.cpp:121] checkForEvent(): Network Event: 11 - STA_START
[  1399][D][STA.cpp:131] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
STA Started
[  1542][V][STA.cpp:222] _onStaEvent(): STA Connected: SSID: Ficeto's AP, BSSID: 76:4d:28:c6:7a:fc, Channel: 8, Auth: WPA_WPA2_PSK
[  1553][V][NetworkEvents.cpp:121] checkForEvent(): Network Event: 13 - STA_CONNECTED
[  1561][D][STA.cpp:131] _onStaArduinoEvent(): Arduino STA Event: 13 - STA_CONNECTED
STA Connected
[  2265][V][NetworkInterface.cpp:66] _onIpEvent(): sta Got New IP: 192.168.254.36 MASK: 255.255.255.0 GW: 192.168.254.1
[  2277][V][NetworkEvents.cpp:121] checkForEvent(): Network Event: 16 - STA_GOT_IP
[  2284][D][STA.cpp:131] _onStaArduinoEvent(): Arduino STA Event: 16 - STA_GOT_IP
[  2291][D][STA.cpp:189] _onStaArduinoEvent(): STA IP: 192.168.254.36, MASK: 255.255.255.0, GW: 192.168.254.1
*sta: <UP,Ficeto's AP,CH:8,RSSI:-56,N,WPA_WPA2_PSK> (DHCPC,GARP,IP_MOD,V6_REP)
      ether 30:C6:F7:F4:2F:8C
      inet 192.168.254.36 netmask 255.255.255.0 broadcast 192.168.254.255
      gateway 192.168.254.1 dns 192.168.254.3
```

lbernstone, 2024-04-17:
I'll dig out another AP & a esp32 to test on.  Failed on my Android 11 phone.

me-no-dev, 2024-04-18:
@lbernstone mine is Mikrotik, maybe has to do with the Android. I tried all chips too

chipguyhere, 2024-06-03:
Can you paste the actual output of the WiFiScan example to show the SSID name?

What we'd be looking for is to inspect the actual unicode of the apostrophe, to see if they are different.  i.e. copy and paste it into something that can inspect the actual character in the SSID name.

On an iPhone, if you type "Varun's Network" into the Settings app, the iPhone keyboard is going to substitute a different apostrophe, to give it a nicer looking slant.  It will not be considered the same SSID name anymore.

Have a look at this:  I copied and pasted the name of my iPhone out of the Settings app: "Mike’s iPhone 14", which is different than if I just type "Mike's iPhone 14" with my keyboard into GitHub.  Can you see the difference in the shape of the apostrophe?  Python can -- print(ord(a[4])) gives the character code of the fifth character (the apostrophe in Mike's). 

```
Python 3.9.6 (default, Feb  3 2024, 15:58:27) 
[Clang 15.0.0 (clang-1500.3.9.4)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> a = "Mike’s iPhone 14"
>>> print(ord(a[4]))
8217
>>> a = "Mike's iPhone 14"
>>> print(ord(a[4]))
39
>>> 
```

See how 8217 is different from 39?

39 is what's in your source file.

closed 2025-01-16 by Parsaabasi