espressif/arduino-esp32 #9477: Virus in compilation on Esp32 board

Angelpoot2, 2024-04-10:
### Board

Esp32 Dev Module

### Device Description

Hello, straight to the point. I try to program any code from a blink for the esp32 board, but when compiling it it gives me the message that it is not possible to compile due to a virus or unwanted software.
When compiling on any Arduino board I have no problems, the problem is only on the esp32 board

### Hardware Configuration

.

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

windows 10

### Flash frequency

30

### PSRAM enabled

no

### Upload speed

.

### Description

Hello, straight to the point. I try to program any code from a blink for the esp32 board, but when compiling it it gives me the message that it is not possible to compile due to a virus or unwanted software.
When compiling on any Arduino board I have no problems, the problem is only on the esp32 board

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
Arduino:1.8.18 (Windows 10), Tarjeta:"ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

fork/exec C:\Windows\system32\cmd.exe: No se pudo completar la operación porque el archivo contiene un virus o software potencialmente no deseado.

Error compilando para la tarjeta ESP32 Dev Module.

Este informe podría contener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-04-11:
@Angelpoot2 - There is a similar discussion in https://forum.arduino.cc/t/about-esp32-i-need-help-thanks/1145461/2

I use Win11 with all Virus Protection/Verifciation options enabled + Arduino IDE 1.8.19. 
I have used all the different ESP32 Arduino Core versions and I have never seen such issue.

Angelpoot2, 2024-04-13:
It's actually not a problem with Windows. In general, the problem arises when compiling with the ESP32 board, since if I compile towards some other board without problem it compiles and only marks type writing errors: missing points and commas, an undeclared variable, etc., etc.
I have already run different scans on the computer and it does not have a virus, I have a slight suspicion that a library is causing the problem.

VojtechBartoska, 2024-04-18:
Closing as wontfix.

closed 2024-04-18 by VojtechBartoska

moronya, 2025-05-19:
If you are experiencing this error. Try compiling without saving the file. I don't know why the error comes up once the file is saved on a PC. However, this trick worked for me. If someone has an explanation of why saving the file makes it impossible to compile the file, they could help us.