espressif/arduino-esp32 #9479: Arduino version not compatible with ESP-IDF latest

0x5b62656e5d, 2024-04-10:
### Board

ESP32 Devkit DOIT v1

### Device Description

N/A, only connected to laptop via MicroUSB to USB cable

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

VSCode

### Operating System

macOS 14.4.1

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I can't build or access the configmenu. Trying to do that always give me the same error msg saying the Arduino components can be used between 5.1.0 and 5.1.99 IDF versions, but I'm on ESP-iDF 5.3.0.

### Sketch

```cpp
#include "Arduino.h"

// Also have arduino components included in program
```

### Debug Message

```plain
CMake Error at
  /Users/benjamink/esp/esp-idf/components/arduino/CMakeLists.txt:19
  (message):

    Arduino-esp32 can be used with ESP-IDF versions between 5.1.0 and 5.1.99,
    but a newer version is detected: 5.3.0.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-04-10:
Yes, as expected. Arduino 3.0.0 use as base IDF 5.1.x

0x5b62656e5d, 2024-04-10:
Do I install an older IDF version in vscode then?

lbernstone, 2024-04-11:
From current master:
`ESP-IDF Version   : v5.1.2-185-g3662303f31-dirty`

0x5b62656e5d, 2024-04-11:
I tried installing the 5.1 version and running install.sh
I then went into VSCode and added the 5.1 toolchain via ESP-IDF Configure Extension (not the configmenu one)
However, when I run build, same message still appears.

SuGlider, 2024-04-11:
you must run both, `install.sh` and `export.sh` in order to export the correct IDF version.

VojtechBartoska, 2024-09-12:
Hello, closing as answered.

closed 2024-09-12 by VojtechBartoska