espressif/arduino-esp32 #9480: More Zigbee examples for ESP32-C6

jorisvervuurt, 2024-04-10:
### Related area

Zigbee

### Hardware specification

ESP32-C6

### Is your feature request related to a problem?

There is just two (pretty much hidden - not visible in Arduino IDE examples list) examples regarding Zigbee on the ESP32-C6. I would like some more examples, especially for a temperature, humidity, pressure and CO2 sensor.

### Describe the solution you'd like

There is just two (pretty much hidden - not visible in Arduino IDE examples list) examples regarding Zigbee on the ESP32-C6. I would like some more examples, especially for a temperature, humidity, pressure and CO2 sensor.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2024-04-11:
@P-R-O-C-H-Y  - PTAL, Thanks! That may be a nice new task.

lbernstone, 2024-04-11:
A scanner would be an extremely useful example

lboue, 2024-04-15:
For those looking for the link: https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Zigbee

P-R-O-C-H-Y, 2024-05-16:
Zigbee examples got updated to work with latest esp-zigbee-sdk. 
Also 2 new examples have been added to work together: 
- Temperature sensor (End device)
- Thermostat (Coordinator)

mothergooose, 2024-05-18:
> Zigbee examples got updated to work with latest esp-zigbee-sdk. Also 2 new examples have been added to work together:
> 
>     * Temperature sensor (End device)
> 
>     * Thermostat (Coordinator)
> 
> 
> Everything mentioned is included in the latest release 3.0.0-rc3. I appreciate any feedback or suggestions on what we can improve or add next time :)

Great!
I would consider smart metering and motion detection amazing examples.
Thank you for your work!

Hedda, 2024-05-31:
> Zigbee examples got updated to work with latest esp-zigbee-sdk. Also 2 new examples have been added to work together:

For reference; here are the links:

https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Zigbee

Synchronized from esp-zigbee-sdk examples:

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/

PS: Also see related discussion here:

https://github.com/espressif/esp-zigbee-sdk/issues/228

P-R-O-C-H-Y, 2024-05-31:
> @P-R-O-C-H-Y maybe a good idea to add a few more simple "Zigbee HA" profiles (ZHA profile specs) from the esp-zigbee-sdk?
> 
> https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample

I think there is no need to port more IDF zigbee-sdk examples, anyone can easily port them as you can use the 4 examples I ported as a reference. There is no much to change in the example. It's just about adding `setup()` and `loop()` and handle buttons, leds, sensors in the Arduino way. I think it's better to spent more energy on the wrapper, than porting new examples, which will be probably removed when wrapper is done :) Let me know your opinion.

But also if anyone does port any new example from zigbee-sdk, I will be really happy to see a PR adding it to the repo :)

elgerg, 2024-08-06:
Hi @P-R-O-C-H-Y 
Thanks for the effort so far.

I am really looking forward to a wrapper that makes this easier to use as to be fair I'm so confused on what needs setting where as I'd like to extend the temp sensor example to include luminosity but have no idea how to start/where to add it into the existing cluster.

Do we have any ideas when the wrapper will be available for testing?

Thanks in advance!

Hedda, 2024-08-06:
> think it's better to spent more energy on the wrapper, than porting new examples, which will be probably removed when wrapper is done :) Let me know your opinion.

What is the status of that wrapper and can we follow the development of it anywhere?

P-R-O-C-H-Y, 2024-08-07:
Hi, thank you all for the interest for the wrapper. If you want you can follow my [dev branch in my fork](https://github.com/P-R-O-C-H-Y/arduino-esp32/tree/feature/zigbee-library). Its still in early stages, but if you have any suggestions, feel free to open comment here on open an issue in my fork :). Any help or suggestions are welcome.

I have not done much on Zigbee wrapper for last month, as there was a lot of other higher priority tasks, but I am now starting to do some progress again :).

P-R-O-C-H-Y, 2024-08-09:
@Hedda Update pushed to the branch. You can test the on_off light (ED) + switch(Coordinator). Very simple sketches now :)

elgerg, 2024-08-09:
> @Hedda Update pushed to the branch. You can test the on_off light (ED) + switch(Coordinator). Very simple sketches now :)

Oh my! This looks amazing, super simple easy enough for anyone to use!

Thanks for the effort :)

elgerg, 2024-08-09:
One thing I did notice.. how do you set the Manufacturer and model name?

IE
```
/* Attribute values in ZCL string format
 * The string should be started with the length of its own.
 */
#define MANUFACTURER_NAME \
  "\x0B"                  \
  "ESPRESSIF"
#define MODEL_IDENTIFIER "\x09" CONFIG_IDF_TARGET
```

P-R-O-C-H-Y, 2024-08-09:
> One thing I did notice.. how do you set the Manufacturer and model name?
>
> IE
>
> ```
> /* Attribute values in ZCL string format
>  * The string should be started with the length of its own.
>  */
> #define [...]

I am not at this moment, that's something I still need to implement :)
I have opened a feature request to have a tracker for Zigbee Wrapper #10135 to not have multiple issues opened.

Feel free to add any requests or questions there :)

lboue, 2024-08-16:
> > A scanner would be an extremely useful example
> 
> @lbernstone @lboue Zigbee network scan have been implemented in the dev branch.

Thanks, I'll try it as soon as I can.

estegewr, 2024-09-12:
@P-R-O-C-H-Y: I'm not that familiar with Arduino IDE and have read that search paths for header files and libraries cannot be added that easily. To compile your examples, Jan, I need to include sources from the ZigBee example root directory. How do you reference these. I don't want to adapt the file structure by copy & paste all resources in a Arduino project.

labmaster, 2024-09-15:
same here, i'am not able to get the example compiled.  (

Error, see here:
```
c:\Users\m\Documents\Arduino\libraries\Zigbee\src\ZigbeeCore.cpp: In function 'void esp_zb_task(void*)':
c:\Users\m\Documents\Arduino\libraries\Zigbee\src\ZigbeeCore.cpp:76:3: error: 'esp_zb_stack_main_loop' was not declared in this scope
   76 |   esp_zb_stack_main_loop();
      |   ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\meyer\Documents\Arduino\libraries\Zigbee\src\ZigbeeCore.cpp: In member function 'const char* ZigbeeCore::getDeviceTypeString(esp_zb_ha_standard_devices_t)':
c:\Users\m\Documents\Arduino\libraries\Zigbee\src\ZigbeeCore.cpp:363:14: error: 'ESP_ZB_HA_LIGHT_SENSOR_DEVICE_ID' was not declared in this scope; did you mean 'ESP_ZB_HA_SIMPLE_SENSOR_DEVICE_ID'?
  363 |         case ESP_ZB_HA_LIGHT_SENSOR_DEVICE_ID: return "Light Sensor";
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |              ESP_ZB_HA_SIMPLE_SENSOR_DEVICE_ID

Using library ESP32 at version 3.1.0 in folder: C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\libraries\ESP32 
Using library Zigbee at version 1.0.0 in folder: C:\Users\m\Documents\Arduino\libraries\Zigbee 
```

P-R-O-C-H-Y, 2024-09-16:
> @P-R-O-C-H-Y: Can you please share a few words on how to use your ZigBee examples with Arduino? Do I have to compile and install the library first? I tried to compile the examples but got reference errors to the ep header files.

Hi @estegewr, there is a PR opened for Zigbee wrapper library for Arduino. You can checkout that PR and try the examples which are included. You can take a look to the README of the examples to check what setup is needed. The esp-zigbee-sdk libraries are precompiled and included with the esp32 arduino core.

P-R-O-C-H-Y, 2024-09-16:
> same here, i'am not able to get the example compiled. (
> 
> Error, see here:
> 
> ```
> C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\libraries\Zigbee\src\ZigbeeCore.cpp: In function 'void esp_zb_task(void*)':
> C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\libraries\Zigbee\src\ZigbeeCore.cpp:76:3: error: 'esp_zb_stack_main_loop' was not declared in this scope
>    76 |   esp_zb_stack_main_loop();
>       |   ^~~~~~~~~~~~~~~~~~~~~~
> C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\libraries\Zigbee\src\ZigbeeCore.cpp: In member function 'const char* ZigbeeCore::getDeviceTypeString(esp_zb_ha_standard_devices_t)':
> C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\libraries\Zigbee\src\ZigbeeCore.cpp:363:14: error: 'ESP_ZB_HA_LIGHT_SENSOR_DEVICE_ID' was not declared in this scope; did you mean 'ESP_ZB_HA_SIMPLE_SENSOR_DEVICE_ID'?
>   363 |         case ESP_ZB_HA_LIGHT_SENSOR_DEVICE_ID: return "Light Sensor";
>       |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>       |              ESP_ZB_HA_SIMPLE_SENSOR_DEVICE_ID
> 
> Using library ESP32 at version 3.0.4 in folder: C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\libraries\ESP32 
> Using library Zigbee at version 1.0.0 in folder: C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\libraries\Zigbee 
> exit status 1
> 
> Compilation error: exit status 1
> ```

@labmaster I see you are on the 3.1.0-RC1 and Zigbee sdk libraries were not updated there probably yet to the last version. Thats why it cannot be compiled. I will ask to update the libs also there :)

labmaster, 2024-09-17:
...
> 
> @labmaster I see you are on the 3.1.0-RC1 and Zigbee sdk libraries were not updated there probably yet to the last version. Thats why it cannot be compiled. I will ask to update the libs also there :)
> 
> EDIT: There is PR pending for 3.1 branch with the 1.5.0 Zigbee sdk libraries: #10272

Hmmm, thats what it shows when compiling:
...
riscv32-esp-elf-g++ -c @C:\Users\meyer\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-**release_v5.3-466a392a**\esp32h2/flags/cpp_flags
...

P-R-O-C-H-Y, 2024-09-17:
@labmaster Are you using git install or the release of 3.1.0-rc1 in Board Manager?

labmaster, 2024-09-18:
@P-R-O-C-H-Y 
I tried both with different results but in both cases no working compilation.

P-R-O-C-H-Y, 2024-09-18:
> @P-R-O-C-H-Y 
> 
> I tried both with different results but in both cases no working compilation.
> 
> 

On both cases you have same compilation error? If you use master or 3.0.5 release + the Zigbee PR compilation should be fine.

labmaster, 2024-09-19:
Still getting:

Detecting libraries used...
C:\Users\m\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2405/bin/riscv32-esp-elf-g++ -c @C:\Users\m\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-466a392a\esp32h2/flags/cpp_flags -w -Os -Werror=return-type -w -x c++ -E -CC -DF_CPU=96000000L -DARDUINO=10607 -DARDUINO_ESP32H2_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32H2_DEV" -DARDUINO_VARIANT="esp32h2" -DARDUINO_PARTITION_zigbee -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32h2:UploadSpeed=921600,CDCOnBoot=default,FlashFreq=64,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=none,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=ed" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DZIGBEE_MODE_ED @C:\Users\m\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-466a392a\esp32h2/flags/defines -IC:\Users\m\Documents\Arduino\Zigbee_Color_Dimmable_Light -iprefix C:\Users\m\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-466a392a\esp32h2/include/ @C:\Users\m\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-466a392a\esp32h2/flags/includes -IC:\Users\m\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-466a392a\esp32h2/qio_qspi/include -IC:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\cores\esp32 -IC:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\variants\esp32h2 @C:\Users\m\AppData\Local\Temp\arduino\sketches\9C15DBB43541F3B20B026FA2E289D680/build_opt.h @C:\Users\m\AppData\Local\Temp\arduino\sketches\9C15DBB43541F3B20B026FA2E289D680/file_opts C:\Users\m\AppData\Local\Temp\arduino\sketches\9C15DBB43541F3B20B026FA2E289D680\sketch\Zigbee_Color_Dimmable_Light.ino.cpp -o nul
C:\Users\m\Documents\Arduino\Zigbee_Color_Dimmable_Light\Zigbee_Color_Dimmable_Light.ino:34:10: fatal error: ZigbeeCore.h: No such file or directory
   34 | #include "ZigbeeCore.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
Alternatives for ZigbeeCore.h: []
ResolveLibrary(ZigbeeCore.h)
  -> candidates: []
exit status 1

Compilation error: ZigbeeCore.h: No such file or directory

P-R-O-C-H-Y, 2024-09-19:
This is really weird. How did you added the Zigbee library to the core? (as its still a PR and not merged yet)

labmaster, 2024-09-19:
i downloaded zip of https://github.com/P-R-O-C-H-Y/arduino-esp32/blob/feature/zigbee-library/package.json

Problem seems to be that i'am doing something worng with it.
So what is the right place to put it or to do with it ?
Sketchbook location is
c:\Users\m\Documents\Arduino

labmaster, 2024-09-19:
@P-R-O-C-H-Y 
now is tarted all over, clean everything, reinstalled Arduino ...

I get a new error: 

Compiling library "Zigbee"
"C:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32/tools/riscv32-esp-elf/bin/riscv32-esp-elf-g++" -MMD -c "@C:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\tools\\esp32-arduino-libs\\esp32h2/flags/cpp_flags" -w -Os -Werror=return-type -DF_CPU=96000000L -DARDUINO=10607 -DARDUINO_ESP32H2_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32H2_DEV\"" "-DARDUINO_VARIANT=\"esp32h2\"" -DARDUINO_PARTITION_zigbee "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32h2:UploadSpeed=921600,CDCOnBoot=default,FlashFreq=64,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=none,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=ed\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DZIGBEE_MODE_ED "@C:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\tools\\esp32-arduino-libs\\esp32h2/flags/defines" "-IC:\\Users\\m\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024819-4252-wa2ki4.2hjgh\\Zigbee_On_Off_Light" -iprefix "C:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\tools\\esp32-arduino-libs\\esp32h2/include/" "@C:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\tools\\esp32-arduino-libs\\esp32h2/flags/includes" "-IC:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\tools\\esp32-arduino-libs\\esp32h2/qio_qspi/include" "-IC:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\cores\\esp32" "-IC:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\variants\\esp32h2" "-IC:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\libraries\\Zigbee\\src" "@C:\\Users\\m\\AppData\\Local\\Temp\\arduino\\sketches\\2466DE2D61991CDB9445ABB317207B4D/build_opt.h" "@C:\\Users\\m\\AppData\\Local\\Temp\\arduino\\sketches\\2466DE2D61991CDB9445ABB317207B4D/file_opts" "C:\\Users\\m\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\libraries\\Zigbee\\src\\ep\\ZigbeeThermostat.cpp" -o "C:\\Users\\m\\AppData\\Local\\Temp\\arduino\\sketches\\2466DE2D61991CDB9445ABB317207B4D\\libraries\\Zigbee\\ep\\ZigbeeThermostat.cpp.o"
Using previously compiled file: C:\Users\m\AppData\Local\Temp\arduino\sketches\2466DE2D61991CDB9445ABB317207B4D\libraries\Zigbee\ZigbeeCore.cpp.o
Using previously compiled file: C:\Users\m\AppData\Local\Temp\arduino\sketches\2466DE2D61991CDB9445ABB317207B4D\libraries\Zigbee\ZigbeeHandlers.cpp.o
Using previously compiled file: C:\Users\m\AppData\Local\Temp\arduino\sketches\2466DE2D61991CDB9445ABB317207B4D\libraries\Zigbee\ep\ZigbeeColorDimmableLight.cpp.o
Using previously compiled file: C:\Users\m\AppData\Local\Temp\arduino\sketches\2466DE2D61991CDB9445ABB317207B4D\libraries\Zigbee\ep\ZigbeeLight.cpp.o
Using previously compiled file: C:\Users\m\AppData\Local\Temp\arduino\sketches\2466DE2D61991CDB9445ABB317207B4D\libraries\Zigbee\ZigbeeEP.cpp.o
Using previously compiled file: C:\Users\m\AppData\Local\Temp\arduino\sketches\2466DE2D61991CDB9445ABB317207B4D\libraries\Zigbee\ep\ZigbeeTempSensor.cpp.o
Using previously compiled file: C:\Users\m\AppData\Local\Temp\arduino\sketches\2466DE2D61991CDB9445ABB317207B4D\libraries\Zigbee\ep\ZigbeeColorDimmerSwitch.cpp.o
Using previously compiled file: C:\Users\m\AppData\Local\Temp\arduino\sketches\2466DE2D61991CDB9445ABB317207B4D\libraries\Zigbee\ep\ZigbeeSwitch.cpp.o
C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\libraries\Zigbee\src\ep\ZigbeeThermostat.cpp:151:6: error: no declaration matches 'void ZigbeeThermostat::getSensorSettings()'
  151 | void ZigbeeThermostat::getSensorSettings(){
      |      ^~~~~~~~~~~~~~~~
C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\libraries\Zigbee\src\ep\ZigbeeThermostat.cpp:151:6: note: no functions named 'void ZigbeeThermostat::getSensorSettings()'
In file included from C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\libraries\Zigbee\src\ep\ZigbeeThermostat.cpp:1:
C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\libraries\Zigbee\src\ep\ZigbeeThermostat.h:32:7: note: 'class ZigbeeThermostat' defined here
   32 | class ZigbeeThermostat : public ZigbeeEP {
      |       ^~~~~~~~~~~~~~~~

Using library Zigbee at version 1.0.0 in folder: C:\Users\m\AppData\Local\Arduino15\packages\esp32\hardware\esp32\libraries\Zigbee 
exit status 1

Compilation error: exit status 1

labmaster, 2024-09-19:
for now i found the problem.
In ZigbeeThermostat.h at "class ZigbeeThermostat : public ZigbeeEP"
"void getSensorSettings();" was missing.

So you are right, it seems to be there is work in progress from your site . ;-)

I'a not 100% familiar with git, so i'a not pretty sure what you mean with "you need to select previous commit". You mean i should use an olrder commit ?

P-R-O-C-H-Y, 2024-11-14:
Hi everyone, I am closing this issue as there are now 8 examples for the new and easy to use Zigbee library. The old examples which were just port from esp-zigbee-sdk were removed. All the examples can be found here: https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples

closed 2024-11-14 by P-R-O-C-H-Y

cers-idis, 2025-03-24:
Hello,

I'm looking for some help, I'm trying to find (whith no results) on an example about to set a ESP32-C6 as a Zigbee Router. I just wonder how does it need to be declared. 

Thanks in advance!

P-R-O-C-H-Y, 2025-03-24:
Hi @cers-idis, please ask questions on our Discord channel or in Discussions and I will help you there to not spam this closed issue. Thanks and ping me there :)