espressif/arduino-esp32 #9487: State of the art of the ability to establish an internet connexion over IPv6 with Arduino framework

Gspohu, 2024-04-11:
### Board

esp32-s3-N4

### Device Description

Plain module

### Hardware Configuration

Irrelevant

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Linux Ubuntu 22.04

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello @espressif team,

I am currently working on a project that requires making API calls over IPv6 with an ESP32-S3, using the Arduino framework through PlatformIO. My firmware works flawlessly with IPv4 connections, but I encounter significant issues when attempting to connect via IPv6. Specifically, my device can obtain a local IPv4 address but fails to establish any internet connection through IPv6.

I encountered errors when trying to compile the IPv6 example code ([WiFiIPv6](
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiIPv6/WiFiIPv6.ino)) provided in the Arduino-ESP32 library. In Debug message section are the details of the error messages received during compilation. Moreover, I have no relevant code to show to help resolve the issue, it would be a good start to have an example code that can compiling and established a viable connection over Ipv6.

It appears that certain IPv6-related functions are not recognized, which suggests a potential issue or lack of support within the framework or environment setup.

I have explored various issues and documentation, including a specific issue on GitHub ([#6590](https://github.com/espressif/arduino-esp32/issues/6590)) and the official ESP-IDF guide on LWIP without finding a clear solution or workaround.

I am using PlatformIO on Linux Ubuntu 22.04 with the latest platform pack (version 6.6.0 and 6.2.0) for the Arduino framework. I have extensive experience with ESP32 development, but this issue has left me at a loss.

**Objective**: My goal is to establish a reliable connection to an API over IPv6.

**Questions**:
1. Is there a known issue or limitation regarding IPv6 support in the Arduino framework for ESP32-S3, particularly with the setup I described?
2. Can you provide any guidance or recommendations on how to successfully implement IPv6 connectivity for API calls on the ESP32-S3 using the Arduino framework?
3. Are there any updates or patches planned to address IPv6 support in the very near future? The 3.0.0 seem needed a lot of work before stability.

Thank you in advance for your support and guidance. And a big thank for all the work of the contributor.

### Sketch

```cpp
None
```

### Debug Message

```plain
Processing esp32-s3-N4 (platform: espressif32@6.6.0; board: adafruit_feather_esp32s3; framework: arduino)
----------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/adafruit_feather_esp32s3.html
PLATFORM: Espressif 32 (6.6.0) > Adafruit Feather ESP32-S3 2MB PSRAM
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.21.3
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio/build/esp32-s3-N4/src/main.cpp.o
src/main.cpp:9:1: error: 'NetworkUDP' does not name a type
 NetworkUDP ntpClient;
 ^~~~~~~~~~
src/main.cpp: In function 'void wifiOnConnect()':
src/main.cpp:16:5: error: 'ntpClient' was not declared in this scope
     ntpClient.begin(2390);
     ^~~~~~~~~
src/main.cpp:16:5: note: suggested alternative: 'DNSClient'
     ntpClient.begin(2390);
     ^~~~~~~~~
     DNSClient
src/main.cpp: In function 'void wifiConnectedLoop()':
src/main.cpp:42:3: error: 'ntpClient' was not declared in this scope
   ntpClient.beginPacket(address, 123); //NTP requests are to port 123
   ^~~~~~~~~
src/main.cpp:42:3: note: suggested alternative: 'DNSClient'
   ntpClient.beginPacket(address, 123); //NTP requests are to port 123
   ^~~~~~~~~
   DNSClient
src/main.cpp: In function 'void WiFiEvent(arduino_event_id_t)':
src/main.cpp:82:33: error: 'class WiFiClass' has no member named 'linkLocalIPv6'; did you mean 'localIPv6'?
             Serial.println(WiFi.linkLocalIPv6());
                                 ^~~~~~~~~~~~~
                                 localIPv6
src/main.cpp:86:33: error: 'class WiFiClass' has no member named 'softAPlinkLocalIPv6'; did you mean 'softAPIPv6'?
             Serial.println(WiFi.softAPlinkLocalIPv6());
                                 ^~~~~~~~~~~~~~~~~~~
                                 softAPIPv6
src/main.cpp: In function 'void setup()':
src/main.cpp:107:10: error: 'class WiFiClass' has no member named 'softAPenableIPv6'; did you mean 'softAPenableIpV6'?
     WiFi.softAPenableIPv6();
          ^~~~~~~~~~~~~~~~
          softAPenableIpV6
src/main.cpp:110:10: error: 'class WiFiClass' has no member named 'enableIPv6'; did you mean 'enableIpV6'?
     WiFi.enableIPv6();
          ^~~~~~~~~~
          enableIpV6
*** [.pio/build/esp32-s3-N4/src/main.cpp.o] Error 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-04-11:
for IPv6 you need the latest master. @Jason2866 ptal on PIO

Gspohu, 2024-04-11:
Ok that explain, why I cannot obtain any result.

I will start simple, Do you know if the example code car compile on the latest master.

Gspohu, 2024-04-11:
@me-no-dev Is the 2.0.15 version you released 3 hours ago good for IPv6, or do I really need to switch to the master branch?

me-no-dev, 2024-04-11:
arduino 2.0.x is based on IDF 4.4. Arduino 3.0.x is based on IDF 5.1. You really do need master

Gspohu, 2024-04-11:
Ok I didn't understand that, I need IDF 5.1 for IPv6.

Jason2866, 2024-04-11:
@Gspohu This setting should work as long you don't use the new MCUs
```
[env:esp32]
board = esp-wrover-kit
platform = https://github.com/platformio/platform-espressif32.git
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/nathannau/arduino-esp32.git#your_branch_name
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
```

Gspohu, 2024-04-11:
Thank you @Jason2866 !

I will try that

Jason2866, 2024-04-11:
Btw. we are using core 3.0.0 since half a year with development builds for Tasmota
and it is incredible stable. So imho no reason for waiting to use.

Gspohu, 2024-04-11:
Humm, I tried to compile the example code (tell me if it's compatible with core 3.0.0). And :

```
#include "WiFi.h"
^~~~~~~~
compilation terminated.
```

I will explore more tomorrow, btw thanks a lot

Gspohu, 2024-04-29:
Thank you all for your advice.

I can confirm that after a significant migration effort, IPv6 is fully functional with core version 3.0.0, downloaded from the master.

You can close the issue.

closed 2024-04-29 by Gspohu