espressif/arduino-esp32 #9488: Conexión de ESP32 con PC

FulvioSanchez, 2024-04-11:
### Board

ESP32 devkit v1

### Device Description

-

### Hardware Configuration

-

### Version

v2.0.13

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hola, vengo aquí por un problema de mi ESP, hace un mes hice dos proyectos, uno de encendido de un led y el otro de el encendido de este pero por bluetooth, guardé mi ESP todo este tiempo en la caja y ayer que lo quiero utilizar se quedó con el proyecto del bluetooth.

solo me aparece el led de alimentación, cuando presiono el botón EN no hace nada pero la placa sigue emitiendo el ssid del proyecto de bluetooth.

intenté con varios cables y varias pc pero sigue sin poder conectarse a mi pc, es como que la pc no lo detecta, quisiera que alguien me diera un norte por donde empezar a buscar la solución para esto.

### Sketch

```cpp
#include "BluetoothSerial.h"
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run make menuconfig to and enable it
#endif

#define RedLed 22
#define GreenLed 23

String recv = "";

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  pinMode(RedLed, OUTPUT);
  pinMode(GreenLed, OUTPUT);

  SerialBT.begin("ESP32_LED_Control"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop()
{
  while (SerialBT.available()) // Read until the bluetooth client is sending.
  {
    char string = SerialBT.read();
    recv = recv + string;
    delay(1);
    
  }
  
  if (recv == "A")
  {
    digitalWrite(RedLed, HIGH);
  }
  if (recv == "a")
  {
    digitalWrite(RedLed,LOW);
  }
  if (recv == "B")
  {
    digitalWrite(GreenLed, HIGH);
  }
  if (recv == "b")
  {
    digitalWrite(GreenLed,LOW);
  }
  if (recv == "AB")
  {
    digitalWrite(RedLed, HIGH);
    digitalWrite(GreenLed, HIGH);    
  }
  if (recv == "ab")
  {
    digitalWrite(RedLed, LOW);
    digitalWrite(GreenLed, LOW);   
  }
  recv = ""; // clearing the string.
}
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-04-11:
Por favor, este foro es para problemas con el código en el repositorio, no para problemas con su código. esp32.com es un mejor lugar para preguntas general.
Si este programa anteriormente funcionaba, y ahora no enciende la luz, quizás el led haya fallado. Si no utilizas una resistencia adecuada, puedes sobrecargar el led y quemarlo.
Siempre debes tener un retraso, aunque sea corto (`delay(10);`), en tu función de loop.

FulvioSanchez, 2024-04-11:
gracias, no sabía.

closed 2024-04-11 by FulvioSanchez