espressif/arduino-esp32 #9489: Dev board index doesn't point at latest unstable version

tyeth, 2024-04-11:
### Board

All

### Device Description

-

### Hardware Configuration

-

### Version

other

### IDE Name

arduino-cli

### Operating System

win11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

Can't install 3.0.0-alpha3

### Sketch

```cpp
any
```

### Debug Message

```plain
C:\dev\arduino\Adafruit_Wippersnapper_Arduino [add-MCP9601-i2cScanFix ≡ +0 ~2 -0 | +1 ~2 -0 !]> arduino-cli.exe --additional-urls=https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json core install esp32:esp32@3.0.0-alpha3

Error during install: Platform 'esp32:esp32@3.0.0-alpha3' not found: required version 3.0.0-alpha3 not found for platform esp32:esp32
```

```
C:\dev\arduino\Adafruit_Wippersnapper_Arduino [add-MCP9601-i2cScanFix ≡ +0 ~2 -0 | +1 ~2 -0 !]> arduino-cli.exe --additional-urls=https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json core install esp32:esp32
Platform esp32:esp32@2.0.15 already installed
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brentru, 2024-04-11:
@me-no-dev As you released "[Arduino Release v2.0.15 based on ESP-IDF v4.4.7](https://github.com/espressif/arduino-esp32/releases/tag/2.0.15)", could you take a look at this issue? Thank you in advance.

lucasssvaz, 2024-04-12:
@me-no-dev I think the bot replaced it in https://github.com/espressif/arduino-esp32/commit/544b4a2b209293a26a14ee0b096ae8452db34846

me-no-dev, 2024-04-12:
yup... this is the first time we release lower version after alpha with higher one... so script failed :) we need to rework it in the future. We are about to release RC1 of 3.0.0 so it will be back to normal in no time

VojtechBartoska, 2024-04-12:
same issue as https://github.com/espressif/arduino-esp32/issues/9492. closing as fixed by 3.0.0-RC1 release.

closed 2024-04-12 by VojtechBartoska

tyeth, 2024-04-12:
Thanks all, appreciate the quick release