espressif/arduino-esp32 #9491: Legacy integration in Arduino IDE

PM04290, 2024-04-12:
### Board

ESP32

### Device Description

na

### Hardware Configuration

na

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

na

### PSRAM enabled

yes

### Upload speed

na

### Description

With 2.0.14 installed since few month with preferece : 
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
note: it's same thing with preference : 
https://espressif.github.io/arduino-esp32/package_esp32_index.json
question : is it possible to préserve compatibility with old version (legacy) ?
2.0.14 is not displayed, so it's imposible to uninstall it, and if i want to stay in 2.0.14, the core is not displayed on available board.
[image]

### Sketch

```cpp
na
```

### Debug Message

```plain
na
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

PM04290, 2024-04-12:
All ESP32 have desappeared, and i can't remove 2.0.14 to install 2.0.15
[image]

CemDemirkartal, 2024-04-13:
I need 2.0.14 version too, Is there a manual way to install it?

2.0.15 has bugs, Serial 1 and Serial 2 do not work on nodemcu esp32s v1.1 module
It was working well on 2.0.14 version, now working on 2.0.11 version which available previous version to install.

me-no-dev, 2024-04-13:
seems to me you forgot to load the proper JSON, so you see only the versions that are released by Arduino.cc

CemDemirkartal, 2024-04-13:
With https://espressif.github.io/arduino-esp32/package_esp32_index.json

"2.0.15" , "2.0.11" and "2.0.10" versions are listed

lyusupov, 2024-04-13:
@me-no-dev 
both 'stable' and 'dev' JSON files are truncated:
- "2.0.15" is the only listed in the 'stable' one: https://espressif.github.io/arduino-esp32/package_esp32_index.json
- "3.0.0-rc1" is the only listed in the 'dev' file: https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json

me-no-dev, 2024-04-13:
holly cow... something really went crazy with those jsons. I guess we have work to do for next release

me-no-dev, 2024-04-13:
I have updated both JSONs. Can you please confirm that you see the older versions on the stable json and alpha3 on the dev json

lyusupov, 2024-04-13:
@me-no-dev 

> Can you please confirm that you see the older versions ...

**Your update have positive effect !**

### Before

```
$ arduino-cli core install esp32:esp32@2.0.14
Error during install: Platform 'esp32:esp32@2.0.14' not found: required version 2.0.14 not found for platform esp32:esp32
```

### After

```
$ arduino-cli core update-index ;
Downloading index: package_index.tar.bz2 downloaded
Downloading index: package_esp32_dev_index.json downloaded
Downloading index: package_esp32_index.json downloaded
$ arduino-cli core install esp32:esp32@2.0.14
Tool esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0 already installed
Tool esp32:xtensa-esp32s2-elf-gcc@esp-2021r2-patch5-8.4.0 already installed
Tool esp32:xtensa-esp32s3-elf-gcc@esp-2021r2-patch5-8.4.0 already installed
Tool esp32:xtensa-esp-elf-gdb@11.2_20220823 already installed

< ... skipped  ... >
```

Thank you so much!

me-no-dev, 2024-04-13:
@lyusupov I have no idea why it failed. Script has worked for years without issues and it worked for 2.0.14, yet we have not changed it a bit in it since.
Dev channel also failed when we started releasing alphas, but nobody noticed until now.

Thanks for your investigation. It really pinpointed the issue. Others saying that they see other versions made no sense, except if they were looking at the Arduino.cc released versions, which were exactly those.

Anyway, we should be good for now and we will fix them for future releases

khryssmar, 2024-04-13:
Had the same issue yesterday where only .11 and .15 shows up and had errors on Serial2. But today, almost all versions showed up and I can now revert to .14 instead of .11 only.

PM04290, 2024-04-14:
@me-no-dev you suggest 2.0.15 and there is a Serial2 bug ? i use it in some projects.... what should i do ? go back to 2.0.14 ?

me-no-dev, 2024-04-14:
you can go to 2.0.14 or just call `Serial2.setPins(rx, tx)`. As far as I understand, the issue is only with default pins

lyusupov, 2024-04-15:
@PM04290 

[image]

closed 2024-04-15 by PM04290

ajfdez, 2025-01-19:
Hello to all.

I am debugging some SPI Displays issues (ili9488 SPI screen) in an Arduino ESP32, and seems like i need Arduino ESP Boards 2.0.14 because it was the last working version for my problem.

But 2.0.14 is not available in the boards manager.

[image]

@me-no-dev, is it possible that what happened and you solved on Apr/24, is happening again?

I already have the [https://espressif.github.io/arduino-esp32/package_esp32_index.json](url) in the Additional Boards URL's

Xylopyrographer, 2025-01-19:
@ajfdez 
 I think you want to pick the **_esp32 by Espressif Systems_** board package, not the **_Arduino ESP32 Boards by Arduino_** package. 

[image]

ajfdez, 2025-01-19:
No, i need the Arduino ESP32 Boards. 

I have readed somewhere that you need to use the Arduino's and not the Espressis's ones if you are using an Arduino ESP32.

Although trying doesn't hurt, I followed your suggestion and I'm still at the same point. The problem I have is quite widespread, and all the solutions I see involve installing Arduino ESP32 Boards 2.0.14."

me-no-dev, 2025-01-19:
> The problem I have is quite widespread

what problem is quite widespread? Our SPI library works just fine on all supported chips with all SPI screens. Maybe the library you are using is not properly written?

ajfdez, 2025-01-19:
The problem is an ili9488 Spi screen with an Arduino ESP32. 
If you google _Arduino ESP32 Ili9488 _ you can see is something that happens often, and many says is something impossible today.

They do not work. Or at least i do not find the way. The only posts you can see with some success are people that installed 2.0.14

Examples:

[](https://github.com/Bodmer/TFT_eSPI/issues/3329#issuecomment-2370475614)
[](https://github.com/Bodmer/TFT_eSPI/issues/3329#issuecomment-2370475614)
[](https://forum.arduino.cc/t/tft-lcd-display-ili9488-esp32-does-not-work/1169968/21)

I think is something related with SPI declarations. Or maybe is the library as you say, i don't know.

The library is TFT_eSPI.h [](https://github.com/Bodmer/TFT_eSPI/blob/master/README.md) and seems like a well written one.