espressif/arduino-esp32 #9492: 3.0.0 alpha disappeared from dev update site

jsb1, 2024-04-12:
### Board

Lolin D32

### Device Description

Standard

### Hardware Configuration

No

### Version

other

### IDE Name

Sloeber

### Operating System

Ubuntu 23.10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

3.0.0 Alpha disappeared from update site. Sloeber deleted it automatically. I had to restore it from backup, disconnect internet and switch of automatic updates in Sloeber. This is not a bug in sloeber!

What it the Reason for publishing Release 2.0.15 on dev channel? 

### Sketch

```cpp
all
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mightChamp, 2024-04-12:
Facing same issue

DennisMart, 2024-04-12:
Facing same issue mee too,  please provide a response as soon as possible

Hainer1, 2024-04-12:
Same issue

me-no-dev, 2024-04-12:
reason is that the old script was not aware that there is a newer version available. We will release 3.0.0-RC1 shortly and all will be OK again. Sorry for the trouble

JAndrassy, 2024-04-12:
for those who lost the good version
[package_esp32_dev_index.json](https://github.com/espressif/arduino-esp32/files/14957913/package_esp32_dev_index.json)

me-no-dev, 2024-04-12:
don't! I am releasing RC1 @JAndrassy do not use that

closed 2024-04-12 by VojtechBartoska

jsb1, 2024-04-14:
:+1: Thank You very much for quick reaction. And rc1 is looking good - no problems switching from alpha. :+1: