espressif/arduino-esp32 #9495: v3.0.0-RC1 possible regression - WiFi not declared in this scope (was in 3.0.0-alpha3)

tyeth, 2024-04-12:
### Board

adafruit_qtpy_esp32 / ALL ESP32 boards

### Device Description

Building in CI, or locally for adafruit_qtpy_esp32

### Hardware Configuration

-

### Version

latest development Release Candidate (RC-X)

### IDE Name

vscode

### Operating System

win11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

Code that previously compiled and ran in alpha3 doesn't in RC1.
Adding `#include "WiFi.h"` resolves the issue, is that a regression? We were including Wifisecureclient.h and getting wifi for free prior to RC1.

### Sketch

```cpp
/*!
 * @file Wippersnapper_ESP32.h
 *
 * This is a driver for using the ESP32's network interface
 * with Adafruit IO Wippersnapper.
 *
 * Adafruit invests time and resources providing this open source code,
 * please support Adafruit and open-source hardware by purchasing
 * products from Adafruit!
 *
 * Copyright (c) Brent Rubell 2020-2021 for Adafruit Industries.
 *
 * MIT license, all text here must be included in any redistribution.
 *
 */

#ifndef Wippersnapper_ESP32_H
#define Wippersnapper_ESP32_H

#ifdef ARDUINO_ARCH_ESP32
#include "Wippersnapper.h"

#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
#include "Arduino.h"
#include "WiFi.h"  // NOW REQUIRED FOR 3.0.0-RC1
#include <WiFiClientSecure.h>
extern Wippersnapper WS;

/****************************************************************************/
/*!
    @brief  Class for using the ESP32 network interface.
*/
/****************************************************************************/
class Wippersnapper_ESP32 : public Wippersnapper {

public:
  /**************************************************************************/
  /*!
  @brief  Initializes the Adafruit IO class for ESP32 devices.
  */
  /**************************************************************************/
  Wippersnapper_ESP32() : Wippersnapper() {
    _ssid = 0;
    _pass = 0;
    _mqtt_client = new WiFiClientSecure;
  }

  /**************************************************************************/
  /*!
  @brief  Destructor for the Adafruit IO AirLift class.
  */
  /**************************************************************************/
  ~Wippersnapper_ESP32() {
    if (_mqtt_client)
      delete _mqtt_client;
  }

  /********************************************************/
  /*!
  @brief  Sets the WiFi client's ssid and password.
  @param  ssid
            WiFi network's SSID.
  @param  ssidPassword
            WiFi network's password.
  */
  /********************************************************/
  void set_ssid_pass(const char *ssid, const char *ssidPassword) {
    _ssid = ssid;

    // set the AP password
    // check if ssidPassword was "" in secrets.json
    if ((ssidPassword != NULL) && (strlen(ssidPassword) == 0)) {
      _pass = NULL; // Set as NULL for open networks
    } else {
      _pass = ssidPassword;
    }
  }

  /**********************************************************/
  /*!
  @brief  Sets the WiFi client's ssid and password.
  */
  /**********************************************************/
  void set_ssid_pass() {
    _ssid = WS._config.network.ssid;
    _pass = WS._config.network.pass;
  }

  /***********************************************************/
  /*!
  @brief   Performs a scan of local WiFi networks.
  @returns True if `_network_ssid` is found, False otherwise.
  */
  /***********************************************************/
  bool check_valid_ssid() {
    // Set WiFi to station mode and disconnect from an AP if it was previously
    // connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    // Perform a network scan
    int n = WiFi.scanNetworks();
    if (n == 0) {
      WS_DEBUG_PRINTLN("ERROR: No WiFi networks found!");
      return false;
    }

    // Was the network within secrets.json found?
    for (int i = 0; i < n; ++i) {
      if (strcmp(_ssid, WiFi.SSID(i).c_str()) == 0)
        return true;
    }

    // User-set network not found, print scan results to serial console
    WS_DEBUG_PRINTLN("ERROR: Your requested WiFi network was not found!");
    WS_DEBUG_PRINTLN("WipperSnapper found these WiFi networks: ");
    for (int i = 0; i < n; ++i) {
      WS_DEBUG_PRINT(WiFi.SSID(i));
      WS_DEBUG_PRINT(" ");
      WS_DEBUG_PRINT(WiFi.RSSI(i));
      WS_DEBUG_PRINTLN("dB");
    }

    return false;
  }

  /********************************************************/
  /*!
  @brief  Sets the ESP32's unique client identifier
  @note   On ESP32, the UID is the MAC address.
  */
  /********************************************************/
  void getMacAddr() {
    uint8_t mac[6] = {0};
    WiFi.macAddress(mac);
    memcpy(WS._macAddr, mac, sizeof(mac));
  }

  /********************************************************/
  /*!
  @brief  Initializes the MQTT client
  @param  clientID
          MQTT client identifier
  */
  /********************************************************/
  void setupMQTTClient(const char *clientID) {
    if (strcmp(WS._config.aio_url, "io.adafruit.com") == 0) {
      _mqtt_client->setCACert(_aio_root_ca_prod);
    } else {
      _mqtt_client->setCACert(_aio_root_ca_staging);
    }

    // Construct MQTT client
    WS._mqtt = new Adafruit_MQTT_Client(_mqtt_client, WS._config.aio_url, 8883,
                                        clientID, WS._config.aio_user,
                                        WS._config.aio_key);
  }

  /********************************************************/
  /*!
  @brief  Returns the network status of an ESP32 module.
  @return ws_status_t
  */
  /********************************************************/
  ws_status_t networkStatus() {
    switch (WiFi.status()) {
    case WL_CONNECTED:
      return WS_NET_CONNECTED;
    case WL_CONNECT_FAILED:
      return WS_NET_CONNECT_FAILED;
    case WL_IDLE_STATUS:
      return WS_IDLE;
    default:
      return WS_NET_DISCONNECTED;
    }
  }

  /*******************************************************************/
  /*!
  @brief  Returns the type of network connection used by Wippersnapper
  @return ESP32
  */
  /*******************************************************************/
  const char *connectionType() { return "ESP32"; }

protected:
  const char *_ssid;              ///< WiFi SSID
  const char *_pass;              ///< WiFi password
  WiFiClientSecure *_mqtt_client; ///< Pointer to a WiFi client object (TLS/SSL)

  const char *_aio_root_ca_staging =
      "-----BEGIN CERTIFICATE-----\n"
      "TRUNCATED-uoXvg==\n"
      "-----END CERTIFICATE-----\n"; ///< Root certificate for io.adafruit.us

  const char *_aio_root_ca_prod =
      "-----BEGIN CERTIFICATE-----\n"
      "TRUNCATED-8Y=\n"
      "-----END CERTIFICATE-----\n"; ///< Root certificate for io.adafruit.com

  /**************************************************************************/
  /*!
  @brief  Establishes a connection with the wireless network.
  */
  /**************************************************************************/
  void _connect() {

    if (WiFi.status() == WL_CONNECTED)
      return;

    if (strlen(_ssid) == 0) {
      _status = WS_SSID_INVALID;
    } else {
      _disconnect();
      delay(100);
      WiFi.begin(_ssid, _pass);
      _status = WS_NET_DISCONNECTED;
      delay(5000);
    }
  }

  /**************************************************************************/
  /*!
      @brief  Disconnects from the wireless network.
  */
  /**************************************************************************/
  void _disconnect() {
    WiFi.disconnect();
    delay(500);
  }
};

#endif // ARDUINO_ARCH_ESP32_H
#endif // Wippersnapper_ESP32_H
```

### Debug Message

```plain
error: 'WiFi' was not declared in this scope

src/network_interfaces/Wippersnapper_ESP32.h: In member function 'virtual ws_status_t Wippersnapper_ESP32::networkStatus()':
src/network_interfaces/Wippersnapper_ESP32.h:167:13: error: 'WiFi' was not declared in this scope
  167 |     switch (WiFi.status()) {
      |             ^~~~
src/network_interfaces/Wippersnapper_ESP32.h:168:10: error: 'WL_CONNECTED' was not declared in this scope; did you mean 'WS_CONNECTED'?
  168 |     case WL_CONNECTED:
 ```
```

### Other Steps to Reproduce

https://esp32.com/viewtopic.php?f=19&t=39320&p=130587#p130587

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tyeth, 2024-04-12:
@valeros is this a regression, or just a newer way to use it? Are there any docs that mention the change?

me-no-dev, 2024-04-12:
There will be docs. Reason is that WiFiClientSecure can now work on chips without WiFi, so it is not included by default. The whole network interface layer has changed to support H2 (and P4) and to also equalize all network interfaces (with PPP also on the way)

tyeth, 2024-04-12:
Woops, pinged wrong person, sorry.
@VojtechBartoska is this expected?

tyeth, 2024-04-12:
Oh thank you for the reply @me-no-dev 
Much appreciated!