espressif/arduino-esp32 #9500: Serial2 wont work.

khryssmar, 2024-04-13:
### Board

ESP32-WROOM-32D

### Device Description

Breadboard.

### Hardware Configuration

IO17 and IO16 to GPS.

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am using Serial2 to connect to UBX Neo6m. After 2.0.15, it doesnt work anymore.

### Sketch

```cpp
Serial passthrough
```

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-04-13:
@khryssmar - please provide more information and a simple sketch that can be used to reduce the issue.

Thanks in advance.

SuGlider, 2024-04-13:
A quick test reports that Serial2 is using pins 4 and 25 by default... I'll investigate it.

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    37][V][esp32-hal-uart.c:330] uartBegin(): UART2 baud(115200) Mode(800001c) rxPin(4) txPin(25)
[    46][V][esp32-hal-uart.c:416] uartBegin(): UART2 not installed. Starting installation
[    56][V][esp32-hal-uart.c:463] uartBegin(): UART2 initialization done.

```

SuGlider, 2024-04-13:
Yes... it was the PR https://github.com/espressif/arduino-esp32/pull/8800 from 3.0.0 that has been imported into 2.015 by https://github.com/espressif/arduino-esp32/pull/9176

This breaks the UART pins...
|          |   Name  | ESP32 | S2 | S3 | C3 |
|:--------:|:-------:|:-----:|:--:|:--:|:--:|
| UART0 RX | SOC_RX0 |   3   | 44 | 44 | 20 |
| UART0 TX | SOC_TX0 |   1   | 43 | 43 | 21 |
| UART1 RX |   RX1   |   ~9~ -> 26  | ~18~ -> 4 | 15 | 18 | 
| UART1 TX |   TX1   |   ~10~ -> 27  | ~17~ -> 5 | 16 | 19 |
| UART2 RX |   RX2   |  ~16~ -> 4   | -- | 19 | -- |
| UART2 TX |   TX2   |   ~17~ -> 25  | -- | 20 | -- |

SuGlider, 2024-04-13:
@khryssmar - a quick workaround would to initialize `Serial2` passing the pin, for example:
`Serial2.being(115200, SERIAL_8N1, 16, 17); // baud rate, mode, RxPin, TxPin`

lbernstone, 2024-04-13:
The Serial2 pins were changed because psram is typically on 16 & 17. With a WROOM module, they are available.

SuGlider, 2024-04-13:
It was changed for 3.0.0 as a breaking change. 
But for 2.0.x it should reamain the same, not breaking previous sketches.

closed 2024-04-15 by VojtechBartoska

sblendorio, 2024-04-21:
I'm experiencing the same problem, as I installed 2.0.15 version. How can I take the 2.0.16? The IDE shows 2.0.15 as latest version:

[image]

VojtechBartoska, 2024-04-23:
Hello,

2.0.16 is not released yet, we are still working on it. We will release it in 1 - 2 weeks.

sblendorio, 2024-04-23:
@VojtechBartoska thanks.

I read that 3.0 will have a breaking change. Where do I find the corresponding code for using Serial2?

Thanks again!

khryssmar, 2024-08-12:
I would like to reopen the issue. Encountered this again. Please fix. 
 
[image]
[image]

SuGlider, 2024-08-13:
> @VojtechBartoska thanks.
> 
> I read that 3.0 will have a breaking change. Where do I find the corresponding code for using Serial2?
> 
> Thanks again!

https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#id14

SuGlider, 2024-08-13:
> I would like to reopen the issue. Encountered this again. Please fix. @SuGlider
> 
> ![image](https://private-user-images.githubusercontent.com/50015320/357098328-a4fd750a-bd0b-4ac7-96e3-76b3898879ca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1NzgxMDEsIm5iZiI6MTcyMzU3NzgwMSwicGF0aCI6Ii81MDAxNTMyMC8zNTcwOTgzMjgtYTRmZDc1MGEtYmQwYi00YWM3LTk2ZTMtNzZiMzg5ODg3OWNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEzVDE5MzY0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ3ZDhlNWM0NTU2NmU5MjU0NDhmZGZlOTc0NzMyOGZjMDEwYTZhNzI0OWUwZDYzOGUxNTAzMWFiMGRkNTM0NTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.B1cDDxHokIBGIKx10DFXFZNWtmX3-s_MuQohM4SCGTM) ![image](https://private-user-images.githubusercontent.com/50015320/357098437-2c527486-a0fc-432f-bb06-7bcf79e4f47e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1NzgxMDEsIm5iZiI6MTcyMzU3NzgwMSwicGF0aCI6Ii81MDAxNTMyMC8zNTcwOTg0MzctMmM1Mjc0ODYtYTBmYy00MzJmLWJiMDYtN2JjZjc5ZTRmNDdlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEzVDE5MzY0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAzMjgyMDAxY2I3MTQ5MzMxZWMzYzQyYTEwZGNhNWQ3YjQ1YzEzNTNjZjBiY2Q4YzY0ZWFhYTJiOGJjMmVmOTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.GzZUBJ_hWOCxUCLl-y2_n74dEa_pxhDvQfApa4OWg-c)

This change is valid for Arduino Core 3.x.x
Explanation in https://github.com/espressif/arduino-esp32/issues/10152#issuecomment-2286991672
------

Arduino Core 2.0.15, 2.0.16 and 2.0.17 had a mistake using new pins.
Arduino Core 2.0.14 and lower use the original pins.

Arduino Core from `release/v2.x` branch uses the same pins as before:
Fixed with https://github.com/espressif/arduino-esp32/pull/9502