espressif/arduino-esp32 #9506: esp32 arduino release "2.0.15" does not show up in boards manager both in 1.8.x and 2.x

aalucia, 2024-04-14:
### Board

esp32

### Device Description

is a generic problem in arduino ide

### Hardware Configuration

is a generic problem in arduino ide

### Version

v2.0.14

### IDE Name

2.3.2 / 1.8.12

### Operating System

Linux Suse 15

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

When updating the sw the new version does not show up

### Sketch

```cpp
is not related to sketch
```

### Debug Message

```plain
no debug
```

### Other Steps to Reproduce

tried on two arduino's IDE

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

aalucia, 2024-04-14:
Hope it helps :)

JAndrassy, 2024-04-14:
do you have this URL in Preferences: `https://espressif.github.io/arduino-esp32/package_esp32_index.json` ?

aalucia, 2024-04-14:
Yes, absolutely.

aalucia, 2024-04-14:
My bad: a typo in the url. Sorry  and thanks !

aalucia, 2024-04-14:
My bad. A Type in URL. Sorry and thanks.

closed 2024-04-14 by aalucia