espressif/arduino-esp32 #9509: WiFi.macAddress doesn't work anymore to retrieve mac, it returns 00:00:00:00:00:00 - Network.macAddress does perform as expected.

tyeth, 2024-04-14:
### Board

ESP32

### Device Description

-

### Hardware Configuration

-

### Version

latest master (checkout manually)

### IDE Name

-

### Operating System

-

### Flash frequency

-

### PSRAM enabled

yes

### Upload speed

-

### Description

-

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tyeth, 2024-04-14:
@me-no-dev think this is not expected

JAndrassy, 2024-04-14:
I can't reproduce it. It returns the MAC address in my test

tyeth, 2024-04-14:
Oh strange, thanks, I'll investigate further

tyeth, 2024-04-14:
Strictly speaking I'm on this branch with one commit to make the actions on the main adafruit repo pass (or at least my local using platformio):
https://github.com/tyeth/Adafruit_Wippersnapper_Arduino/tree/add-VL53L4CD

me-no-dev, 2024-04-14:
since the network refactoring, macAdress comes from the interface and you are asking for it before you have started that interface. Network.macAddress gives you the base MAC of the ESP chip. From it are derived STA, AP and ETH MACs

tyeth, 2024-04-14:
Thank you so much for the explanation! 

So effectively it's not set on either of those three interfaces until they connect or are brought up, does that sound right?
 Then presumably just scanning with WiFi for available networks would not be sufficient to populate either of the WiFi interface (or specifically STA) Mac address properties?

me-no-dev, 2024-04-14:
WiFi Scan will enable STA, so it should give you the MAC after it runs. You can get the MAC right after the proper `WiFi.mode()` also. No need to connect to anything. 

By far the best way is to use the new network interfaces:
```cpp
// Get the base MAC of the ESP chip (works always)
Network.macAddress(mac);

// Get the MAC address of WiFi STA (must first start interface)
WiFi.STA.begin();
WiFi.STA.macAddress(mac);

// Get the MAC address of WiFi AP (must first start interface)
WiFi.AP.begin();
WiFi.AP.macAddress(mac);

// Get the MAC address of ETH (must first start interface)
ETH.begin();
ETH.macAddress(mac);
```

JAndrassy, 2024-04-15:
off topic:  it would be nice if the mac in the examples above could be of MacAddress type

me-no-dev, 2024-04-15:
@JAndrassy yup. on the way also :)

craigt543nz, 2024-06-22:
Thanks everyone. Very useful info.

closed 2025-01-16 by Parsaabasi