espressif/arduino-esp32 #9510: S3 Mini - After upgrade from 2.0.15 to 3.0.0rc1 "Compilation error: 'ledcSetup' was not declared in this scope"

Mark-Roly, 2024-04-14:
### Board

LOLIN S3 Mini

### Device Description

ESP32-S3 based LOLIN WEMOS S3 Mini

### Hardware Configuration

None that affect the issue

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 11

### Flash frequency

Flash Mode: QIO 80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

After upgrading 2.0.15 to 3.0.0rc1 then use of ledcSetup is failing to compile for ESP32-S3-based LOLIN M3 Mini with the following error:

C:\Users\redacted\redacted\redacted\redacted.ino:1434:3: error: 'ledcSetup' was not declared in this scope
 1434 |   ledcSetup(channel, freq, resolution);
      |   ^~~~~~~~~
C:\Users\redacted\redacted\redacted\redacted.ino:1435:3: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
 1435 |   ledcAttachPin(used_pin, channel);
      |   ^~~~~~~~~~~~~
      |   ledcAttach

### Sketch

```cpp
#define used_pin 14

int freq = 2000;
int channel = 0;
int resolution = 8;

void setup() {
  pinMode(used_pin, OUTPUT);

  ledcSetup(channel, freq, resolution);
  ledcAttachPin(used_pin, channel);

}

void loop() {
}
```

### Debug Message

```plain
C:\Users\redacted\AppData\Local\Temp\.arduinoIDE-unsaved2024315-33988-4c0j6b.gjicv\sketch_apr15a\sketch_apr15a.ino: In function 'void setup()':
C:\Users\redacted\AppData\Local\Temp\.arduinoIDE-unsaved2024315-33988-4c0j6b.gjicv\sketch_apr15a\sketch_apr15a.ino:10:3: error: 'ledcSetup' was not declared in this scope
   10 |   ledcSetup(channel, freq, resolution);
      |   ^~~~~~~~~
C:\Users\redacted\AppData\Local\Temp\.arduinoIDE-unsaved2024315-33988-4c0j6b.gjicv\sketch_apr15a\sketch_apr15a.ino:11:3: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
   11 |   ledcAttachPin(used_pin, channel);
      |   ^~~~~~~~~~~~~
      |   ledcAttach

exit status 1

Compilation error: 'ledcSetup' was not declared in this scope
```

### Other Steps to Reproduce

Be on 3.0.0rc1
use S3 Mini board
Try to compile above sketch

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-04-15:
This is a major version change with breaking API changes.
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#ledc

Mark-Roly, 2024-04-18:
Thank you, I have updated my code and it is working as exxpected.

closed 2024-04-18 by Mark-Roly