espressif/arduino-esp32 #9514: Getting core dump in with 2.0.15 ESP32-D0WD-V3

eloebo, 2024-04-15:
### Board

ESP32-D0WD-V3 audiokit

### Device Description

Audiokit 2.2

### Hardware Configuration

Default Audiokit

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

After new ESP board package 2.0.15 I got this problem. Since the default arduino behaviour is just to reboot and not show crashdump I cannot see in detail what went wrong.

### Sketch

```cpp
I use a precompiled library compiled with esp-idf which I include. (467kB)
```

### Debug Message

```plain
E (331) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184 //331 _data_end
load:0x40078000,len:13260 //446 _loader_text_start
load:0x40080400,len:3028 //233 _init
entry 0x400805e4	 //273 call_start_cpu0

abort() was called at PC 0x400f69eb on core 0

292            0x40080F80          0x00000005          GLOBAL | FUNC       DEFAULT        0x0002         bootloader_print_banner

Backtrace: 0x40083c21:0x3ffe3970 0x4008a4b1:0x3ffe3990 0x4008f531:0x3ffe39b0 0x400f69eb:0x3ffe3a30 0x400f6a32:0x3ffe3a50 0x400f6993:0x3ffe3a70 0x400d8c4f:0x3ffe3a90 0x400d8c8f:0x3ffe3ad0 0x400dc513:0x3ffe3b10 0x400dc711:0x3ffe3bb0 0x400dcab7:0x3ffe3bf0 0x400ee103:0x3ffe3c20 0x40083335:0x3ffe3c50 0x40079306:0x3ffe3c90 |<-CORRUPTED
```

### Other Steps to Reproduce

Suggestion: Add in the boot code to print out readable crashdump from flash from latest crash like: 
https://github.com/MathieuDeprez/ESP32_CoreDump_Arduino_1.0.6

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-04-15:
we can not help just like that, goig just by the info you gave. No code, you did not decode the backtrace with the exception decoder... all I can say so far is that you probably have corrupted heap

eloebo, 2024-04-16:
Giving up using esp32 for arduino

closed 2024-04-16 by eloebo

me-no-dev, 2024-04-16:
you do not need coredump in the logs above. You need the exception decoder to turn the backtrace into meaningful trace. And while we do not offer that specific thing, we do offer other ways to understand what is going on (JTAG, ExceptionDecorder)

eloebo, 2024-04-16:
> you do not need coredump in the logs above. You need the exception decoder to turn the backtrace into meaningful trace. And while we do not offer that specific thing, we do offer other ways to understand what is going on (JTAG, ExceptionDecorder)

I tried a J-link debugger but could not get connection to the cores either through esp-if tools (openocd and gdb running esp32.cfg with interface/jlink.cfg) or the one in arduino package (even worse, since could not connect to the debugger since some standard config with some ROOVER board)
Is ExeptionDecoder something that relies on HW or is it something in SW that could be added easily?

me-no-dev, 2024-04-16:
https://github.com/me-no-dev/EspExceptionDecoder it's software and runs in the Arduino IDE. Even if your project is not in the IDE, once you compile it, you can start the tool in the ide and point it to your project elf file. The rest will happen after you paste the backtrace

me-no-dev, 2024-04-16:
there are other versions of it online that do not require IDE, but I can not say how well they work