espressif/arduino-esp32 #9516: tflite-micro is breaking complex.h

pschatzmann, 2024-04-15:
### Board

Any

### Device Description

The issue is with all boards: I tried with the ESP32 and a C6

### Hardware Configuration

n/a

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Linux Mint

### Flash frequency

n/a

### PSRAM enabled

no

### Upload speed

n/a

### Description

The new version (4.0.0) contains __espressif__esp-tflite-micro__ which has a complex.h in the src directory:
~/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32/include/espressif__esp-tflite-micro/signal/src/complex.h

This is preventing that the regular __complex.h__ from the c and c++ library can be used!  
So all libraries containing complex.h are broken and will produce compile errors.

### Sketch

```cpp
#include <complex.h>

std::complex<double> mycomplex(10.0, 2.0);

void setup() {}

void loop() {}
```

### Debug Message

```plain
Compile error:

/tmp/.arduinoIDE-unsaved2024315-466177-1nzbc94.mycl/sketch_apr15c/sketch_apr15c.ino:3:6: error: 'complex' in namespace 'std' does not name a template type
    3 | std::complex<double> mycomplex(10.0, 2.0);
      |      ^~~~~~~
/tmp/.arduinoIDE-unsaved2024315-466177-1nzbc94.mycl/sketch_apr15c/sketch_apr15c.ino:2:1: note: 'std::complex' is defined in header '<complex>'; did you forget to '#include <complex>'?
    1 | #include <complex.h>
  +++ |+#include <complex>
    2 | 

exit status 1

Compilation error: 'complex' in namespace 'std' does not name a template type
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pschatzmann, 2024-04-15:
I messed up with the version info: of cause it is 3.0.0-rc1

lucasssvaz, 2024-04-15:
The CPP STD header name is `complex` without the `.h`

This will compile without any issue:
```
#include <complex>

std::complex<double> mycomplex(10.0, 2.0);

void setup() {}

void loop() {}
```

me-no-dev, 2024-04-15:
`#include <complex>` not `<complex.h>`. Would that work?

me-no-dev, 2024-04-15:
If you think that complex.h is broken, maybe file an issue in the toolchain repository: https://github.com/espressif/crosstool-NG/issues

pschatzmann, 2024-04-15:
So it is the toolchain and not arduino-esp32 that decides to include tflite-micro ?

me-no-dev, 2024-04-16:
@pschatzmann we did resolve your original issue. For the others it's too tedious for you to properly report. I am still trying to offer you a way to resolve the problem, for which I only know that it has to do with complex.h.

pschatzmann, 2024-04-16:
I can create an additional simple example (on top of the mentioned libraries that give compile errors) if that helps you to understand the issue. Please let me know!

me-no-dev, 2024-04-16:
Yes it will. thanks!

pschatzmann, 2024-04-16:
In Arduino, create a new tab: test.c with the following lines:
```C
#include <complex.h> 
double complex  z  = 3.2 + 4.1 * I; 
```

This gives the following error:
```
In file included from /tmp/.arduinoIDE-unsaved2024316-5867-1odft3o.2uay/sketch_apr16a/test.c:1:
/home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32/include/espressif__esp-tflite-micro/signal/src/complex.h:23:10: error: expected '=', ',', ';', 'asm' or '__attribute__' before '<' token
   23 | template <typename T>
      |          ^
/tmp/.arduinoIDE-unsaved2024316-5867-1odft3o.2uay/sketch_apr16a/test.c:2:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'z'
    2 | double complex  z  = 3.2 + 4.1 * I;
      |                 ^

exit status 1

Compilation error: expected '=', ',', ';', 'asm' or '__attribute__' before 'z'
```
Deleting the file ~/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f312/esp32/include/espressif__esp-tflite-micro/signal/src/complex.h is correcting the issue! 
So this is the root cause of the issue!

pschatzmann, 2024-04-16:
I [cross posted the issue in tflite-micro](https://github.com/espressif/esp-tflite-micro/issues/80) as well, since it could be resolved easily there

me-no-dev, 2024-04-16:
This would be easily fixable in the header itself. Seems to not have guards for CPP and also does not include the next complex.h. With the following changes your example compiled
[image]

pschatzmann, 2024-04-16:
Just for my better understanding: am I correct in the assumption that it was the decision of the arduino-esp32 project to include tflite-micro into the basic functionality instead of using an Arduino library ?

me-no-dev, 2024-04-16:
differences is that this "version" of tflite is written to support the hardware accelerators in some of our chips, so on S3 you would get better results. I do not see how external Arduino library would help in this case. The header comes from tflite itself.

me-no-dev, 2024-04-16:
Also tflite is precompiled, so you only need to link against it and not compile the whole thing (it's a lot)

pschatzmann, 2025-01-03:
Just a friendly reminder that this issue is still open and has not been fixed in 3.1.
All Arduino libraries which are implemented in C which are using complex.h are still broken:
- example https://github.com/pschatzmann/arduino-codec2

lucasssvaz, 2025-01-21:
Fixed in https://github.com/espressif/esp-tflite-micro/releases/tag/v1.3.3

Fix will be available in next release of Arduino.

closed 2025-01-21 by lucasssvaz