espressif/arduino-esp32 #9530: HTTPUpdate stuck at 9% or 10%.

hitecSmartHome, 2024-04-18:
### Board

ESP32-Wrover

### Device Description

Custom device with ETH_LAN8720

### Hardware Configuration

ET_LAN8720

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

HTTPUpdate stuck at 9% or 10%.

### Sketch

```cpp
HTTPUpdateResult result = Updater.update(secureClient, url.c_str(), "", [](HTTPClient *http) {
    http->addHeader("Accept", "application/octet-stream");
});
```

### Debug Message

```plain
[ 33060][D][HTTPUpdate.cpp:252] handleUpdate():  - current Sketch Size: 2210656
[ 33770][D][HTTPUpdate.cpp:309] handleUpdate(): runUpdate flash...
[ 33771][D][Updater.cpp:133] begin(): OTA Partition: app1
[ 39395][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-28928) SSL - Bad input parameters to function
[ 39397][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
[369997][E][HTTPUpdate.cpp:433] runUpdate(): Update.writeStream failed! (Stream Read 
Timeout)

[369997][E][HTTPUpdate.cpp:359] handleUpdate(): Update failed

[370000][D][HTTPClient.cpp:408] disconnect(): tcp is closed

[370005][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
[VersionHandler] - Stage 1/2 error code: 0
[VersionHandler] - Update error: Stream Read Timeout
```

### Other Steps to Reproduce

Just HTTPUpdate with ssl

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-04-22:
Hello @hitecSmartHome, can you please retest on latest 2.0.15 release? I have tested HTTPUpdate without any issues.
Or if possible, send small sketch where I can reproduce the error. Thanks

hitecSmartHome, 2024-04-22:
Thank you for the response.
Sorry the arduino version number probably not right. Iam using arduino as a component with platformio. Will check the correct version number.

lbernstone, 2024-04-22:
An SSL error after receiving 200k of data makes me think you are having a memory issue, although maybe there is some weird timing out that is passing an empty string to ssl_client.  Add the following header, and then call `MEMCK;` at appropriate test points to watch your memory consumption.
```
const char* fmtMemCk = "\nFree: %d\tMaxAlloc: %d\t PSFree: %d\n\n";
#define MEMCK Serial.printf(fmtMemCk,ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())
```

hitecSmartHome, 2024-04-22:
Iam also using external ram and I set malloc so it will allocate from that ram. Ext ram is working fine and i got plenty from both ext and internal. But will print it in the middle of the update. Meanwhile the app is doing a websocket connection to a server as a client with ssl. I have tried to disable that without success. Also my app suspends every other task if the update starts. It is not a real suspend hovewer. Just a flag which puts a vTaskDelay to every task and returns.

hitecSmartHome, 2024-04-23:
**Here are the details of my environment**

- PLATFORM: Espressif 32 (6.6.0) > Espressif ESP-WROVER-KIT
- HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash <-- THIS IS FALSE. It has 16mb flash.
- DEBUG: Current (ftdi) On-board (ftdi) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)

**PACKAGES:**
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - framework-espidf @ 3.40406.240122 (4.4.6)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

**Sketch**

```
RAM:   [=         ]  12.4% (used 40724 bytes from 327680 bytes)
Flash: [=======   ]  65.1% (used 2047733 bytes from 3145728 bytes)
```

**Partition table**

```csv
# Name,	    Type,	SubType,    Offset,	    Size,	    Flags
nvs,        data,	nvs,	    0x9000,         0x5000,
otadata,    data,	ota,        0xE000,         0x2000,
app0,	    app,	ota_0,	    0x10000,        0x300000,
app1,	    app,	ota_1,	    0x310000,       0x300000,
user,	    data,	spiffs,     0x610000,       0x400000,
system,	    data,	spiffs,     0xA10000,       0x5DC000,
coredump,   data,	coredump,   0xFEC000,       0x10000,
```

**Here is my own memory and system info log during a normal operation**

```json
{
  "firmware": {
    "version": "1.0.6",
    "serialBaudRate": 115200,
    "partitionAddress": 65536,
    "partitionSize": 3145728
  },
  "chip": {
    "features": {
      "Embedded Flash": "No",
      "Embedded PSRAM": "No",
      "Wi-Fi 2.4GHz support": "Yes",
      "IEEE 802.15.4 support": "No",
      "Bluetooth Classic support": "Yes",
      "Bluetooth LE support": "Yes"
    },
    "model": "ESP32",
    "cores": 2,
    "revision": 3,
    "fullRevision": 300,
    "featuresRep": 50,
    "idfVersion": "4.4.6"
  },
  "partition": {
    "nvs": {
      "address": 36864,
      "size": 20480,
      "type": "DATA",
      "subtype": "DATA_NVS",
      "encrypted": false
[... 246 lines cut ...]
        "state": 2,
        "priority": 24,
        "basePriority": 24,
        "stackHighWaterMark": 904,
        "coreId": 1,
        "number": 2,
        "runTimeCounter": 3962123,
        "runTimePercent": 0
      },
      {
        "name": "Tmr Svc",
        "state": 2,
        "priority": 1,
        "basePriority": 1,
        "stackHighWaterMark": 1556,
        "coreId": 0,
        "number": 7,
        "runTimeCounter": 43,
        "runTimePercent": 0
      }
    ],
    "totalRunTime": 21026389
  }
}
```

P-R-O-C-H-Y, 2024-04-23:
Please check the memory consumption as lbernstone commented. Thanks

hitecSmartHome, 2024-04-23:
I'm about to test task stack and memory consumption during the updating. The problem is that these are internal functions and I must tweak the library to be able to print these out while running.

lbernstone, 2024-04-23:
[This](https://github.com/espressif/mbedtls/blob/09bba150d0d822aad2e58d71723f5407da5c21e0/library/ssl_msg.c#L5672-L5674) looks to be the source of the error, so either you are out of memory and the ssl context is not getting populated (very likely) or else the context has fallen out of scope (not seeing the code, you would have to determine if this is even possible).

hitecSmartHome, 2024-04-23:
I could not test it with my real server today but I really think that it was not a memory issue. I have 60-80k internal ram sometimes even 100k at once. AFAIK ssl needs max 40k. How could it fall out of scope? I do spin up a task and delete this task right after the update ends but this is blocking code so the task is not deleted until the function returns.

hitecSmartHome, 2024-04-25:
So it seems it is either some weird timeout issue or ssl issue.
If I host my server on localhost with http only, and the esp connects to it locally, it works fine.

hitecSmartHome, 2024-05-02:
So i placed the macro in the onProgress cb

```cpp
[ 45805][D][Updater.cpp:133] begin(): OTA Partition: app0

Free: 131851    MaxAlloc: 57344  PSFree: 3974075

[VersionHandler] - Stage 1/2 Progress: 0.00 %
Free: 131839    MaxAlloc: 57344  PSFree: 3975587

[VersionHandler] - Stage 1/2 Progress: 0.00 %
Free: 131803    MaxAlloc: 57344  PSFree: 3974059

[VersionHandler] - Stage 1/2 Progress: 0.19 %
Free: 131799    MaxAlloc: 57344  PSFree: 3976995

[VersionHandler] - Stage 1/2 Progress: 0.39 %
Free: 131799    MaxAlloc: 57344  PSFree: 3976995

[VersionHandler] - Stage 1/2 Progress: 0.58 %
Free: 131799    MaxAlloc: 57344  PSFree: 3976995

[VersionHandler] - Stage 1/2 Progress: 0.77 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 0.97 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 1.16 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 1.36 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 1.55 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 1.74 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 1.94 %
Free: 131795    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 2.13 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 2.32 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 2.52 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 2.71 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 2.91 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 3.10 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 3.29 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 3.49 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 3.68 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 3.87 %
Free: 131799    MaxAlloc: 57344  PSFree: 3975771

[VersionHandler] - Stage 1/2 Progress: 4.07 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 4.26 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 4.46 %
Free: 131643    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 4.65 %
Free: 131831    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 4.84 %
Free: 131823    MaxAlloc: 57344  PSFree: 3975851

[VersionHandler] - Stage 1/2 Progress: 5.04 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 5.23 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 5.42 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 5.62 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 5.81 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 6.01 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 6.20 %
Free: 131795    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 6.39 %
Free: 131783    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 6.59 %
Free: 131783    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 6.78 %
Free: 131795    MaxAlloc: 57344  PSFree: 3974323

[VersionHandler] - Stage 1/2 Progress: 6.97 %
Free: 131795    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 7.17 %
Free: 131795    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 7.36 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 7.56 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 7.75 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 7.94 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 8.14 %
Free: 131823    MaxAlloc: 57344  PSFree: 3975851

[VersionHandler] - Stage 1/2 Progress: 8.33 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 8.52 %
Free: 131795    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 8.72 %
Free: 131795    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 8.91 %
Free: 131795    MaxAlloc: 57344  PSFree: 3975787

[VersionHandler] - Stage 1/2 Progress: 9.11 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 9.30 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 9.49 %
Free: 131859    MaxAlloc: 57344  PSFree: 3978843

[VersionHandler] - Stage 1/2 Progress: 9.69 %
Free: 131831    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 9.88 %
Free: 131823    MaxAlloc: 57344  PSFree: 3977315

[VersionHandler] - Stage 1/2 Progress: 10.07 %
Free: 131771    MaxAlloc: 57344  PSFree: 3974243

[VersionHandler] - Stage 1/2 Progress: 10.27 %[ 50356][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-28928) SSL - Bad input parameters to function

[ 50362][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.

```

hitecSmartHome, 2024-05-02:
**MEMCK inside `size_t UpdateClass::writeStream(Stream &data)` in `Updater.cpp`**

```cpp

[ 24093][D][Updater.cpp:133] begin(): OTA Partition: app1

[VersionHandler] - Stage 1/2 Progress: 0.00 %
Free: 131795    MaxAlloc: 55296  PSFree: 3973051

[VersionHandler] - Stage 1/2 Progress: 0.00 %
[VersionHandler] - Stage 1/2 Progress: 0.19 %
Free: 131735    MaxAlloc: 55296  PSFree: 3974443

[VersionHandler] - Stage 1/2 Progress: 0.39 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 0.58 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 0.77 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 0.97 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 1.16 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 1.36 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 1.55 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 1.74 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 1.94 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 2.13 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 2.32 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 2.52 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 2.71 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 2.91 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 3.10 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 3.29 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 3.49 %
Free: 131771    MaxAlloc: 55296  PSFree: 3975971

[VersionHandler] - Stage 1/2 Progress: 3.68 %
Free: 131735    MaxAlloc: 55296  PSFree: 3974443

[VersionHandler] - Stage 1/2 Progress: 3.87 %
Free: 131735    MaxAlloc: 55296  PSFree: 3974443

[... 91 lines cut ...]
Free: 131755    MaxAlloc: 55296  PSFree: 3974443

[VersionHandler] - Stage 1/2 Progress: 10.07 %
Free: 131755    MaxAlloc: 55296  PSFree: 3974443

[VersionHandler] - Stage 1/2 Progress: 10.27 %
Free: 131755    MaxAlloc: 55296  PSFree: 3974443

[ 28917][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-28928) SSL - Bad input parameters to function

[ 28918][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.

Free: 149983    MaxAlloc: 55296  PSFree: 3982431

[161698][E][HTTPUpdate.cpp:433] runUpdate(): Update.writeStream failed! (Stream Read Timeout)

[161698][E][HTTPUpdate.cpp:359] handleUpdate(): Update failed

[161701][D][HTTPClient.cpp:408] disconnect(): tcp is closed

[161706][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.

[VersionHandler] - Stage 1/2 error code: 0
[VersionHandler] - Update error: Stream Read Timeout
```

hitecSmartHome, 2024-05-02:
@lbernstone or @P-R-O-C-H-Y do you guys have any idea?
As you can see i have plenty of free ram. I'm using WiFiClientSecure and setting insecure mode.
Handshake and other negotiations should be fine since the update starts.

IT spits out this error

`[ 28917][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-28928) SSL - Bad input parameters to function`

as soon as it gets to 10.27%. The Updater class does not wait at all.

lbernstone, 2024-05-02:
Do you get this error if you use a minimal sketch to pull the firmware from that URL?

hitecSmartHome, 2024-05-02:
Will create a minimal sketch.

hitecSmartHome, 2024-05-02:
**I have created a minimal sketch with wifi instead of ethernet and the result is the same**

```cpp
#include <Arduino.h>
#include <HTTPClient.h>
#include <HTTPUpdate.h>
#include <Update.h>
#include <WiFiClientSecure.h>

const char* fmtMemCk = "\nFree: %d\tMaxAlloc: %d\t PSFree: %d\n\n";

#define URL "https://testServer/version/downloadSpecific?version=1.0.7"
#define SSID "WIFI_SSID"
#define PASSWORD "WIFI_PASS"
#define MEMCK Serial.printf(fmtMemCk,ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())

WiFiClientSecure secureClient;
HTTPUpdate Updater;

void updateSuccess() {
    printf("Update success\n");
}

void updateStart() {
    printf("Update started\n");
}

void updateProgressReport(float percent) {
    printf("Progress: %.2f %%\n", percent);
    MEMCK;
}

void updateError(int error) {
    printf("%s\n", Updater.getLastErrorString().c_str());
}

void connectToWiFi() {
    WiFi.mode(WIFI_STA);
    WiFi.begin(SSID, PASSWORD);
    uint8_t result = WiFi.waitForConnectResult();
    if (result != WL_CONNECTED) {
        printf("WiFi connection failed: %d\n", result);
        ESP.restart();
        return;
    }
    printf("WiFi connected\n");
}

void setup() {
    Serial.begin(115200);
    connectToWiFi();

    secureClient.setInsecure();
    secureClient.setTimeout(5);

    Updater.rebootOnUpdate(false);
    Updater.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);

    Updater.onStart([]() { updateStart(); });
    Updater.onEnd([]() { updateSuccess(); });
    Updater.onError([](int err) { updateError(err); });
    Updater.onProgress([](int current, int total) {
        updateProgressReport(100.0 * current / total);
    });

    HTTPUpdateResult result = Updater.update(secureClient, URL, "", [](HTTPClient *http) {
        http->addHeader("Authorization", "{\"token\":\"noInitYet\"}");
        http->addHeader("Accept", "application/octet-stream");
    });
    if (result != HTTP_UPDATE_OK) {
[... 214 lines cut ...]
Progress: 9.30 %

Free: 244219    MaxAlloc: 110592         PSFree: 4179319

Progress: 9.49 %

Free: 242299    MaxAlloc: 110592         PSFree: 4179319

Progress: 9.69 %

Free: 247675    MaxAlloc: 110592         PSFree: 4179319

Progress: 9.88 %

Free: 245755    MaxAlloc: 110592         PSFree: 4179319

Progress: 10.07 %

Free: 244219    MaxAlloc: 110592         PSFree: 4179319

[  7486][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-28928) SSL - Bad input parameters to function

[  7487][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.

```

me-no-dev, 2024-05-02:
it is rather very strange that you always get to the about the same progress and then it fails. Could you try with a different firmware file? Can you try to just fake-download the file using the HTTPClient library and see if that passes OK? By fake download I mean to just not write the result to flash, but print the progress

hitecSmartHome, 2024-05-02:
I could try that. Will create an other Virgin sketch and try with plain http.

I can download the firmware with the same URL from my browser and i have also tried with postman and it is working. So the server is not the problem

me-no-dev, 2024-05-02:
I do not suspect the server, but rather something else to be happening. Really strange to always stop at the same place. Some bytes in the firmware cause SSL to break? There is something else received that the lib does not expect? Not sure at this point, but if you can download the firmware to the ESP using just HTTPClient, then we can start digging deeper into the updater, rather than SSL

hitecSmartHome, 2024-05-02:
Here it is with bare HTTP stream read

```cpp
#include <Arduino.h>
#include <HTTPClient.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>

const char* fmtMemCk = "\nFree: %d\tMaxAlloc: %d\t PSFree: %d\n\n";

#define URL "https://testServer/version/downloadSpecific?version=1.0.7"
#define SSID "WIFI_SSID"
#define PASSWORD "WIFI_PASS"
#define MEMCK Serial.printf(fmtMemCk,ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())

HTTPClient https;
WiFiClientSecure secureClient;

void connectToWiFi() {
    WiFi.mode(WIFI_STA);
    WiFi.begin(SSID, PASSWORD);
    uint8_t result = WiFi.waitForConnectResult();
    if (result != WL_CONNECTED) {
        printf("WiFi connection failed: %d\n", result);
        ESP.restart();
        return;
    }
    printf("WiFi connected\n");
}

void restart() {
    vTaskDelay(5000);
    ESP.restart();
}

void setup() {
    Serial.begin(115200);
    connectToWiFi();

    secureClient.setInsecure();
    https.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);
    https.addHeader("Authorization", "{\"token\":\"noInitYet\"}");
    https.addHeader("Accept", "application/octet-stream");

    if (!https.begin(secureClient, URL)) {
        printf("Could not connect to server\n");
        restart();
    }

    int httpCode = https.GET();
    if (httpCode <= 0) {
        printf("Error on HTTP request: %d\n", httpCode);
        restart();
    }

    uint8_t buff[128];
    int len = https.getSize();
    int totalBytesRead = 0;
    WiFiClient* stream = https.getStreamPtr();
    while (https.connected() && (len > 0 || len == -1)) {
        size_t size = stream->available();
        if (size) {
            int bytesRead = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
            totalBytesRead += bytesRead;
            float percent = 100.0 * totalBytesRead / len;
            printf("Progress: %.2f %%\n", percent);
            if (len > 0) { len -= bytesRead; }
        }
    }
    https.end();
    printf("HTTP download Done\n");
}

void loop() {
    vTaskDelay(1);
}
```

**Again, same thing.**

```serial
Progress: 11.40 %
Progress: 11.41 %
Progress: 11.42 %
Progress: 11.43 %
Progress: 11.43 %
Progress: 11.44 %
Progress: 11.44 %
[  6320][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-28928) SSL - Bad input parameters to function

[  6330][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.

[  6337][D][HTTPClient.cpp:408] disconnect(): tcp is closed

HTTP download Done

```

hitecSmartHome, 2024-05-02:
Maybe it is some MBEDTLS misconfiguration in the sdkconfig??

hitecSmartHome, 2024-05-02:
[image]
[image]

Or maybe it is a PSRAM issue?

[image]

hitecSmartHome, 2024-05-02:
Here it is with bare HTTP stream read

```cpp
#include <Arduino.h>
#include <HTTPClient.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>

const char* fmtMemCk = "\nFree: %d\tMaxAlloc: %d\t PSFree: %d\n\n";

#define URL "https://testServer/version/downloadSpecific?version=1.0.7"
#define SSID "WIFI_SSID"
#define PASSWORD "WIFI_PASS"
#define MEMCK Serial.printf(fmtMemCk,ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())

HTTPClient https;
WiFiClientSecure secureClient;

void connectToWiFi() {
    WiFi.mode(WIFI_STA);
    WiFi.begin(SSID, PASSWORD);
    uint8_t result = WiFi.waitForConnectResult();
    if (result != WL_CONNECTED) {
        printf("WiFi connection failed: %d\n", result);
        ESP.restart();
        return;
    }
    printf("WiFi connected\n");
}

void restart() {
    vTaskDelay(5000);
    ESP.restart();
}

void setup() {
    Serial.begin(115200);
    connectToWiFi();

    secureClient.setInsecure();
    https.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);
    https.addHeader("hsh-identity", "{\"hshKey\":\"noInitYet\"}");
    https.addHeader("Accept", "application/octet-stream");

    if (!https.begin(secureClient, URL)) {
        printf("Could not connect to server\n");
        restart();
    }

    int httpCode = https.GET();
    if (httpCode <= 0) {
        printf("Error on HTTP request: %d\n", httpCode);
        restart();
    }

    uint8_t buff[128];
    int payloadSize = https.getSize();
    int bytesLeft = payloadSize;
    int totalBytesRead = 0;
    WiFiClient* stream = https.getStreamPtr();

    printf("Downloading %d bytes\n", bytesLeft);

    while (https.connected() && (bytesLeft > 0 || bytesLeft == -1)) {
        vTaskDelay(1);
        size_t size = stream->available();
        if (size) {
            int bytesRead = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
            totalBytesRead += bytesRead;
            float percent = (100.0 * totalBytesRead) / payloadSize;
            printf("Progress: %.2f %%\r", percent);
            if (bytesLeft > 0) {
                bytesLeft -= bytesRead;
            }
        }
    }
    https.end();
    printf(
        "\nHTTP download Done. Downloaded %d bytes of %d bytes\n",
        totalBytesRead,
        payloadSize);
}

void loop() {
    vTaskDelay(1);
}
```

**Again, same thing.**

```serial
Progress: 11.40 %
Progress: 11.41 %
Progress: 11.42 %
Progress: 11.43 %
Progress: 11.43 %
Progress: 11.44 %
Progress: 11.44 %
[  6320][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-28928) SSL - Bad input parameters to function

[  6330][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.

[  6337][D][HTTPClient.cpp:408] disconnect(): tcp is closed

HTTP download Done

```

A little refactor of the test code.

hitecSmartHome, 2024-05-02:
Okay i think i found the problem

[image]

The Assymetric in/out fragment length was turned on in the menuconfig.
I turned it off and now it completes....

Any clue why is that? I would want to know the reason.

me-no-dev, 2024-05-02:
```
If enabled, this option allows customizing TLS in/out fragment length
in asymmetric way. Please note that enabling this with default values
saves 12KB of dynamic memory per TLS connection.
```

hitecSmartHome, 2024-05-02:
Yeah, i have seen the helper text but it is still not clear to me why could this cause that i was experiencing.

me-no-dev, 2024-05-02:
You could get a better answer in the IDF repo tracker. To me, this does not sound like it should break anything, but maybe nobody stumbled upon this before, or they may know how to fix it

hitecSmartHome, 2024-05-02:
Well, thank you all for the help. Closing this

closed 2024-05-02 by hitecSmartHome