espressif/arduino-esp32 #9532: V2.0.15 Serial,println() works WITHOUT a Serial.begin()

def1149, 2024-04-18:
### Board

Adafruit Feather ESP32-S2

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Serial.print() works WITHOUT a Serial.open()

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  Serial.println("\nThis Prints !!??");
  delay(2000);
}
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-04-19:
@def1149 - `Adafruit Feather ESP32-S2` uses `Native USB CDC` to print using the `Serial` object from Arduino.
Indeed, `Serial.println("\nThis Prints !!??")` works because USB OTG CDC is initialized when the USB CDC object (`Serial`) is created and `size_t USBCDC::write()` doesn't check if `USBCDC::begin()` has been called prior to that.

SuGlider, 2024-04-19:
@me-no-dev - The call to `Serial.begin()` in `main.cpp->void app_main()` started with #5422 followed by #5466 and then finally in #6341

In #5422 there is the commit d425a3fd5f94ace0e8c19d1174e5a54f4d0124c9 that says "USB CDC should also be started in main".
Is it still valid? Do you remember the reason for that?

me-no-dev, 2024-04-20:
@SuGlider sorry, I did not understand this is USB. It would then make sense that it's ON

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

SuGlider, 2025-04-13:
This shall be investigated.

Parsaabasi, 2026-10-14:
Hello, 

Closing this issue because it refers to an older version. Please update to the latest version and check if the issue persists. If you still need help, feel free to reopen this issue or create a new one. Thanks!

closed 2026-10-14 by Parsaabasi