espressif/arduino-esp32 #9533: LittleFS directory wrong date 1970

mikrocoder, 2024-04-19:
### Board

Arduino Nano ESP32

### Device Description

Arduino Nano ESP32 only

### Hardware Configuration

Arduino Nano ESP32 only

### Version

v2.0.13

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 11

### Flash frequency

Default

### PSRAM enabled

yes

### Upload speed

Default

### Description

Hi,

when I experiment with the IDE 2.3.2 and Arduino Nano ESP32 Core 2.0.13 and example `LittleFS_time` I notice that a current date is not taken from the time server.
Compared to the [ESP8266 LittleFS](https://github.com/esp8266/Arduino/tree/master/libraries/LittleFS/src) , file methods are also missing. For example `file.getCreationTime()`.

Is ESP32 LittleFS still being developed further?

```
Now is : 2024-04-19 10:22:37

----create and work with file----
Appending to file: /mydir/hello.txt
Message appended
Appending to file: /newOrder/hello.txt
Message appended

Listing directory: /
  DIR : mydir  LAST WRITE: 1970-01-01 00:59:59
Listing directory: /mydir
  FILE: hello.txt  SIZE: 20  LAST WRITE: 2024-04-19 10:22:37
  DIR : newDir  LAST WRITE: 1970-01-01 00:59:59
```
Thanks.

### Sketch

```cpp
Arduino IDE 2.3.2 and Arduino Nano ESP32 Core 2.0.13 and example `LittleFS_time`
```

### Debug Message

```plain
.
```

### Other Steps to Reproduce

.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

heidnerd, 2024-06-11:
I believe this is also a problem with the Espressif ESP32 SD library. The data structures for the file FAT entry are not in the .h files ,  e.g.  FAT_TIME,  FAT_DATE.

Typically you'd write a simple dateTimeCallback function that would include the current time & date information when ever a directory or file is created.   That's not an option in the Espressif versions - the structure definition's are missing in the .h

See: https://forum.arduino.cc/t/add-date-and-time-to-your-sd-card-files/318901

igrr, 2024-06-11:
> Typically you'd write a simple dateTimeCallback function that would include the current time & date information when ever a directory or file is created.

littlefs and fatfs ports for ESP32 use standard C library or POSIX functions such as `time` and `gettimeofday` to obtain current time. So if you set the current time using `settimeofday`, then these filesystems will use correct time.

heidnerd, 2024-06-13:
Thanks goes to Igrr,  he pointed that out.  I'd missed it.  Easy to skim over the code and miss the includes.

Easy example of using the Unix/Posix time set...

https://github.com/lbernstone/ESP32_settimeofday/blob/master/settimeofday.ino

closed 2025-01-15 by Parsaabasi