espressif/arduino-esp32 #9545: EP32 C3 Supermini only boots when USB is connected?

satbeginner, 2024-04-20:
### Board

ESP32 C3 Supermini

### Device Description

ESP32C3 Dev Module 

### Hardware Configuration

No pins connected, just the build-in LED is used.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows-10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi all, I was used that when I use a Serial.begin(115200) in Setup() and I have USB CDC on Boot enabled I can see any Serial.print messages in the Serial Monitor of the IDE.
And yes, that works.
However, like since one week or so, when I flash the code that still works, but now the ESP32C3 no longer boots when the board is NOT connected to a usb port of my PC.
Even with the IDE not started it will not boot on pressing reset or at power on.
But, as soon as I start the IDE or open the serial port (COM12 in my case) eg. by using Putty on that serial port, at that moment YES, the board boots and will start to output the Serial.print messages.

I was used that a board would run anyway, even with no serial port connected.
If I flash with USB CDC on boot Disabled, it won't run either.

I can make it run only when I remove all Serial.xxxx statements from my code before flashing??

I have the feeling this started about a week ago, because I had code running that I flashed a week ago that was running fine when using an external 5V supply. (Yes, I checked the voltages)

I changed some small stuff in the code and flashed again this week, and now it won't boot anymore, only when connected to either the IDE or Putty using the proper COM-port, or commenting out ALL Serial.xxxx statements?
When I use Putty to see the output of the Serial.xxx statements the ESP32 stops working when I disconnect Putty...

Is this because the ESP32C3 always expects a serial-port active when declared as Serial.begin(115200)??

Is there a way of keeping the Serial.xxxx statements in there but setting the serial port to ignore when not there?

I am stumped, but it does reproduce 100%.

Un saludo,

Leo

### Sketch

```cpp
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(8, OUTPUT);
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(8, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(8, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
  Serial.println("ESP32 is running");
```

### Debug Message

```plain
15:57:39.843 -> ESP32 is running
15:57:41.808 -> ESP32 is running
15:57:43.797 -> ESP32 is running
15:57:45.847 -> ESP32 is running
15:57:47.809 -> ESP32 is running
15:57:49.834 -> ESP32 is running
15:57:51.869 -> ESP32 is running
15:57:53.855 -> ESP32 is running
15:57:55.883 -> ESP32 is running
```

### Other Steps to Reproduce

Flased without any Serial.xxxxx statements it will boot and run.
Also when flashed with the Serial- statements connected to PC, but with Putty active instead of the Arduino IDE, it will boot and run too.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-04-24:
@satbeginner - Thanks for reporting.
It seems to be a bug within `Serial.print()`.

Let me investigate it.

SuGlider, 2024-04-24:
It was fixed in 2.0.15 but not in 3.0.0-RC1... I'll submit the necessary PR.

Gspohu, 2024-04-24:
I confirm this bug on ESP32-S3. @SuGlider Can you provide this PR or commit that fix this issue in 2.0.15 ? Thanks !

satbeginner, 2024-04-30:
Hi all, thanks for your work.
In Arduino IDE the 3.0.0-rc1 just installed, I can confirm not yet fixed in rc1, will wait for Arduino update to rc2.

Thanks again for your work !

Leo

satbeginner, 2024-05-15:
Hi all, today my Arduino IDE upgraded the board library to 3.0.0-rc2 so I could test it on a ESp32 C3: Success!!
It gives the Serial.print output visible in the IDE Serial Monitor, and most important: the ESp32C3 boots and runs perfect, even when powered from a usb power brick.
So no need anymore to have an active USB Serial port attached to the eSp32.

Many thanks,

Leo

ojsheep, 2025-02-03:
Is this problem fixed? Becouse i have problem works with usbcable and when i try to power with powerbrick or from 5v supply (via pins), program not boot

SuGlider, 2025-02-04:
I'll reopen this issue and investigate it.

JohnConnorNPC, 2025-02-11:
Using 3.1.1 Same issue with ESP32 C3 Super Mini

Removed all Serial lines and it will work off a stand alone power supply

ojsheep, 2025-02-11:
Ok Thanks, i saw that is you are using delay after inistilizig
serial.begin. module works but i think not stable.

SuGlider, 2025-02-11:
It seems that the fixes were not propagated to the Arduino Core 3.x
I'll add all necessary PRs in order to fix 3.0.x, current 3.1.x and future 3.2.x

ojsheep, 2025-02-11:
Ok, thanks for the update.

JohnConnorNPC, 2025-02-25:
3.1.3 worked

I just need at the start, as I am using a C3 supermini:
#define LED_BUILTIN 8

So I went back to 3.1.1 and it works too now.

PR #9565 (2024-04-29): feat (hwcdc): ports changes made in 2.0.15
## Description of Change
Ports many changes, fixes and improvements made in 2.0.15:
- correct use of timeout
- avoids problems with CDC ISR not reading data
- fixes problems with transmiting many bytes to USB Host
- changes how USB SOF and CDC connection is detected
- improves flushing TX data, forcing a FIFO policy when overfow is detected

## Tests scenarios
TBD 

## Related links
Fixes #9545
closed 2025-03-12 by Jason2866