espressif/arduino-esp32 #9546: printf format warnings (-Wformat) regarding type definition change from Arduino 2 to Arduino 3 for uint32_t and int32_t

mathieucarbou, 2024-04-20:
### Board

All

### Device Description

All

### Hardware Configuration

N/A

### Version

latest development Release Candidate (RC-X)

### IDE Name

PlatformIO

### Operating System

macOS

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Switching to Arduino 3 leads to several printf warnings regarding type definition changed from int to long.

Example:

```
/Users/mat/Data/Workspace/me/WebSerialLite/examples/demo_ap/demo_ap.ino: In function 'void loop()':
/Users/mat/Data/Workspace/me/WebSerialLite/examples/demo_ap/demo_ap.ino:53:22: warning: format '%u' expects argument of type 'unsigned int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Wformat=]
   53 |     WebSerial.printf("Free heap=[%u]\n", ESP.getFreeHeap());
      |                      ^~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~~~
      |                                                         |
      |                                                         uint32_t {aka long unsigned int}
```

The problem is that printf format is setup by the developer according to the expected type of the argument. If those type change, the format might become incorrect.

In the example above, fixing the format for Arduino 3 would make it wrong for Arduino 2...

What would be the solution for library developer to maintain a compatibility with Arduino 2 & 3 in such case ?

I saw some old issues referencing this same problem but they seemed to be closed or left:
- https://github.com/espressif/arduino-esp32/issues/5086
- https://github.com/espressif/esp-idf/issues/6906

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

igrr, 2024-04-20:
Have replied in the linked issue, but just to keep this one answered: the recommended approach for library developers is to use fixed-width format specifier macros from `<inttypes.h>` when printing `uint32_t` and `int32_t` values: `PRIu32`, `PRId32`, `PRIx32` and so on. This is the most portable approach, these macros should always resolve to the correct specifiers for any given toolchain.

VojtechBartoska, 2024-04-22:
Closing this ticket as asnwered, if needed you can reopen. Thank!

closed 2024-04-22 by VojtechBartoska

mathieucarbou, 2024-04-22:
I confirme that using `PRIu32` where applicable solved the warnings for me.