espressif/arduino-esp32 #9547: ESP32S3 SPI2_HOST could not support 6 device, only 4 work.

uta-guy, 2024-04-20:
### Board

ESP32S3 Wroom 1 N8R2

### Device Description

ESP32S3 supports only 4 SPI device on SPI2 bus while claims supporting 6 device. I new installed Arduino 2.3.2 and installed ESPRESSIF V2.0.11, found the file soc_caps.h indicates it supports "#define SOC_SPI_PERIPH_CS_NUM(i)            3".  However, in ESP-IDF V5.2.1, same file says: "#define SOC_SPI_PERIPH_CS_NUM(i)            (((i)==0)? 2: (((i)==1)? 6: 3))". Copied the latest file to Arduino and upload with issue still exists. Strongly suggest upgrading Arduino-esp32 library as well to handle the 4 SPI issue. THANKS!!

### Hardware Configuration

ESP32S3 N8R2 reading 16x Honey ABP SPI sensors. 

### Version

v2.0.10

### IDE Name

Arduino IDE

### Operating System

WIN11

### Flash frequency

80MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

ESP32S3 supports only 4 SPI device on SPI2 bus while claims supporting 6 device. I new installed Arduino 2.3.2 and installed ESPRESSIF V2.0.11, found the file soc_caps.h indicates it supports "#define SOC_SPI_PERIPH_CS_NUM(i)            3".  However, in ESP-IDF V5.2.1, same file says: "#define SOC_SPI_PERIPH_CS_NUM(i)            (((i)==0)? 2: (((i)==1)? 6: 3))". Copied the latest file to Arduino and upload with issue still exists. Strongly suggest upgrading Arduino-esp32 library as well to handle the 4 SPI issue. THANKS!!

### Sketch

```cpp
SPI interface 16x ABP sensors reading.
```

### Debug Message

```plain
Reporting 0xFF.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

uta-guy, 2024-04-21:
Same issue mentioned here but only handled in ESP-IDF : https://github.com/espressif/esp-idf/issues/8876

lbernstone, 2024-04-21:
The [development version](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide) has the new setting.

uta-guy, 2024-04-21:
Thanks Lbernstone for quick reply! I updated the line you highlighted in arduino-esp32 but compilation has issue, mentions missing files. I had to download from idf-c432c692fa and added all missing files to 2.0.11 folder still complains lots of "not declared in this scope" or suggested alternative ...    It seems I need an entire version of arduino-esp32 to handle the SPI issue, not only modifying the line of "#define SOC_SPI_PERIPH_CS_NUM(i)            (((i)==0)? 2: (((i)==1)? 6: 3))". 

Pls help!

lbernstone, 2024-04-21:
The first link is instructions to install the development version.  The second link shows where the files are coming from.

uta-guy, 2024-04-22:
Added the DEV link to Arduino and downloaded ESP32 3.0.0RC1. Double checked the soc_caps.h with 6 device supported. I connected 3 SPI sensors to SPI2 (HSPI I believe, set SCK 12, MOSI 11, MISO 13) with 3 CS ports.  If I enable all 3, none working or only the 2nd is working, others reporting 0xFF. However, when I test them one by one, each works well. I guess it might be a code issue? I'm using standard SPI code, not HSPI code from ESP.

void setup() {

  Serial.begin(115200);
  SPI.begin();
  SPI.setClockDivider(SPI_CLOCK_DIV4);
  SPI.setDataMode(SPI_MODE1);
  SPI.setBitOrder(MSBFIRST);
  //Serial.println("Start...");
  //Serial.print("MISO:");
  //Serial.println( MISO);
  //Serial.print("SCK:");
  //Serial.println(SCK);
  //Serial.print("MOSI:");
  //Serial.println(MOSI);
}

lbernstone, 2024-04-22:
Perhaps you have another issue.  Are you sure all those devices will work at the settings you have provided?  The ESP32S3 can certainly handle multiple SPI devices. See https://wokwi.com/projects/395883588108483585 for an example

uta-guy, 2024-04-22:
Thanks Lbernstone! You gave me an excellent link for my reference. The wiring is exactly what I need; however, the code is not so straight forward to me. I attached the code  and my concerns, pls help. THX!
I have to say, there are so many ESP32S3 SPI info online but seldom is useful. 

#include <SD.h>

SPIClass SPI2(2); // If I have 3 SPI device, should I create 3 constructors here? 
#define mySPI SPI2

void setup() {
  Serial.begin(115200);
  Serial.println("Hello, ESP32-S3!");

  mySPI.begin(12, 13, 11, 15);    // Here creates 1 SPI device. 
  if (SD.begin(15, mySPI)) {        // Could not understand?? 
    Serial.println("SD0 started");
  }
  if (SD.begin(16)) {       // Not connected in the circuit, why shows SD1 started?
    Serial.println("SD1 started");
  }
  if (SD.begin(14)) {    // Not connected in the circuit, why shows SD2 started?
    Serial.println("SD2 started");
  }
  if (SD.begin(5)) {
    Serial.println("SD3 started");
  }
  if (SD.begin(19)) {
    Serial.println("SD4 started");
  }
}

**Following is a SPI from ESPRESSIF:**

#include <SPI.h>

// Define ALTERNATE_PINS to use non-standard GPIO pins for SPI bus

#ifdef ALTERNATE_PINS
  #define HSPI_MISO   26
  #define HSPI_MOSI   27
  #define HSPI_SCLK   25
  #define HSPI_SS     32
#else
  #define HSPI_MISO   MISO
  #define HSPI_MOSI   MOSI
  #define HSPI_SCLK   SCK
  #define HSPI_SS     SS
#endif

static const int spiClk = 1000000; // 1 MHz

//uninitalised pointers to SPI objects
SPIClass * hspi = NULL;

void setup() {
  //initialise instance of the SPIClass attached to HSPI
  hspi = new SPIClass(HSPI);
  
  //clock miso mosi ss
#ifndef ALTERNATE_PINS
  //initialise hspi with default pins
  //SCLK = 14, MISO = 12, MOSI = 13, SS = 15
  hspi->begin();
#else
  //alternatively route through GPIO pins
  hspi->begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS); //SCLK, MISO, MOSI, SS // Should I create 3 times with different SSs?
#endif

  //set up slave select pins as outputs as the Arduino API
  //doesn't handle automatically pulling SS low
  pinMode(HSPI_SS, OUTPUT); //HSPI SS
}

// the loop function runs over and over again until power down or reset
void loop() {
  hspi_send_command();
  delay(100);
}

lbernstone, 2024-04-22:
This site is not meant to be a tutorial- it is for errors in the code in the repository here.  Please ask general questions at https://esp32.com
SPI is a bus.  You can run multiple devices on a single bus, although only one can actually communicate with the cpu at a time.  They take turns by signalling on the individual cable select line.  The numbers in Arduino don't correspond to the numbers in ESP-IDF, so your original issue is irrelevant here.  On the S3, there are 4 SPI peripherals, but only 2 are available for use as generic SPI (0 & 1/HSPI & VSPI in Arduino).  An error in my code made it unclear what I was demonstrating there.  Take a look at the updated version.  If you only need one bus, use the SPI object which is pre-defined.  It uses the pin numbers set in the [variants file](https://github.com/espressif/arduino-esp32/blob/master/variants/esp32s3/pins_arduino.h#L28-L31) by default.  You can change the pins by calling SPI.begin as shown in the wokwi example. It then calls SD.begin to initialize the SD cards (a generic SPI device) using the cable select pin and SPI object parameters.

VojtechBartoska, 2024-09-12:
Hello @uta-guy, can this ticket be consider as asnwered?

uta-guy, 2024-09-12:
Yes， solved。 Thanks

[External]

closed 2024-09-13 by VojtechBartoska