espressif/arduino-esp32 #9549: ESP32S3 failed to auto upload in v2.0.15

moononournation, 2024-04-21:
### Board

ESP32-S3

### Device Description

ESP32-S3 board with USB CDC

### Hardware Configuration

GPIO 19 & 20 to USB CDC

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS 14.4.1

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

After upgrade to v2.0.15 and uploaded once, it will failed to auto upload anymore. Cross check with downgrade to v2.0.14, auto upload resumed after manual upload once.

### Sketch

```cpp
any sketch can reproduce it.
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2024-04-22:
Hi @moononournation, thanks for the report. You you please provide the exact configuration you are using and the steps to reproduce it ? So I can test and investigate it. Thanks in advance!

Revvy-D, 2024-05-08:
Hi,
I am experiencing this issue too. I am using a Seeed Studio XIAO ESP32S3.
I can confirm that this issue isn't present in version 2.0.14 but is present in version 2.0.15 and 2.0.16.
I can also confirm that the first upload after installing one of the newer version will upload correctly but all subsequent upload will fail.
I would like to add the caveat though that this only happens if you are using serial communications in your sketch. If you are not using serial communications then you will not have this issue.
To get around this issue and upload some new code you can either start the board in bootloader mode or if you add a delay before you call Serial.begin() you can plug in the board just before the upload starts and it works fine.

ErtBo, 2024-05-15:
Experiencing the same issue with ESP32-S3 and ESP32-C3 devkits when USB CDC is used to upload from Arduino IDE Windows 10/11. Below few devkits where the issue arises

1. ESP32-S3-DevKitC-1 (Espressif)
2. ESP32-S3-Zero (Waveshare Electronics)
3. LILYGO T8-C3 (LILYGO)
4. ESP32-C3-WROOM-02 (custom board)
5. ESP32-S3-WROOM-01 (custom board)

lucasssvaz, 2024-05-15:
@SuGlider Is this related to the issue you are working on ?

me-no-dev, 2024-05-15:
@lucasssvaz yes. We have found the problem and are working on getting it resolved (that particular one we have already fixed, but others remain)

lucasssvaz, 2024-05-15:
Fixed in https://github.com/espressif/arduino-esp32/pull/9628 Please take a look if this fixes your issues

SuGlider, 2024-05-15:
The PR #9628 isn't backported to Arduino Core 2.0.x yet.
It will be done by next week, directly into `release/v2.x` branch.

But please feel free to try to apply the same changes to your Arduino Core local installation.

SuGlider, 2024-05-23:
@moononournation - The fix is already merged to `release/v2.x` branch by PR #9660

closed 2024-05-23 by SuGlider