espressif/arduino-esp32 #9550: ESP32 encounter failures when launching.

ICheng-Chou, 2024-04-23:
### Board

u-blox NINA-W10 series (ESP32)

### Device Description

no

### Hardware Configuration

no

### Version

v2.0.14

### IDE Name

Arduino IDE, VS code

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am using a very simple code, but I often encounter failures when launching the program. 
I'm currently using the NINA-W106B module with an ESP32-D0WD-v3 chip.
I've tried burning the same code with VS Code using idf version 4.2.2, and I don't experience this issue. 
I suspect there might be a problem with the Arduino IDE plugin. 
Can anyone assist me?

### Sketch

```cpp
void setup() {
  Serial.begin(115200);
  Serial.println("Setup done.");
}

void loop() {
  delay(500);
  Serial.println("Running...");
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
Setup done.
[... 453 lines cut ...]
1150 mmu set 00030000, pos 00030000
1150 mmu set 00040000, pos 00040000
1150 mmu set 00050000, pos 00050000
1150 mmu set 00060000, pos 00060000
1150 mmu set 00070000, pos 00070000
1150 mmu set 00080000, pos 00080000
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
Setup done.
Running...
Running...
Running...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ICheng-Chou, 2024-04-24:
Thank you for your reply. I have conducted a cross-check to confirm the usability of the cable:

I used the same cable and programming using VS Code without encountering any boot failure.
I changed the cable and still encountered a boot failure when programming with Arduino.
Therefore, I conclude that the issue is not with the cable.

me-no-dev, 2024-04-24:
That is really strange, because the first reboot in your logs is `POWERON_RESET` followed by `SW_RESET` and followed by a lot of `TG0WDT_SYS_RESET` which points to bootloop caused by probably power, from which the chip goes crazy and loops a lot more with fails long before the firmware is even booted. What you use to compile and flash should not matter.

Just in case... can you try with 3.0.0-RC1?

kapyaar, 2024-04-24:
I have seen something along this line. Not sure if it applies to you. With some digging, I came across some references to a bug in the ESP chip. See page 5, bug list item 3.4. IT turned out the version I had was affected. For which I tried disabling the brown out when the board boots up, and after the main tasks are done, re enable it. 

```
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"

setup(){
uint32_t brown_reg_temp = READ_PERI_REG(RTC_CNTL_BROWN_OUT_REG);  //save WatchDog register
WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);                        //disable brownout detector. Enable after wifi is connected.
....setup things......
WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, brown_reg_temp);  //enable brownout detector.
}
  
```

Again, worth a try if you are not making progress.

closed 2024-04-25 by ICheng-Chou