espressif/arduino-esp32 #9551: Why do i get a warning when i got a break on UART?

hitecSmartHome, 2024-04-23:
### Board

ESP32-Wrover

### Device Description

-

### Hardware Configuration

-

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I wrote a custom modbus implementation which uses the HardwareSerial.
I'm exchanging modbus messages over uart with 115200 baud.
If I set the debug level to 5, i got an UART1 RX break warning on serial constantly.

The communication is in a tight loop because I want to exchange messages as fast as i can.
I'm expecting a break, thats why I know that I got a packet. It is not a problem at all.

### Sketch

```cpp
Serial1.begin(MBUS_BAUD, SERIAL_8N1, MBUS_RX, MBUS_TX);
Serial1.setPins(-1, -1, -1, MBUS_RTS);
Serial1.setMode(MODE_RS485_HALF_DUPLEX);
Serial1.setTxBufferSize(RESPONSE_BUFFER_SIZE);
Serial1.setRxBufferSize(RESPONSE_BUFFER_SIZE);
Serial1.setRxTimeout(RX_TIMEOUT);
Serial1.onReceive([this](void) { handleRawPacket(); },true);
Serial1.onReceiveError([this](hardwareSerial_error_t error) { handlePacketError(error); });
```

### Debug Message

```plain
[189919][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189923][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189928][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189935][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189945][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189953][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189961][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189969][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189973][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189983][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189990][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[189997][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[190004][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[190011][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[190015][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[190019][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[190023][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[190027][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[190032][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

[190040][W][HardwareSerial.cpp:313] _uartEventTask(): UART1 RX break.

....
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2024-04-23:
possibly related to https://github.com/espressif/arduino-esp32/issues/9500?

hitecSmartHome, 2024-04-23:
> possibly related to #9500?

I'm on 2.0.14 Arduino as a component of IDF using platformIO.

hitecSmartHome, 2024-04-23:
Really frustrating because my serial is flooded with these warning messages and I can't debug other part of my code.

hitecSmartHome, 2024-04-23:
I'M specifiing pins by default.

```cpp
#define MBUS_BAUD 115200
#define MBUS_RX 35
#define MBUS_TX 32
#define MBUS_RTS 33

// RX_TIMEOUT / (MBUS_BAUD / 10) = millisec
// 4 / (115200 / 10) = 3.4 ms
#define RX_TIMEOUT 5
#define WAIT_TIMEOUT_MS 100
```

SuGlider, 2024-04-24:
In order to get rid of this message, you may change the `HardwareSerial.cpp` code from the Arduino Core folder.
Just comment out the `log_w("UART%d RX break.", uart->_uart_nr);` line

https://github.com/espressif/arduino-esp32/blob/58b9f079d25155304c75f9d5c7c4dbe6b78e48c1/cores/esp32/HardwareSerial.cpp#L255-L258

SuGlider, 2024-04-24:
Another way to get rid of this messages would be to set the log level to Error only.
Not sure if any of those possible workarounds would fit your need.

Please let me know.

SuGlider, 2024-04-24:
From our end, it would make more sense to change these log messages to Verbose instead of Warning Level.

hitecSmartHome, 2024-04-24:
Thank you for the suggestions. Isnt rx break a normal behavior? How else should we determine a packet?

SuGlider, 2024-04-24:
> Thank you for the suggestions. Isnt rx break a normal behavior? How else should we determine a packet?

Yes, but it depends on the protocol. In the case of MODBUS or some RS485 protocols, it may be common.
Some other protocols use a data structure with specific field values to communicate end of transmission, ACK, NACK,  and so on.

hitecSmartHome, 2024-04-25:
Okay, i just commented the line out. Thank you very much for the help!

PR #9563 (2024-04-29): Change log level of UART events
## Description of Change
Changes from Warning to Verbose UART event warnings in order to make debugging easier, specially with UART BREAK.

Affected Events: BREAK, parity error, frame error.

## Tests scenarios
CI Only.

## Related links
Fix #9551
closed 2024-04-25 by hitecSmartHome