espressif/arduino-esp32 #9557: ESP32-S3 Serial1，TXpin：9，Rxpin：10，Unacceptable data

guess-name, 2024-04-23:
### Board

ESP32-S3-WROOM-1U

### Device Description

Connect to RS232 chip, chip model SP3232EY-L /TR

 Serial1：Sending is OK, not receiving data

### Hardware Configuration

NULL

### Version

latest master (checkout manually)

### IDE Name

arduino

### Operating System

windows10

### Flash frequency

240MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

 Serial1：Sending is OK, not receiving data

### Sketch

```cpp
void setup()
{
  Serial0.begin(115200);
  Serial1.begin(115200);
  Serial1.setPins(10, 9);
}

void loop()
{

  Serial1.println(Serial0.available());
  // Serial0.println(digitalRead(10));
  Serial0.println(Serial1.available());
  delay(2000);
}
```

### Debug Message

```plain
No errors were reported
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-04-23:
Does it work if you call setPins before the begin, or set the pins as part of the begin parameters?

SuGlider, 2024-04-24:
@guess-name - Please inform what is the Arduino Core version that you are using.

SuGlider, 2024-04-24:
I have tested it with Arduino Core 2.0.15 and 3.0.0-rc1. No issues found.
In order to teste I have used a external USB-UART board based on the CH-340 chip.

SuGlider, 2024-04-24:
`setpin()` can go before or after `begin()`. It doesn't matter.

This is the sketch I've used to test S3 Serial with the CH340 USB-UART board.

``` cpp
void setup()
{
  // setPins() can go before or after begin(). It works both ways.
  //Serial1.setPins(10, 9);
  Serial1.begin(115200);
  Serial1.setPins(10, 9);
}

void loop()
{
  // Serial1 prints lower case
  Serial1.println("output test");
  // Serial1 prints back read data in upper case
  if (Serial1.available()) {
    while (Serial1.available()) {
      Serial1.print((char) (Serial1.read() - ('a' - 'A')));
    }
    Serial1.println();
  }
  delay(2000);
}
```

guess-name, 2024-04-25:
I changed the library to 3.0 and it worked

guess-name, 2024-04-25:
why？

SuGlider, 2024-04-25:
> why？

What was the previous Core version that you were using?

guess-name, 2024-04-25:
I used PIO before and later changed PIO's platformio.ini file to the following

[env:lilygo-t-display-s3]
; platform = espressif32
board = lilygo-t-display-s3
framework = arduino
board_build.partitions = ESP32-SRT-16M.csv
build_flags = -D BOARD_HAS_PSRAM
board_build.arduino.memory_type = qio_opi
upload_speed = 230400
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
board_build.arduino.upstream_packages = no
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
    platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
	; tool-esptoolpy@1.40700.0
	tool-xtensa-esp-elf-gdb
lib_deps = 
	bblanchon/ArduinoJson@^7.0.2
	circuitstate/CSE_ModbusRTU@^0.0.7
	circuitstate/CSE_ArduinoRS485@^1.0.12

SuGlider, 2024-04-25:
OK. There were a few issues with `setPins()` in some previous Core versions. But it shall be all fine now.
Can we close this issue?

closed 2024-04-25 by SuGlider