espressif/arduino-esp32 #9559: ESP32 won't connect to wifi with 2.x (but works with 1.x)

AndyKoder, 2024-04-23:
### Board

ESP32-Wroom32 (ESP32 Dev Module)

### Device Description

Nothing

### Hardware Configuration

Nothing

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

ubuntu linux

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Can't connect to the WiFi router.

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientConnect/WiFiClientConnect.ino
Exactly this example (changed only ssid+pass)
```

### Debug Message

```plain
it compiles + uploads fine, serial monitor shows:
[WiFi] WiFi Status: 0
[WiFi] WiFi Status: 0
[WiFi] WiFi Status: 0
[WiFi] Failed to connect to WiFi!

same with all 2.x versions
```

### Other Steps to Reproduce

but with 1.0.6 it works, same code, different serial output:

[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi is disconnected
[WiFi] WiFi Status: 0
[WiFi] WiFi is connected!
[WiFi] IP address: 192.168.2.102

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-04-23:
What is your router/AP hardware?
Do you have any unusual characters in your SSID/pass?

lbernstone, 2024-04-24:
Ah, right @SuGlider . https://docs.espressif.com/projects/arduino-esp32/en/latest/troubleshooting.html#id3

AndyKoder, 2024-04-24:
No unusual chars in the SSID/PASS
Router does WPA2 max (set to be only WPA2)

It works now with: WiFi.setMinSecurity(WIFI_AUTH_WPA2_PSK);
(not with WIFI_AUTH_WPA_PSK, as it was set to WPA2 only)
Thank you @SuGlider / @lbernstone very much!

Maybe it should be in the example code too? (commented out)

closed 2024-04-24 by AndyKoder