espressif/arduino-esp32 #9562: Linking completely breaks on 3.0.0(-rc1)

rhysperry111, 2024-04-24:
### Board

UM TInyS3

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino CLI

### Operating System

Linux (rolling release, up-to-date)

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

When building a project when using 3.0.0 (in order to make use of the new ledcAttach function) it seems to fail with linking the majority of functions.

I have attached verbose build logs (and the build command) compiling with 2.0.11 vs 3.0.0-rc1.

`arduino-cli compile -v --export-binaries --libraries=editor/arduino --build-property compiler.c.extra_flags="-Ieditor/arduino/src/lib" --build-property compiler.cpp.extra_flags="-Ieditor/arduino/src/lib" -b esp32:esp32:esp32s3 editor/arduino/examples/Baremetal/Baremetal.ino`

[v3.0.0-rc1.buildlog.txt](https://github.com/espressif/arduino-esp32/files/15094729/v3.0.0-rc1.buildlog.txt)
[v2.0.11.buildlog.txt](https://github.com/espressif/arduino-esp32/files/15094730/v2.0.11.buildlog.txt)

Currently blocking: thiagoralves/OpenPLC_Editor#106

### Sketch

```cpp
- unable to upload, large project, generated by OpenPLC -
```

### Debug Message

```plain
See attached.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-04-24:
@rhysperry111 - It is not fully clear to me what is the issue. Please depict it further.

Arduino Core 3.0.0 has **many breaking changes** when compared to 2.0.x.
Most embedded project codes shall need to be changed in order to use the new 3.0.x Core.
There is some documentation about it:

https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

Regarding `ledcAttach()`, it is in the 3.0.x Core but not n the 2.0.x.
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#ledc

rhysperry111, 2024-04-24:
@SuGlider 
>  It is not fully clear to me what is the issue. Please depict it further.

The project compiles and links perfectly fine in v2.0.x (apart from the ledcAttach bit, but that's expected), however under v3.0.0 it fails to even link things like `micros()` (part of the standard set of Arduino functions)

SuGlider, 2024-04-24:
OK, so the issue happens when using `arduino-cli compile -v --export-binaries --libraries=editor/arduino --build-property compiler.c.extra_flags="-Ieditor/arduino/src/lib" --build-property compiler.cpp.extra_flags="-Ieditor/arduino/src/lib" -b esp32:esp32:esp32s3 editor/arduino/examples/Baremetal/Baremetal.ino` building process.

Reading the logs, I think that it would be good to clear all cache, compiled binary folders, etc and start a new fresh build for Core 3.0.0.

There are a changed in IDF, library folders and others that may be picking up remainings binaries from 2.0.14 and using it to link the 3.0.0 version.

rhysperry111, 2024-04-24:
I've cleared as many caches as I can find, and also `2.0.11` isn't mentioned once in the verbose log so I don't think it's a problem with some old components being left behind.

SuGlider, 2024-04-24:
maybe @me-no-dev may take a look into the logs and quickly see the issue related to using `arduino-cli` to build it...

rhysperry111, 2024-04-24:
I'd provide the source files if I could, but the files OpenPLC tends to generate tend to be a bit of a mess and scattered over a few locations

me-no-dev, 2024-04-24:
You are overwriting some things that we use: https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L66-L67

Just add those to your custom extra flags and it should be fine

rhysperry111, 2024-04-24:
Holy crap you're a lifesaver <3

Thanks for solving the issue, it's genuinely appreciated

Is this worth mentioning in the migration guide since it seems to have changed behavior since the 2.0.x releases?

me-no-dev, 2024-04-24:
Sure thing! @VojtechBartoska please make sure this ends up in the migration guide. cc @P-R-O-C-H-Y @lucasssvaz

VojtechBartoska, 2024-04-29:
Reopening to remind us that documentation update is needed.

SuGlider, 2024-04-29:
> Reopening to remind us that documentation update is needed.

I had open an issue about it... https://github.com/espressif/arduino-esp32/issues/9566

VojtechBartoska, 2024-04-29:
Lets close both issues with new PR with docs update.

PR #9574 (2024-05-02): docs(flags): Add build system breaking changes to migration guide
## Description of Change
Add section in the migration guide for build system and add note about the "Extra Flags" change.

## Related links

Closes #9562 
Closes #9566
closed 2024-05-02 by me-no-dev