espressif/arduino-esp32 #9564: "'I2S_MODE_ADC_BUILT_IN' was not declared in this scope" On ESP32-S3 using example

LabtronicsDesign, 2024-04-24:
### Board

ESP32S3 Dev Module

### Device Description

DevKit

### Hardware Configuration

Example code does not compile. No hardware required

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

Cannot compile the example code I2S/HiFreqADC

### Sketch

```cpp
see example code I2S/HiFreqADC
```

### Debug Message

```plain
'I2S_MODE_ADC_BUILT_IN' was not declared in this scope
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-04-24:
This mode was only available for original ESP32.  v3.0 uses the new [continuous mode API](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogReadContinuous/AnalogReadContinuous.ino) which will work on all esp32 devices. Download the [development version](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide) to test.

LabtronicsDesign, 2024-04-24:
> This mode was only available for original ESP32. v3.0 uses the new [continuous mode [...]

Thanks I will try this out to tomorrow m. For now I will close the issue

closed 2024-04-24 by LabtronicsDesign