espressif/arduino-esp32 #9570: ESP32-C6 WiFI.softAP is always auth open

mcuw, 2024-04-25:
### Board

ESP32-C6

### Device Description

nanoESP32-C6

### Hardware Configuration

nothing

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

macOS 14.4

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

2000000

### Description

Create a WiFI.softAP should create a protected WiFi AP when define a SSID and a passphrase.
Instead of of using these values the WiFi AP is always named as ESP_OC... and the wifi auth is "open". So the issue is that a secure connection is not possible.
Please double check!

### Sketch

```cpp
WiFi.mode(WIFI_AP);
delay(10);
WiFi.softAP("myssid", "mysecret");
```

### Debug Message

```plain
no errors are shown
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-04-25:
Are you sure there isn't another AP floating around?
I am unable to reproduce
[image]

mcuw, 2024-04-27:
Hi @lbernstone,
thanks a lot for your support and it works indeed.
I figured out my mistake ...
When I started to create an AP without an passphrase - the open auth is in place.
Then I added a passphrase with a length of less then 8 characters which is not a valid value for the `WiFi.softAP`.
I missed to check the return value and strange that the wifi connection keep on showing the open auth AP. Even when the new WiFi.softAP is failing with a new firmware.

I have to figure out how I can activate the logging ... with that I would see this error message: `log_e("passphrase too short!");`

thanks

closed 2024-04-27 by mcuw