espressif/arduino-esp32 #9571: Impossible to recover from a WiFi signal lost due to fadding

pberna67, 2024-04-28:
### Board

NINA-W101, NINA-W106

### Device Description

EVK-NINA-W101 from u-blox 

### Hardware Configuration

ESP32 I'm running  the v2.0.15 (Not listed in the Issue form)

https://www.u-blox.com/en/product/evk-nina-w10

### Version

v2.0.14

### IDE Name

Arduino 1.8.19

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I've encountered a problem using the library when the WiFi connection is lost due to a fading of the WiFi signal between the ESP32 module and the router. If the connection is lost, the library is no longer able to restore it again. The conditions for reproducing the problem are as follows:

Hw: EVK-NINA-W101/106 based on ESP32
https://www.u-blox.com/en/product/evk-nina-w10
Board lib package: ESP32 2.0.15 installed
Sketch example: See attached, 

How to reproduce the problem:
The app Mailbox_Change_Notification connects to internet via a router. Move slowly the EVN-NINA-W102 (ESP32) board far from the router wifi range, until the ESP32 will lost the Wifi connection (do not switch off the router and then switch on it, to simulate the Wifi drop, otherwise the problem doesn’t occurs). Wait some second (1 minutes, see Serial.Dump, wait for that time or more, shorter time couldn't trigger the problem) with ESP32 position out of Wifi range of the router, then slowly enter the ESP 32 EVK in the range of the Wifi router. The result is that the sketch is not able to reconnect to the WiFi

Regards
Paolo

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>

#define WIFI_SSID     "mySSID"
#define WIFI_PASSWORD "myPASS"

#define  NINA_W101

#define LED_R    23                  // Pin IO23 J4 LED RED
#define LED_R_CH 1 
#if defined(NINA_W101) || defined(NINA_W102)
#define LED_G    33                  // Pin IO33 J2 LED GREEN  per NINA-W101/102
#elif defined(NINA_W106)
#define LED_G    32                  // Pin IO32 J2 LED GREEN  per NINA-W106
[... 221 lines cut ...]
13:28:54.904 -> [126636][D][WiFiGeneric.cpp:1086] _eventCallback(): WiFi AutoReconnect Running
13:28:55.093 -> ........[129059][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
13:28:57.337 -> [129066][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 201 - NO_AP_FOUND
13:28:57.337 -> [129073][D][WiFiGeneric.cpp:1086] _eventCallback(): WiFi AutoReconnect Running
13:28:57.477 -> ....[130206][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
13:28:58.460 -> [130213][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 2 - AUTH_EXPIRE
13:28:58.460 -> [130220][D][WiFiGeneric.cpp:1086] _eventCallback(): WiFi AutoReconnect Running
13:28:58.694 -> ....[131332][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
13:28:59.586 -> [131340][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 2 - AUTH_EXPIRE
13:28:59.586 -> [131347][D][WiFiGeneric.cpp:1086] _eventCallback(): WiFi AutoReconnect Running
13:28:59.858 -> ...[132458][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
13:29:00.744 -> [132466][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 2 - AUTH_EXPIRE
13:29:00.744 -> [132473][D][WiFiGeneric.cpp:1086] _eventCallback(): WiFi AutoReconnect Running
13:29:00.789 -> ........[134895][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
13:29:03.169 -> [134903][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 201 - NO_AP_FOUND
13:29:03.169 -> [134910][D][WiFiGeneric.cpp:1086] _eventCallback(): WiFi AutoReconnect Running
13:29:03.169 -> ....[135956][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
13:29:04.241 -> [135964][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 39 - TIMEOUT
13:29:04.381 -> ..........................................................................................................................................................................................................................................[206109][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 9 - STA_LOST_IP
13:30:14.554 -> ........................
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pberna67, 2024-04-28:
At this time

13:29:04.381 -> ..........................................................................................................................................................................................................................................[206109][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 9 - STA_LOST_IP
13:30:14.554 -> ........................

The Module is moved closed to the Access point, but it doesn't reconnect.
Is the AutoReconnect procedure set by default ?

chipguyhere, 2024-06-03:
Is there more than one access point providing this same SSID?

closed 2025-01-15 by Parsaabasi