espressif/arduino-esp32 #9572: E (496) esp_core_dump_flash: No core dump partition found! Esp32 cam

uraannnn, 2024-04-28:
### Board

esp32 cam with esp 32 cam mb

### Device Description

esp32 cam with esp 32 cam mb

### Hardware Configuration

.

### Version

latest master (checkout manually)

### IDE Name

arduino ide

### Operating System

win10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I recently bought a esp 32 com and wanted to flash it and make the stream, I flashed it and it got to 100% but it gives the following error e�3Rյ � �$ 0::�SHBC!� �01 = UI��YUM QUI�E'�1d PS�e�*�e M��%=J60��*j��'�b S%� ��H kM_!ɲ��b�}���00`,!}"ɖi0,10Eɲ�0`,A�Eɲ�0,q"ɖ�0�j
��'T�Q ,�]c-"���1
!�+:�3�� ��&K: � S�+:� �  ���b)K��� SH�t+:� �� ��bS���S �j
0�.V `4�� �% j
E (496) esp_core_dump_f�f6�: No core dump partition found!
E (496) esp_core_dump_flash: No core dump partition found! when I click on the rst button, help me solve the problem

### Sketch

```cpp
.
```

### Debug Message

```plain
I recently bought a esp 32 com and wanted to flash it and make the stream, I flashed it and it got to 100% but it gives the following error e�3Rյ � �$ 0::�SHBC!� �01 = UI��YUM QUI�E'�1d PS�e�*�e M��%=J60��*j��'�b S%� ��H kM_!ɲ��b�}���00`,!}"ɖi0,10Eɲ�0`,A�Eɲ�0,q"ɖ�0�j
��'T�Q ,�]c-"���1 !�+:�3�� ��&K: � S�+:� � ���b )K��� SH�t+:� �� ��bS���S �j
0�.V `4�� �% j
E (496) esp_core_dump_f�f6�: No core dump partition found!
E (496) esp_core_dump_flash: No core dump partition found! when I click on the rst button, help me solve the problem
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-04-28:
Change your Serial.begin statement to use 115200 baud, matching with the setting in the serial monitor. Use a partition scheme with coredump (all the built-in partition schemes should have coredump).

5chufti, 2024-08-07:
it is throwing the same error message even with built in partition schemes ...

quantumDSflux, 2024-09-24:
yep me getting the same problem......!!!

5chufti, 2024-09-26:
solution:
ad a file "partitions.csv" to your sketch directory with following content:

```
# Name,   Type,  SubType, Offset,   Size,    Flags
nvs,      data,  nvs,     0x9000,   0x5000,
otadata,  data,  ota,     0xe000,   0x2000,
app0,     app,   ota_0,   0x10000,  0x3d0000,
fr,       data,        ,  0x3e0000, 0x20000,
coredump, data, coredump,         , 0x10000,
```
and select "partition scheme = custom" in settings

lbernstone, 2024-09-27:
If you enable "Erase all flash before upload", and then choose the default partition scheme, you will repair the coredump partition. Note that this will delete anything saved in nvs/Preferences/EEPROM.