espressif/arduino-esp32 #9579: ULP function not functioning properly on ESP32S3

Llgok, 2024-05-02:
### Board

ESP32S3 Dev Module

### Device Description

ESP32S3 Dev Module

### Hardware Configuration

not anything

### Version

v2.0.14

### IDE Name

Arduino IDE and PlatformIO IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I attempted to compile a program for ULP counting using C Macros on ESP32S3 with Arduino IDE and PlatformIO IDE (2.0.14), but it did not work on ESP32S3 and threw an error "ulp: program too big: 11 words, max is 0 words". The same program worked fine on ESP32. Later, when I switched the library version to 3.0.0-alpha2, it failed to compile with the error "fatal error: esp32/ulp.h: No such file or directory".

Is the ULP functionality for ESP32S3 not well adapted on Arduino IDE yet? It's worth mentioning that in the latest ESP-IDF version, both ESP32S3 and ESP32 can successfully utilize the ULP functionality.

### Sketch

```cpp
#include <Arduino.h>
#include "esp32s3/ulp.h"
#include "driver/rtc_io.h"
#include "sdkconfig.h"

#define ULP_START_OFFSET 32

void ULP_Initialization()
{
    memset(RTC_SLOW_MEM, 0, CONFIG_ESP32S3_ULP_COPROC_RESERVE_MEM);

    const ulp_insn_t program[] = {
        I_MOVI(R0, 0), // R0 =0
        I_MOVI(R1, 0), // R1 = 0;

        M_LABEL(1), // do {

        // delay(1000);
        I_MOVI(R0, 200), // R0 = n * 1000 / 5, where n is the number of seconds to delay, 200 = 1 s
        M_LABEL(2),      // do {
                         // delay (5);
                         // since ULP runs at 8 MHz
                         // 40000 cycles correspond to 5 ms (max possible delay is 65535 cycles or 8.19 ms)
        I_DELAY(40000),
        I_SUBI(R0, R0, 1), // R0 --;
        M_BGE(2, 1),       // } while (R0 >= 1); ... jump to label 2 if R0 > 0

        I_ADDI(R1, R1, 1), // R1 ++
        // store sample count to RTC_SLOW_MEM [0]
        I_ST(R1, 0, 0), // RTC_SLOW_MEM [0] = R1;

        I_MOVR(R0, R1), 
        M_BL(1, 100),   // ... jump to label 1 if R0 < 100

        I_END(), 
    };
    
    size_t size = sizeof(program) / sizeof(ulp_insn_t);
    ulp_process_macros_and_load(0, program,&size);
    ulp_run(0);

    // in which it also prints the debug output mentioned earlier
    Serial.printf("ULP Mem in main app: %d\n", CONFIG_ESP32S3_ULP_COPROC_RESERVE_MEM);
}

void setup()
{
    Serial.begin(115200);
    ULP_Initialization();
}

void loop()
{
    for (int i = 0; i < 10; i++)
    {
        Serial.printf("RTC_SLOW_MEM[%d]: %d\n", i, RTC_SLOW_MEM[i] & 0xFFFF);
    }

Serial.printf("\n");
    delay(1000);
}
```

### Debug Message

```plain
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4201e66e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
E (2) ulp: program too big: 11 words, max is 0 words
ULP Mem in main app: 0
RTC_SLOW_MEM[0]: 47214
RTC_SLOW_MEM[1]: 0
RTC_SLOW_MEM[2]: 33694
RTC_SLOW_MEM[3]: 0
RTC_SLOW_MEM[4]: 58295
RTC_SLOW_MEM[5]: 50288
RTC_SLOW_MEM[6]: 32486
RTC_SLOW_MEM[7]: 4171
RTC_SLOW_MEM[8]: 38900
RTC_SLOW_MEM[9]: 30073
```

### Other Steps to Reproduce

No response

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-05-02:
Looks like CONFIG_ULP_COPROC_TYPE_FSM needs to be added for S2 & S3. I'll open an issue in the lib-builder.

Llgok, 2024-05-02:
> Looks like CONFIG_ULP_COPROC_TYPE_FSM needs to be added for S2 & S3. I'll open an issue in the lib-builder. Just FYI, the S2 & S3 include a [RISC-V ULP](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/ulp-risc-v.html) as well which can be programmed directly with C.

Hello, I have just added CONFIG_ULP_COPROC_TYPE_FSM=y & CONFIG_ULP_COPROC_TYPE_RISCV=y in the sdkconfig file of ESP32S3, but the issue remains the same as before. Did I modify the sdkconfig file incorrectly?

# CONFIG_ESP32S3_TRAX is not set
CONFIG_ESP32S3_TRACEMEM_RESERVE_DRAM=0x0
# CONFIG_ESP32S3_ULP_COPROC_ENABLED is not set
CONFIG_ULP_COPROC_TYPE_FSM=y
CONFIG_ULP_COPROC_TYPE_RISCV=y
CONFIG_ESP32S3_ULP_COPROC_RESERVE_MEM=1024
CONFIG_ESP32S3_DEBUG_OCDAWARE=y
CONFIG_ESP32S3_BROWNOUT_DET=y
CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_7=y
# CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_6 is not set
# CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_5 is not set
# CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_4 is not set
# CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_3 is not set
# CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_2 is not set
# CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_1 is not set

Arduino: sdkconfig file path -> ...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\tools\sdk\esp32s3\sdkconfig 
PlatformIO: sdkconfig file path -> ...\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32s3\sdkconfig

lbernstone, 2024-05-02:
You would need to recompile the sdk from scratch. It will get fixed when the espressif guys make a new release.

Llgok, 2024-05-02:
> You would need to recompile the sdk from scratch. It will get fixed when the espressif guys make a new release.

Just now I cleared the compilation cache and recompiled from scratch, but the problem remains the same. The counting register did not change. Is there any feature that I haven't enabled yet?

me-no-dev, 2024-05-02:
First you need to install 3.0.0-rc1 from the board manager (DEV JSON link) and then you can try to replace the libs in `...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-rc1\tools\esp32-arduino-libs` with the ones from [here](https://github.com/espressif/esp32-arduino-lib-builder/actions/runs/8921956430/artifacts/1466720146)

This should enable ULP for you on S2 and S3. We will not be backporting it to 2.0.x

closed 2024-05-02 by me-no-dev

Llgok, 2024-05-03:
> First you need to install 3.0.0-rc1 from the board manager (DEV JSON link) and then you can try to replace the libs in [...]

I tried to adapt ULP in 3.0.0-rc1 but encountered an error when including the header file.
﻿
Error message ->
“fatal error: esp32s3/ulp.h: No such file or directory
11 | #include "esp32s3/ulp.h"
|          ^~~~~~~~~~~~~~~
compilation terminated.
﻿
exit status 1
﻿
Compilation error: esp32s3/ulp.h: No such file or directory”

mmirio, 2025-04-15:
This is "normal."
None of the examples compile for ESP32-S3..
ulp_riscv_uart.c:8:10: fatal error: ulp_riscv.h: No such file or directory   
ulp_riscv_adc_ulp_core.h:10:10: fatal error: hal/adc_ll.h: No such file or directory

I haven't found any solution.