espressif/arduino-esp32 #9582: ESP32 S3 serial and USB HID stall when DTR is activated a second time

ChrGri, 2024-05-02:
### Board

ESP32 S3 dev

### Device Description

esp32-s3-devkitc-1

### Hardware Configuration

USBOTG connection to PC

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Win 11

### Flash frequency

-

### PSRAM enabled

no

### Upload speed

-

### Description

ESP32 S3 runs simple program that should provided USB HID gamepad and serial messages via USB OTG port. Im using the Windows inbuilt USB gamecontroller program to monitor the gamepad data and [this serial monitor app](https://sites.google.com/site/terminalbpp/) to monitor the serial messages. The serial monitor app allows to set the DTR/RTS states manually, by pressing the red marked software buttons<br>
[image]

The first activation of DTR enables the serial output as expected. When DTR is disabled, USB HID output continiues, which is also expected. But, when DTR is activated a second time, USB HID and serial output stall, which is not as expected.

Apart from that, when serial and USB HID output are activated, both will stall after a while, [see](https://github.com/espressif/arduino-esp32/issues/7554#issuecomment-2088049999).

### Sketch

```cpp
Can be found here:
https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/tree/develop/Validation/Joystick_Serial_test
```

### Debug Message

```plain
-
```

### Other Steps to Reproduce

1. Flash firmware
2. Open serial monitor app
3. Connect to COM port
4. Enable DTR
5. Check USB HID and serial output
6. Deactivate DTR
7. USB HID output still updating
8. Activate DTR
9. USB HID and serial output stall 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-05-06:
Is this about USB Serial Device (CDC-ACM) signals?

SuGlider, 2024-05-06:
@ChrGri - please keep in mind that DTR and RTS signals of the USB CDC port actually control "reset" and "boot mode" signals of the ESP32-S3.

SuGlider, 2024-05-06:
It may be putting the S3 into download mode... "freezing/halting" the S3 firmware execution...

SuGlider, 2024-05-06:
@ChrGri - Please read this: https://github.com/espressif/arduino-esp32/issues/6762#issuecomment-1182821492

[image]

ChrGri, 2024-05-06:
@SuGlider Thank you very much. I understood, that to enter the download mode on a regular ESP, one has to execute a specific DTR/RTS [sequence](https://electronics.stackexchange.com/questions/473443/esp32-dev-board-dtr-rts), as shown below
[image].

From the documentation you referenced, I understood that only the download mode flag is toggled, but the chip is never asked to reset:
[image]

I think setting [enableReboot](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/usb_cdc.html) has improved the behaviour and the USB HID output is continuous independent of the DTR state. However, simultaneous USB HID and serial output are still buggy and both stall after some time.

SuGlider, 2024-05-06:
>However, simultaneous USB HID and serial output while RTS/DTR sates beeing const. are still buggy and both stall after some time. Single output of serial data or single output of USB HID data works fine though.

I see that you are using PlatformIO. What is the Arduino Core version used with it?
If possible, could you test it with Arduino IDE using Arduino Core 3.0.0-RC1?

ChrGri, 2024-05-06:
> I see that you are using PlatformIO. What is the Arduino Core version used with it? If possible, could you test it with Arduino IDE using Arduino Core 3.0.0-RC1?

I've changed the platformIO as shown below. No difference. 
[image]

You can find the sample code [here](https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/tree/develop/Validation/Joystick_Serial_test). I've played around with `Serial.setTxTimeoutMs(0);` and `Serial.flush();` without success. The observation is always, as soon as USB HID and serial output are activated together, the output will stall at some point.

BR
Chris

ChrGri, 2024-05-06:
One thing which helped a lot, was adding a delay between the serial write and USB HID output command, as such
[image].

SuGlider, 2024-05-06:
I understand that your project uses USB OTG and TinyUSB for both, HID and CDC.
The delay seems to give TinyUSB some room to run different tasks from those 2 endpoints and/or their tasks...

It may be necessary to run a deep investigation of this issue.

ChrGri, 2024-05-06:
> I understand that your project uses USB OTG and TinyUSB for both, HID and CDC. The delay seems to give TinyUSB some room to run different tasks from those 2 endpoints and/or their tasks...
> 
> It may be necessary to run a deep investigation of this issue.

Yes, CDC and HID. Do you have an idea, why `Serial.flush()` apparently doesn't give that room too?

Is there something I can to help to find a proper fix?

BR
Chris

SuGlider, 2024-05-06:
I'd need to analyse it better and try to understand why it could fail. 

Regarding `Serial. Flush()`, it actually just waits until all the data is sent to the USB Host.
By other hand, `delay(time_ms)`  releases the FreeRTOS Scheduler to run another task while waiting for that time to pass.

Arduino has a 1 ms time slice defined for FreeRTOS. 
`delay(5);` means waiting for 5 time slices while other tasks could be executed - maybe something related to HID... 
This is the "room" that I've mentioned about.

It is necessary to see the whole picture and try to understand the sequence of actions and possible reasons for the issue.

ChrGri, 2024-06-05:
Just want to report that 2.0.17 behaves identical.

ChrGri, 2024-07-08:
@SuGlider I dont want to bother you, but is there something I can do to accelerate the investigation?

SuGlider, 2024-07-08:
There is a way to check if the S3 enters in Boot Mode by connecting another terminal application to the UART0 output and check the messages ROM will output when DTR is activated.

ChrGri, 2024-07-08:
Thank you @SuGlider You already explained the RTS/DTR logic above. The remaining issue is, that the serial output stalls after a while, see https://github.com/espressif/arduino-esp32/issues/9582#issuecomment-2095318759

I initially thought that serial transmission only stalls, when CDC and HID output is activated, but it seams to stall when only CDC is activcated too.

SuGlider, 2024-07-08:
Would it be possible for you to post a basic sketch with detailed explanation in order to help me to reproduce the CDC stalling?

Thanks.

ChrGri, 2024-07-08:
> Would it be possible for you to post a basic sketch with detailed explanation in order to help me to reproduce the CDC stalling?
> 
> Thanks.

For sure. A sample script was posted here:
https://github.com/espressif/arduino-esp32/issues/9582#issuecomment-2096672910

SuGlider, 2024-07-08:
The issue is also only with CDC, right?
I understand that it is not necessary to use HID together with CDC.

How should I reproduce the CDC issue?

ChrGri, 2024-07-10:
> The issue is also only with CDC, right? I understand that it is not necessary to use HID together with CDC.
> 
> How should I reproduce the CDC issue?

I'll try to generate a simple code to reproduce the issue with CDC only. This will take some time though. In the meantime, I suggest looking into the sample code from above (HID + CDC) which will create the serial stall, when the delay call is removed:
[image]

Thank you

ChrGri, 2024-07-29:
@SuGlider Did you had a chance to look into the HID+CDC example already?

SuGlider, 2024-07-30:
Hi @ChrGri - I just returned from vacation. This is in my list of work. Please give me a week and I'll test it.

ChrGri, 2024-08-14:
@SuGlider any updates?

ChrGri, 2024-08-22:
@SuGlider Anything I can do to help?

ChrGri, 2024-09-01:
@SuGlider it's been over a month since you replied. Can you confirm the issue from [here](https://github.com/espressif/arduino-esp32/issues/9582#issuecomment-2258902207) now?

ChrGri, 2024-09-07:
The main issue was described. The cdc+hid issue will be tracked [here](https://github.com/espressif/arduino-esp32/issues/10307).

closed 2024-09-07 by ChrGri