espressif/arduino-esp32 #9585: RTL8201 unable to obtain IP on DHCP

khawajamechatronics, 2024-05-03:
### Board

ESP32E

### Device Description

Hi,

I am using custom esp32 board with RTL8201 connected. I have tested code with ESP32 + LAN8720 and it works like a charm. But when I use same code modified for RTL8201 it is unable to obtain IP hence cannot connect to internet.

It doesnt call ARDUINO_EVENT_ETH_GOT_IP.

### Hardware Configuration

#define ETH_PHY_TYPE        ETH_PHY_RTL8201
#define ETH_PHY_ADDR         0
#define ETH_PHY_MDC         23
#define ETH_PHY_MDIO        18
#define ETH_PHY_POWER       17
#define ETH_CLK_MODE        ETH_CLOCK_GPIO0_IN

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

240

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi,

ETH code for RMII works with LAN8720 on WT32-ETH01 board but doesnt work with custom ESP32 +RTL8201 board.

It doesnt call GOT IP event.

Thanks

### Sketch

```cpp
/*
    This sketch shows the Ethernet event usage

*/

// Important to be defined BEFORE including ETH.h for ETH.begin() to work.
// Example RMII LAN8720 (Olimex, etc.)
#ifndef ETH_PHY_TYPE
#define ETH_PHY_TYPE        ETH_PHY_RTL8201
#define ETH_PHY_ADDR         0
#define ETH_PHY_MDC         23
#define ETH_PHY_MDIO        18
#define ETH_PHY_POWER       17
#define ETH_CLK_MODE        ETH_CLOCK_GPIO0_IN
#endif

#include <ETH.h>

static bool eth_connected = false;

// WARNING: onEvent is called from a separate FreeRTOS task (thread)!
void onEvent(arduino_event_id_t event)
{
  switch (event) {
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      // The hostname must be set after the interface is started, but needs
      // to be set before DHCP, so set it from the event handler thread.
      ETH.setHostname("esp32-ethernet");
      break;
    case ARDUINO_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case ARDUINO_EVENT_ETH_GOT_IP:
      Serial.println("ETH Got IP");
      Serial.println(ETH);
      eth_connected = true;
      break;
    case ARDUINO_EVENT_ETH_LOST_IP:
      Serial.println("ETH Lost IP");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
    Serial.println("ETH Case");
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  NetworkClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  Network.onEvent(onEvent);
  ETH.begin();

  while ( !eth_connected) {
    delay(100);
 // wait for eth connection...
}
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1416
load:0x40078000,len:14804
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
[     2][D][esp32-hal-cpu.c:259] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 1.45
  Cores             : 2
  Frequency         : 240 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   381064 B ( 372.1 KB)
  Free Bytes        :   350756 B ( 342.5 KB)
  Allocated Bytes   :    23208 B (  22.7 KB)
  Minimum Free Bytes:   345280 B ( 337.2 KB)
  Largest Free Block:   114676 B ( 112.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  8388608 B (8 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : May  3 2024 10:38:45
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.2-185-g3662303f31-dirty
  Arduino Version   : 3.0.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : DFROBOT_FIREBEETLE_2_ESP32E
  Arduino Variant   : dfrobot_firebeetle2_esp32e
  Arduino FQBN      : esp32:esp32:dfrobot_firebeetle2_esp32e:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=debug,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none
============ Before Setup End ============
ETH Started
ETH Connected
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-05-03:
Wrong clock mode used. Correct is `ETH_CLOCK_GPIO17_OUT` for the RTL8201

kostaond, 2024-05-06:
@khawajamechatronics could be RMII CLK configuration as indicated by @Jason2866. However, since it's custom board, it could also be other HW design related issue. For start, what [RMII CLK configuration are you planning to use?](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_eth.html#configure-mac-and-phy) You could also share your design so we could try to help you.

khawajamechatronics, 2024-05-06:
Thank you so much, after configuring hardware and software with CLK GPIO 17 its working. :)

me-no-dev, 2024-05-06:
Thanks @kostaond and @Jason2866 for helping out! Closing as solved

closed 2024-05-06 by me-no-dev