espressif/arduino-esp32 #9587: Brownout event halts the system, when the voltage increase it doesn't resume working.

brunofgc, 2024-05-03:
### Board

ESP32-C3 using Arduino

### Device Description

It's a module soldered into my own device.

### Hardware Configuration

Just a Modem SIM800L conencted in.

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi there!
I'm using the ESP32-C3 with the Arduino Framework.

The system uses a battery as backup for AC Power Source, and it works okay. However, when the battery drains too much, the system stops working. Additionally, when the AC Power is restored, it recharges the battery, but the system doesn't resume working. I need to completely turn off and turn on again in order to back working. "Hardware Reset".

Do you know of any system functions or calls that can detect when the system halts/fails for any reason? Perhaps I could detect a "brownout event" and take some actions accordingly?

Thank you in advance.

### Sketch

```cpp
It's a very big program. I'll be making a small one to try reproduce this error and post here soon.
```

### Debug Message

```plain
I couldn't capture it is happening on clients.
```

### Other Steps to Reproduce

Decrease the voltage until it stop working, and then get the voltage back to an operating range.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-03:
There are chips out there (I think they were called "voltage supervisors") that would pull the ESP reset when voltage drops to certain level (must be higher than what the ESP will trigger as too low) and will release the pin once voltage goes above that level again. Such solution is known to work in your exact case

Quindus, 2024-05-26:
Must be issue with the Power. When you using SIM 800L it required 2 Amps while trying to connect network. Esp 32 required 500 mA while connecting to wifi. At the peak your power supply may stuck to produce enough power. So it drops voltae for some milli seconds. Then you got brown out.

Try power up the SIM800L odulw seperately. This may work.

VojtechBartoska, 2024-09-12:
Hello @brunofgc, is this still valid?

brunocavalcanti00px, 2024-09-13:
Hi @VojtechBartoska . Thank you for your support. I resolved it by implementing my own power supervisor and puting the system on "Deep Sleep" mannualy by command. When it cross the low power line, I turn off modem and put it to deep sleep for 2 minutes. When I be back, I check for the battery, if it is now above the minimum level I start the system again, if not I go sleep for 2 minutes. I do it over and over. It was good to prevent being stuck.

closed 2024-09-16 by VojtechBartoska