espressif/arduino-esp32 #9588: ESP32 “SPIClass spi(HSPI)” but with pin definitions?

Rob58329, 2024-05-03:
### Board

NodeMCU-32s

### Device Description

Board connected to "Adafruit_ST7796S_kbv" tft (although not really relevant)  using the HSPI .

### Hardware Configuration

#define SD_CS    15
#define TFT_CS   33 
#define TFT_DC   17 
#define SPI_MISO 12 // default pin
#define SPI_MOSI 13 // default pin
#define SPI_CLK  14 // default pin
#define TFT_RST -1

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE v1.8.19

### Operating System

Windows 10

### Flash frequency

n/a

### PSRAM enabled

no

### Upload speed

n/a

### Description

The 1st below sketch works fine, but only with the default HSPI pin definitions.

```
#define SD_CS    15
#define TFT_CS   33 
#define TFT_DC   17 
#define SPI_MISO 12 // default pin
#define SPI_MOSI 13 // default pin
#define SPI_CLK  14 // default pin
#define TFT_RST -1

#define use_TFT true

#include <SPI.h>
#include <Adafruit_GFX.h>  // https://github.com/adafruit/Adafruit-GFX-Library 
#include <Adafruit_ST7796S_kbv.h> // https://github.com/prenticedavid/Adafruit_ST7796S_kbv1 

SPIClass MySPI(HSPI); // Declare an HSPI bus object (to use or BOTH TFT and SD-card)
Adafruit_ST7796S_kbv tft=Adafruit_ST7796S_kbv(&MySPI, TFT_DC, TFT_CS, TFT_RST);

void setup() {
  SerialD.begin();
  #if (use_TFT)
    tft.setRotation(3);
    tft.setTextSize(1); 
    tft.fillScreen(ST7796S_BLACK); tft.setCursor(0,0);
    tft.setTextColor(ST7796S_WHITE);
  #endif
} 

void loop() {
  SerialD.println("Loop");
  delay(1000);
}
```

The 2nd below sketch with new HSPI pin definitions compiles OK, but **crashes** at the line “SerialD.begin();” [aka tft.begin()]

```
#define SD_CS    15
#define TFT_CS   33 
#define TFT_DC   17 
#define SPI_MISO 12 // default pin
#define SPI_MOSI 13 // default pin
#define SPI_CLK  14 // default pin
#define TFT_RST -1

#define use_TFT true

#include <SPI.h>
#include <Adafruit_GFX.h>  // https://github.com/adafruit/Adafruit-GFX-Library 
#include <Adafruit_ST7796S_kbv.h> // https://github.com/prenticedavid/Adafruit_ST7796S_kbv1 

SPIClass *spi = NULL;
Adafruit_ST7796S_kbv tft=Adafruit_ST7796S_kbv(spi, TFT_DC, TFT_CS, TFT_RST);

#if (use_TFT)
  #define SerialD tft
#else
 #define SerialD Serial
#endif

void setup() {
  #if (use_TFT)
    spi = new SPIClass(HSPI);
    spi->begin(SPI_CLK, SPI_MISO, SPI_MOSI, TFT_CS);
  #endif
  SerialD.begin(); // crashes here
  #if (use_TFT)
    tft.setRotation(3);
    tft.setTextSize(1); 
    tft.fillScreen(ST7796S_BLACK); tft.setCursor(0,0);
    tft.setTextColor(ST7796S_WHITE);
  #endif
} 

void loop() {
  SerialD.println("Loop");
  delay(1000);
}

```

**Is there any way to change the used HSPI pins outside of “setup()” and “loop()” so my above “#define SerialD tft” will work with “SerialD.print” etc?**

The 3rd below sketch does works, but does NOT allow “#define SerialD tft”, and instead requires every “SerialD.print” to be replaced with “tft->print”, which is NOT what I want to do.

```
#define SD_CS    15
#define TFT_CS   33 
#define TFT_DC   17 
#define SPI_MISO 12 // default pin but can be changed
#define SPI_MOSI 13 // default pin but can be changed
#define SPI_CLK  14 // default pin but can be changed
#define TFT_RST -1

#define use_TFT true

#include <SPI.h>
#include <Adafruit_GFX.h>  // https://github.com/adafruit/Adafruit-GFX-Library 
#include <Adafruit_ST7796S_kbv.h> // https://github.com/prenticedavid/Adafruit_ST7796S_kbv1 

SPIClass *spi = NULL;
Adafruit_ST7796S_kbv *tft= NULL; 

#if (use_TFT)
  #define SerialD tft
#else
 #define SerialD Serial
#endif

void setup() {
  Serial.begin(115200);
  #if (use_TFT)
    spi = new SPIClass(HSPI);
    spi->begin(SPI_CLK, SPI_MISO, SPI_MOSI, TFT_CS);
    tft = new Adafruit_ST7796S_kbv(spi, TFT_DC, TFT_CS, TFT_RST);
    tft->begin();
  #else
    SerialD.begin(); 
  #endif
  #if (use_TFT)
    tft->setRotation(3);
    tft->setTextSize(1); 
    tft->fillScreen(ST7796S_BLACK); tft->setCursor(0,0);
    tft->setTextColor(ST7796S_WHITE);
  #endif
} 

void loop() {
  #if (use_TFT)
    tft->println("Loop");
  #else
    SerialD.println("Loop");
  #endif
  delay(1000);
}

```

### Sketch

```cpp
See above
```

### Debug Message

```plain
Sketch 2 crashes at the "SerialD.begin();"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-05-03:
KISS. HSPI is the default, and there is already an object (SPI) defined on that bus.
```
#include <SPI.h>
...
   SPI.begin(SPI_CLK, SPI_MISO, SPI_MOSI, TFT_CS);
```
If you want to use VSPI, you can do so like:
```
#include <SPI.h>
SPIClass VSPI(1);
...
   VSPI.begin(SPI_CLK, SPI_MISO, SPI_MOSI, TFT_CS);
```

closed 2024-05-04 by Rob58329

lbernstone, 2024-05-06:
> Edit2: @lbernstone - many thanks for below info! Interestingly the default "SPI" object "SPI.begin();" appears to use the pre-defined (SCK,MOSI, MISO,SS) variables, which on my board (nodeMCU-32S) are GPIO 18,23,19 & 5 which I understand to be the VSPI interface. But this still works fine.

In arduino-esp32 context, SPI names are just names- HSPI evaluates to bus 2, and should be the horizontal pins (2,4,12,15), but that's just not how they are defined in the [variants file](https://github.com/espressif/arduino-esp32/blob/master/variants/esp32/pins_arduino.h#L12-L15).  Since it is all muxed through the gpio matrix, that doesn't really matter ([although a bit confusing](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.h#L31-L43)). Espressif has gotten away from the H/V naming, since it doesn't really apply to the other variants, but it is kinda stuck in arduino to maintain backwards compatibility. If you are going to define your own objects, I'd stick to using the numbers

eMUQI, 2025-06-25:
@lbernstone Hello, sorry to bother you. Thank you for your detailed explanations in previous discussions. I'd like to make a further inquiry regarding the SPI bus naming convention for ESP32-S3.

While researching SPI bus configuration, I noticed the following usage pattern:

```cpp
SPIClass oledSPI(HSPI);
...
oledSPI.begin(OLED_SCK, -1, OLED_MOSI, OLED_CS);
```

At the same time, I noticed the following definitions in the current version (3.2.0) of `esp32-hal-spi.h`:

```cpp
#ifdef CONFIG_IDF_TARGET_ESP32S2
#define FSPI 1  //SPI 1 bus. ESP32S2: for external memory only (can use the same data lines but different SS)
#define HSPI 2  //SPI 2 bus. ESP32S2: external memory or device  - it can be matrixed to any pins
#define SPI2 2  // Another name for ESP32S2 SPI 2
#define SPI3 3  //SPI 3 bus. ESP32S2: device only - it can be matrixed to any pins
#elif CONFIG_IDF_TARGET_ESP32
#define FSPI 1  //SPI 1 bus attached to the flash (can use the same data lines but different SS)
#define HSPI 2  //SPI 2 bus normally mapped to pins 12 - 15, but can be matrixed to any pins
#define VSPI 3  //SPI 3 bus normally attached to pins 5, 18, 19 and 23, but can be matrixed to any pins
#else
#define FSPI 0
#define HSPI 1
#endif
```

My question is: why doesn't ESP32-S3 have a VSPI definition? From my understanding, ESP32-S3 has 4 SPI buses, with SPI2 and SPI3 being available for users. Following the ESP32 naming convention, shouldn't these two buses correspond to HSPI and VSPI respectively?

I saw your previous comment that "Espressif has gotten away from the H/V naming, since it doesn't really apply to the other variants." Does this mean that for ESP32-S3, the official recommendation is to use HSPI uniformly (or use numeric IDs directly) to access user SPI buses?

@P-R-O-C-H-Y, I noticed you have recent related modifications in #11510. I'd like to ask for your perspective on this naming convention issue.

Looking forward to your guidance, thank you!

Rob58329, 2025-06-30:
@eMUQI From the above:

```
Note that the ESP32-S3 has 2 available/user SPI controllers named:

    SPIClass(HSPI); or SPIClass(1); = GP-SPI2 (or perhaps GP-SPI3)
    SPIClass(FSPI); or SPIClass(0); = GP-SPI3 (or perhaps GP-SPI2)
    (each has slightly different capabilities).
```

You can use which ever naming convention makes most sence to you.