espressif/arduino-esp32 #9590: Core panic when using WiFi and PxMatrix libraries

radical-las, 2024-05-05:
### Board

ESP-WROOM-32 DevKit v1

### Device Description

stock ESP32 DevKit v1

### Hardware Configuration

P4 LED Matrix 32x64 connected via logic level converter (SN74LVC245AN).

### Version

v2.0.14

### IDE Name

VS Code + Platform IO + ArduinoFramework

### Operating System

Windows 10

### Flash frequency

240MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I've tried platforms v1.0.6 and v2.0.14 using WiFi and PxMatrix libraries, on both platforms, if LED matrix initialized before WiFi.begin() - board gets core panic.

On platform v2.0.14, if LED matrix initialized after WiFi.begin() and WiFi connection failed - board gets core panic. But if WiFi connected successfully, board seems to work fine until WiFi reconnection is required for any reason.

On platform v1.0.6, if LED matrix initialized after WiFi.begin() - board works fine, at least for dozens of minutes, didn't tried for longer.

### Sketch

```cpp
#include <Arduino.h>
#include <PxMatrix.h>
#include <WiFi.h>

const char *ssid = "ssid";
const char *password = "password";

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_OE 16
#define MATRIX_WIDTH 64
#define MATRIX_HEIGHT 32
#define PxMATRIX_double_buffer true
PxMATRIX display(MATRIX_WIDTH, MATRIX_HEIGHT, P_LAT, P_OE, P_A, P_B, P_C, P_D);

uint8_t display_draw_time = 5;
hw_timer_t *timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR display_updater() {
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(display_draw_time);
  portEXIT_CRITICAL_ISR(&timerMux);
}

void display_update_enable(bool is_enable) {
  if (is_enable) {
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  } else {
    timerDetachInterrupt(timer);
    timerAlarmDisable(timer);
  }
}

void setup() {
  Serial.begin(115200);

  WiFi.begin(ssid, password);

  int attemptCounter = 0;
  while (WiFi.status() != WL_CONNECTED && attemptCounter <= 10) {
    attemptCounter++;
    delay(1000);
  }

  display.begin(16);
  display.setMuxPattern(SHIFTREG_ABC_BIN_DE);
  display.setFastUpdate(true);
  display.setTextWrap(false);
  display.clearDisplay();
  display_update_enable(true);
  display.showBuffer();

  Serial.println("END OF INIT");
}

void loop() {
  delay(1000);
}
```

### Debug Message

```plain
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

assert failed: xQueueSemaphoreTake queue.c:1554 (!( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) && ( xTicksToWait != 0 ) ))

Backtrace: 0x40083b91:0x3ffbf38c |<-CORRUPTED

  #0  0x40083b91:0x3ffbf38c in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
```

### Other Steps to Reproduce

I'm using buildin WiFi library. PxMatrix can be found here - [PxMatrix by 2dom](https://github.com/2dom/PxMatrix). But I use little bit different version of PxMatrix, in latest releases author removed part of code that provided support of SHIFTREG_ABC_BIN_DE mux pattern, so I added that part from master branch, which is not updated somehow. [Modified PxMatrix.h](https://github.com/radical-las/PxMatrixExtended/blob/master/PxMatrix.h)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-05:
you can not update the display inside an interrupt on any ESP32 platform. You need to set a flag in the ISR and check for it in a Task or in the loop.

closed 2024-05-05 by me-no-dev

radical-las, 2024-05-05:
Hi @me-no-dev! Well, it kinda working now, but display is flickering now. And I'm curious why core goes panic only when I'm trying to use WiFi? And, also, why on platform 1.0.6 it's working (even with some reservations)?

me-no-dev, 2024-05-05:
When you start WiFi many things will run on the chip, so it will interfere. 1.0.6 is very old on now EOL IDF v3.3 where many things were different. Display flickers probably because you are changing and updating the LCD at the same time.