espressif/arduino-esp32 #9591: #include "esp32s3/ulp.h" No such file or directory

Llgok, 2024-05-06:
### Board

ESP32S3 Dev Module

### Device Description

not anything

### Hardware Configuration

fatal error:
   14 | #include "esp32s3/ulp.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: esp32s3/ulp.h: No such file or directory

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino-IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

On the ESP32 Dev Module, you can find the header file "esp32/ulp.h", but on the ESP32S3 Dev Module, the header file "esp32s3/ulp.h" cannot be found.

### Sketch

```cpp
#include <Arduino.h>
#include "driver/rtc_io.h"
#include "sdkconfig.h"
#include "soc/rtc_cntl_reg.h"
#include "esp32s3/ulp.h"

#define ULP_START_OFFSET 32

void ULP_Initialization()
{
    // memset(RTC_SLOW_MEM, 0, CONFIG_ESP32S3_ULP_COPROC_RESERVE_MEM);

    const ulp_insn_t program[] = {
        I_MOVI(R0, 0), // R0 =0
        I_MOVI(R1, 0), // R1 = 0;

        M_LABEL(1), // do {

        // delay(1000);
        I_MOVI(R0, 200), // R0 = n * 1000 / 5, where n is the number of seconds to delay, 200 = 1 s
        M_LABEL(2),      // do {
                         // delay (5);
                         // since ULP runs at 8 MHz
                         // 40000 cycles correspond to 5 ms (max possible delay is 65535 cycles or 8.19 ms)
        I_DELAY(40000),
        I_SUBI(R0, R0, 1), // R0 --;
        M_BG(2, 1),       // } while (R0 >= 1); ... jump to label 2 if R0 > 0

        I_ADDI(R1, R1, 1), // R1 ++
        // store sample count to RTC_SLOW_MEM [0]
        I_ST(R1, 0, 0), // RTC_SLOW_MEM [0] = R1;

        I_MOVR(R0, R1), 
        M_BL(1, 100),   // ... jump to label 1 if R0 < 100

        I_END(), 
    };
    
    size_t size = sizeof(program) / sizeof(ulp_insn_t);
    ulp_process_macros_and_load(0, program,&size);
    ulp_run(0);

    // in which it also prints the debug output mentioned earlier
    // Serial.printf("ULP Mem in main app: %d\n", CONFIG_ESP32S3_ULP_COPROC_RESERVE_MEM);
}

void setup()
{
    Serial.begin(115200);
    ULP_Initialization();
}

void loop()
{
    for (int i = 0; i < 10; i++)
    {
        Serial.printf("RTC_SLOW_MEM[%d]: %d\n", i, RTC_SLOW_MEM[i] & 0xFFFF);
    }

Serial.printf("\n");
    delay(1000);
}
```

### Debug Message

```plain
fatal error:
   14 | #include "esp32s3/ulp.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: esp32s3/ulp.h: No such file or directory
```

### Other Steps to Reproduce

No response

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-05-06:
@me-no-dev - The Lib Builder PR is https://github.com/espressif/esp32-arduino-lib-builder/pull/176

Llgok, 2024-05-06:
> @Llgok - issue confirmed. After investigaing it, I have concluded that the ULP sdkconfig has changed from IDF 4.x to IDF 5.x - the last is used by Arduino Core 3.0.0. Therefore, it is necessary to fix it in Arduino Lib Builder repository and rebuild the tools, sdk folders and all other resources.
> 
> It may be fixed for Arduino Core 3.0.0-RC2, which shall be released along this week.
> 
> @me-no-dev - I have submitted a PR to Lib Builder to adjust the sdkconfig settings to add ULP support for S3, S2 and C6. Please take a look and, if possible, merge it on time for the RC2 release. Thanks!

Thank you very much for your reply. This issue has been bothering me for several days. Being able to use ULP on Arduino is really helpful. I am very grateful for the continuous contributions from your open-source team.

SuGlider, 2024-05-06:
@Llgok | @me-no-dev 
IDF 5.1 only allows for exclusive choice for the S2/S3 ULP: FSM ULP or RISCV ULP. 
It can't have both at the same time in the sdkconfig.

When enabling both FSM and RISCV ULP, as it was done in Lib Builder, it will only build for RISCV ULP, leaving FSM out (no `ulp.h` file with the FSM Macros).

In the PR https://github.com/espressif/esp32-arduino-lib-builder/pull/176, I have disabled RISCV ULP and only enabled FSM ULP.
Therefore, FSM `ulp.h` will be available and this is the only way to use ULP within Arduino at this time.
We don't cover RISCV ULP in Arduino.

The Lib Builder PR adds `ulp.h` file correctly.

Llgok, 2024-05-07:
> @Llgok | @me-no-dev IDF 5.1 only allows for exclusive choice for the S2/S3 ULP: FSM ULP or RISCV ULP. It can't have both at the same time in the sdkconfig.
> 
> When enabling both FSM and RISCV ULP, as it was done in Lib Builder, it will only build for RISCV ULP, leaving FSM out (no `ulp.h` file with the FSM Macros).
> 
> In the PR [espressif/esp32-arduino-lib-builder#176](https://github.com/espressif/esp32-arduino-lib-builder/pull/176), I have disabled RISCV ULP and only enabled FSM ULP. Therefore, FSM `ulp.h` will be available. This is, at this time, the only way to use ULP within Arduino. We don't cover RISCV ULP in Arduino.
> 
> The latest commit I've done to Lib Builder PR#176 adds `ulp.h` file correctly to the include folder.

OK

Jason2866, 2024-05-14:
Fixed in latest master with merge of new built libs

closed 2024-05-14 by Jason2866